David00/rpi-power-monitor #2: I have got some issue

Riri45, 2020-07-01:
The adc which is mcp3008 is overheated, I am not using pcb, I use your early method at https://community.openenergymonitor.org/t/diy-current-monitor-on-raspberry-pi-my-power-calculation-isnt-accurate-help-please/13266/50 by using breadboard and gpio breakout broad, did the post installation setup until the "file written"

David00, 2020-07-01:
Hi Riri45, can you elaborate on the actual issue? What exactly isn't working with your breadboard circuit?

Riri45, 2020-07-01:
Sorry for not being clear at first
I have some questions
1.is it normal for the ADC overheat?
2. 
[image]
The adc does not showing any data, I think I might damage my adc due to overheat

David00, 2020-07-01:
The MCP3008's datasheet says it is rated for operation up to 85 degrees Celsius, and I bet there is still quite a margin for operating in higher temps above 85C.  That's pretty warm! I'm not sure what you mean by overheat. Have you measured the temperature with an infrared temperature gun?

Make sure all your breadboard connections are correct, including the 4 required connections to the "control" side of the MCP3008 (power, ground, clock, and data).

Riri45, 2020-07-01:
sorry I forgot to ask is it okay if I'm using the opamp lm385 instead of lmV321

Riri45, 2020-07-01:
Problem solved the ADC is overheated due to my mistake connecting the t cobbler and the gpio cable wrongly at my raspberry pi, Thank you for sharing your knowledge, much appreciated

David00, 2020-07-01:
Great, glad you figured it out!

closed 2020-07-01 by David00

Riri45, 2020-07-12:
I would like to ask one more question if my input voltage is 230Vac compared to your input voltage 120Vac, at what are I should tweak in your phyton code?

David00, 2020-07-12:
I'm currently working on making the setup process easier for users to input values like this, along with the phase correction constants.  Right now there isn't actually a variable defined for source AC voltage. It's currently set as a ratio of your AC step down transformer multiplied by the ratio of the conditioning circuitry for the AC input (which is not very intuitive to figure out), but I've detailed the process below.

> I will be introducing a new "config.py" file for these values to be specified which should make the process easier, but it would be extremely helpful if you can do the following and confirm whether or not it works for your 230V input.

Find the ratio of your AC to AC step down transformer that puts out 9V AC. Ideally you should take an actual measurement of your A/C voltage source and then a measurement of your A/C transformer's output and do the following calculations based on that because the actual labels on the transformer itself usually aren't very accurate. Let's just say for this example that you measured 230V at the wall, and 9V at the transformer output:

1. Divide your transformer's output voltage by the grid voltage: 230/9 = 25.5556.  
2. Multiply your answer by 11.  25.5556 * 11 = 281.111
    (11 is the ratio of the conditioning circuitry that the 9V AC source goes through on the PCB)

3. In [this line](https://github.com/David00/rpi-power-monitor/blob/master/power-monitor.py#L190) of the code, replace the `126.5` with `281.111`. 

4. Uncomment [line 514](https://github.com/David00/rpi-power-monitor/blob/master/power-monitor.py#L514) so that the software prints the line voltage for every sample.  

5. Save the file and restart your code - you should now see your line voltage readings printed out to your terminal window, and it should be in the area of 230V.  You'll still have to tune a bit for accuracy to improve the accuracy even further.

The accuracy tuning is a constant that is specified here [in line 39](https://github.com/David00/rpi-power-monitor/blob/master/power-monitor.py#L39).  You should compare a direct reading from your multi-meter to the printed Line Voltage message on the terminal. If the terminal is higher than the multi-meter, reduce the `AC_voltage_accuracy_factor` variable. If the terminal is lower than the multi-meter, increase the `AC_voltage_accuracy_factor`.

`AC_voltage_accuracy_factor` is just a constant that is multiplied by the calculated AC voltage. So, in my code, I needed a value of 1.075 for my AC voltage accuracy factor. That means my readings were a bit lower than my multi-meter, so 1.075 has the effect of increasing the calculated value by 7.5%.

Also, I have tried to find a 230V to 9V AC to AC transformer for 230V 50Hz grids and haven't found one. Can you please share the model number of the one you've acquired so we can help other users outside of the US?

Riri45, 2020-07-14:
sorry for a late reply, got some stuff needs to be done. https://www.aliexpress.com/item/4000276268327.html here is the link for the 230V/9V transformer.

i cannot access the influxdb and got some issue at the grafana resulting to this 
[image]

David00, 2020-07-15:
Try to restart grafana with `docker restart grafana`. 

If it continues to happen, can you share the output of `docker ps` and `docker logs grafana`?

Riri45, 2020-07-15:
Tried to run the influx docker but the `docker ps` showing this 
[image]

Riri45, 2020-07-15:
Tried this too eventually resulting the same output
[image]

David00, 2020-07-15:
Something is running on your Pi that is using port 8086. You'll either have to figure out what it is and stop it, or change the port that the Docker influx container is binded to (and make the changes in the Python code to connect to the changed InfluxDB port).

Try this command on your Pi to see what is using 8086:

`sudo netstat -lntp | grep 8086`

Riri45, 2020-07-15:
I did and it shows this
[image]

Riri45, 2020-07-15:
managed to solved the influx issue but encounter with this
[image]

David00, 2020-07-17:
Have you ran the `power-monitor.py` file yet? [Running it for the first time](https://github.com/David00/rpi-power-monitor/wiki/Software-1.-Post-Installation-Setup#running-software-for-the-first-time) will create the database.