David00/rpi-power-monitor #5: 3 Phase Power

HarvsG, 2020-07-24:
Starting an issue to discuss the possibility (and perhaps track the progress) of supporting 3 phase power.

HarvsG, 2020-07-24:
We

1024jon, 2020-07-24:
+1 for me!

nannerbm60, 2020-07-24:
+1 for me too. Need mechanism for 3-phase monitoring with total consumption, total solar and net consumption 

Nice work to date!

David00, 2020-07-26:
This is something I would love to work on, but I don't have a 3-phase supply to test and calibrate against.

If anyone has access to a lab grade 3-phase power supply and would like to assist with the development of 3-phase monitoring features, please send me an email at github@dalbrecht.tech.

fasmide, 2020-07-27:
This would be great and enable the project to be able to measure whole-house consumption across most European countries

As far as I know, the board would need at least 3 voltage measuring inputs for this to work...

i-am-snappy, 2020-08-04:
Just confirming that the 3 phase limitation is in software and not hardware? If that's the case, I'll be registering my interest on the board ASAP.

David00, 2020-08-04:
> Just confirming that the 3 phase limitation is in software and not hardware? If that's the case, I'll be registering my interest on the board ASAP.

The answer to your question is both yes and no. The code can be modified to _reasonably calculate_ 3-phase power usage, but there will be some small accuracy sacrifices. This is because, in the current single/split-phase implementation, there is only one voltage input sample, and the sample isn't just measuring the voltage level. It's also monitoring and capturing the harmonics of the voltage wave so that the current samples can be adjusted.

In a 3-phase system, each phase is **supposedly** offset by 120 degrees. With only one voltage input, we'd have to assume that this is true - we can't actually measure the offset to get its exact value. What if the real offset is 119.2 degrees? Or 121? These small tolerances will induce very small errors in the measurement and subsequent calculations. Yes, the variances I'm talking about here are a very small amount, but I am pointing it out just to say that in theory, it would not be as accurate as the current single/split-phase implementation is. I don't know how quantifiable this theoretical inaccuracy is, but I would take an educated guess and say that it would add an existing 1-2% accuracy variance.

In short, it's more of a software issue. The problem with assuming the phase difference and not being to measure it a rather small issue. Adding support for 3-phase monitoring with the existing hardware would still be much more accurate than the cheap monitoring systems out there that don't take **any** voltage samples. If you're able to assist with testing on a 3-phase system, I am definitely interested in working together to bring the software support for 3-phase to this project.

sethkaz, 2020-08-04:
I would say that you shouldn't worry tooo much about the phase offsets, as that should be pretty constant for a given supply (whether it be utility power or generated power).  What would be of interest is to see the relative voltages.  This would show if one phase is being loaded down more than the other phases. 

This loading would also be interesting in a 2-phase system that is available in US households.  It's not separated by 120 degrees, but the difference in voltages on each leg can show if the loads on the house are put on evenly.  Arguably, with how little a given house draws (compared to a factory or warehouse), that delta isn't going to mandate changes by the given homeowner.

I would say from a HW perspective, you'd need to get a 3-phase transformer or 3 single-phase transformers, and wire them up.  I see that you're using a 9VAC external supply to currently measure the voltage.  You could use this method for the 3-phase, but you'd need to make sure that the supply doesn't regulate the voltage.

fasmide, 2020-08-04:
The phase offset is just one problem with having only 1 voltage measurement, there is no guarantee that the 3 phases are at the same voltage level, they are close, but can in various cases differ quite a bit...

For example, if a neighborhood for whatever reason, loads L1 more then the other L2 and L3 phases, or in my case where one phase is heavily loaded and the others are not.

Here is a screenshot from my own monitoring - I live in Denmark where all houses are installed with a 3-phased system by default:

[image]

at one point they are nearly 10v apart, under load this quickly becomes a few hundred watts

beadon, 2020-09-27:
+1 also very interested in this!

codefordough, 2020-11-15:
Also hope this can be developed. Would like to monitor a 480V 3-Ph feeder circuit that supplies a 75KVA transformer.

ariellevien, 2021-06-03:
Very interested in 3-phase. Our domestic supply voltage is WAY out of spec from the supplier (nominal 230V/ph, actual varies from 180V/ph to 240V/ph). I need to record all three.

David00, 2021-10-11:
Hi everyone, I'm happy to announce that I now have an early alpha version of software that should work for 3-phase systems. I haven't publicly released it yet and I would like some more people with access to 3-phase systems available to test. If you're interested in testing, please get in touch with me via email at "github at dalbrecht.tech"

The limitations mentioned previously in this issue thread still apply; specifically, the difference in voltages between the 3 phases will lead to an inaccuracy in the real power calculation. Accounting for the difference in voltages on each phase will require a hardware revision that I will explore in the future.

SCiunczyk, 2022-08-23:
Hi David,

have you got my email sent do "github at dalbrecht.tech" ? Or maybe it landed somewhere in your spam folder abyss ...

Slawek

David00, 2022-08-24:
@SCiunczyk Yep! Thanks for the reminder. I've been quite busy!

jt-eic, 2023-01-11:
Hello all, I'm rather new to the party and I just built my first (of possibly 3) of these monitor systems. I'd like to add my vote for 3-phase power here as well. Perhaps I'd also like to add that my interest in this isn't as much for 'consumption accuracy' as it is more for overall monitoring, and tracking historic measurements. In an environment where there is power, we need to know if we lost power and what to do next. (IE: sending alerts and beginning shut-down sequences). I know there are solid commercial systems for this type of thing, but I'm interested in how it all works and building custom DIY solutions.

SCiunczyk, 2023-01-11:
I have working 3-Phase setup (David send me the special version) I'm very happy how is working for last half year. Ask @David00  and check.

Cube3SGM, 2023-02-13:
Hi David,
I would interested in a 3 phase solution.  I have a smart meter on the supply already so I can get data in 30 minute intervals from the electricity supplier. I have 3ph coming in to the building which feeds a machine shop, this is also split into 3 x 240v single phase fuse boxes which feed the rest of the building. I would like to log data for the total consumption and the workshop consumption, as the workshop is rented to a 3rd party and not on a separate meter. Any other logging would be a bonus. Regards Simon

David00, 2023-09-10:
Hi folks, I'm going to go ahead and close this issue with the relatively recent inclusion of partial 3-phase support in the project earlier this year.

There's a discussion thread [here](https://github.com/David00/rpi-power-monitor/discussions/94) to provide feedback and ask questions, along with a link to the 3-phase setup instructions in the new docs:

https://david00.github.io/rpi-power-monitor/docs/v0.3.0/3-phase.html

closed 2023-09-10 by David00