David00/rpi-power-monitor #7: Switch from InfluxDB to VictoriaMetrics storage?

valyala, 2020-07-28:
It would be great if [VictoriaMetrics](https://github.com/VictoriaMetrics/VictoriaMetrics) could be used in the project instead of InfluxDB. It supports data ingestion over Influx line protocol, so there is no need to change anything from data ingestion side. It provides MetricsQL - PromQL-inspired query language, which is better suited for building typical dashboards in Grafana comparing to Flux or InfluxQL. See [these docs for details](https://victoriametrics.github.io/MetricsQL.html). VictoriaMetrics usually requires lower amounts of resources (CPU, RAM, persistent storage) comparing to InfluxDB - see [this article for details](https://medium.com/@valyala/insert-benchmarks-with-inch-influxdb-vs-victoriametrics-e31a41ae2893). Docker images for VictoriaMetrics are available [here](https://hub.docker.com/r/victoriametrics/victoria-metrics).

David00, 2020-07-28:
Valyala, thanks for your input. Do you know if a VictoriaMetrics supports dynamic querying capable of grouping on a per-<time interval> basis? 

One issue I've had with InfluxDB is that they don't have any way to support Grafana's more advanced dynamic date filtering capabilities. For instance, if I want to build a Grafana panel that shows me the power consumption on a per-day basis for the last 7 days, Influx can only do it if the time is hardcoded into the query.

valyala, 2020-07-29:
> Do you know if a VictoriaMetrics supports dynamic querying capable of grouping on a per- basis?

VictoriaMetrics supports [Prometheus querying API](https://prometheus.io/docs/prometheus/latest/querying/api/#range-queries), which has support for grouping by time and by labels (aka tags). Grouping by time is controlled via `step` argument that is passed to [/api/v1/query_range](https://prometheus.io/docs/prometheus/latest/querying/api/#range-queries). Grouping by labels is controlled via optional `by (...)` or `without (...)` lists that may be added to aggregate functions. See [these docs](https://prometheus.io/docs/prometheus/latest/querying/operators/#aggregation-operators) for details. It also supports aggregations via sliding window. For instance, `sum_over_time(m[1d])` returns the sum of all the data points for time series with the name `m` over the previous day (24 hours).

If you aren't familiar with PromQL and MetricsQL, then I'd suggest starting with the following introductory article - https://medium.com/@valyala/promql-tutorial-for-beginners-9ab455142085 .

> One issue I've had with InfluxDB is that they don't have any way to support Grafana's more advanced dynamic date filtering capabilities. For instance, if I want to build a Grafana panel that shows [...]

I believe all this functionality is covered by [Prometheus datasource in Grafana](https://grafana.com/docs/grafana/latest/features/datasources/prometheus/). VictoriaMetrics is compatible with this datasource - see https://victoriametrics.github.io/#grafana-setup for details.

David00, 2020-08-04:
I welcome anyone who wants to make this modification themselves through a fork. Unfortunately, I'm not able to add this functionality now given my availability and existing commitments. Closing this issue for now.

closed 2020-08-04 by David00