David00/rpi-power-monitor #9: 30A Current Transformers

nickcmeyer, 2020-08-04:
This could be a documentation suggestion or possibly expanded config parameters.

Currently using the ECS1030-L72 with a 75ohm resistor as suggested for 30A circuits.

The SCT-013 (100A) transformers needed a accuracy factor of 0.9. In comparison before I changed the accuracy calibration parameter for the ECS1030-L72 I get 82A for a 22A load measured by a clamp multimeter. That's a correction factor of 0.26. Initially this struck me as odd as I had interpreted the purpose of the accuracy calibration to only correct any variation between current transformers, burden resistors, etc. I realized the back end math isn't taking into account the size of the burden resistor or CT turns and just relying on this calibration factor. Looks like it's set up for the PZCT-02 10ohm and 1000/1 so a ECS1030 at 1000/.5 and 22ohm is 10% high like I saw. Voltage across the 75ohm burden resistor at 1000/.5 would therefore be 82/(75/(10*2))=22A, spot on. 

I'm wondering if you might want to add config parameters for current transformer turn count, as well as burden resistor size, and do that bit of math in the back end. This way "calibration" would be more true to description and optional for those that don't have a clamp meter. Since you sent 1% burden resistors I'm guessing the accuracy would already be acceptable for many.

Also, kind of a side note, if there is a 30A CT on a 15/20A circuit is there any reason not to do the burden resistor math using the circuit breaker size as max? Obviously inrush current could go over 3.3v Vref but the ADC can obviously handle 5v+, although I couldn't find a spec for breakdown. Even with a 150ohm for a 15A circuit it looks like 30A of inrush would still be under 5v and slightly negative on the other half. This would realistically be the same ADC input range as a 75ohm on my 30A air conditioner circuit which has a 60A locked rotor current.

-Nick

David00, 2020-08-04:
Nick, thanks for opening this issue! I'm glad to have your feedback on this matter because it will definitely lead to me improving the project. In my personal deployment I only used 100A CTs, so it doesn't surprise me that this issue completely slipped by me. I do have some 20A and 30A CTs on the way to test with so I will be able to experiment with various changes to find a good long term resolution to this issue soon.

I can tell you have a solid understanding of what's going on, but for the sake of other's reading this in the future, I want to elaborate.  The burden resistor is chosen so that the current output by a particular CT can be turned into a voltage that has the potential to use the full range of the ADC input channel, which ranges between 0 - 3.3V. This is so that we can squeeze the most resolution out of the CT.  So, differenty rated CTs (ie, a 30A vs. a 100A, like in this case) will output the same values, but these values mean different things. A 30A CT putting out 2V at the ADC input is different than a 100A CT putting out 2V at the ADC input. But you're exactly right - the software is tailored for 100A CTs right now.

I've already been planning to introduce "profiles" for various CTs which would include a base configuration and calibration value to get the user in a relatively close range to being accurate.

> Also, kind of a side note, if there is a 30A CT on a 15/20A circuit is there any reason not to do the burden resistor math using the circuit breaker size as max?

I think this would only be safely accomplished if the CT is rated higher than the breaker of the circuit that it's monitoring. And by safe, I mean, not over-volting the ADC input channel and potentially damaging the ADC. The burden resistor calculation actually takes into account the peak _potential_ value of the CT, which is `CT rating * sqrt(2)`, and chooses a suitable resistor value that will keep the voltage output of the secondary transformer coil under 3.3V while at peak load. The absolute max voltage for an input channel is Vdd + 0.6, so that would be about 3.9V. However, the recommended input should be no more than Vref, which is 3.3V. Remember, the CT output voltage is biased +1.65V, so you really only have 1.65V worth of "resolution" to obtain from a CT.

nickcmeyer, 2020-08-05:
Sounds good, working on getting the services figured out and interface running but I'm sure the .26 calibration factor will work just fine.

Not sure how I missed the input/out max in the datasheet, it's right there now that I look at it today... I really should get off the computer earlier. Honestly I would have thought this would have been tolerant of higher and given that it isn't I'm actually more concerned given the possibility for inrush. As you stated the burden resistors are based on the math for peak not RMS but they don't take into account inrush. I think this has the possibility for issues even for 75ohm on a 30A CT on a 30A CB (ratios are the same for 150ohm on a 15A but my air conditioner on a 30A is a better example).

So looking at just the positive side of the wave, 30A produces 1.65Vpk plus the 1.65 bias for the full range 3.3v at the ADC. It only takes ~137% to exceed the 3.9v of the input. My air conditioner compressor has a running current of 13A, and a locked rotor current of 63A. (63*sqrt(2)*75)/2000=3.34Vpk. 3.3+1.65=4.95v or 1.65-3.3=-1.65v at the ADC. I may have to come up with a pair of Zeners or a TVS if I can find one that works to clamp across the burden resistor. Or I suppose do the opposite of my original proposition and go with a 33ohm resistor on the 30A CTs.

Out of curiosity I took a look at the trip curve for the Eaton BR circuit breakers in my panel. For a 15A breaker the the trip curve shows as much as 9-10x rated current for a 10ms interruption. So with a 1000:1 burden resistor sized for the CB (15A/150, 30A/75, 100A/22, etc) the ADC could still see 15.9v+1.65v=17.55v and -14.25v. Granted there is some validity to just popping a new ADC in vs making the inputs tolerant of someone sticking a fork in an outlet.

-Nick

David00, 2020-08-05:
> Sounds good, working on getting the services figured out and interface running but I'm sure the .26 calibration factor will work just fine.

Yes, the .26 calibration factor will work no problem because that factor is simply a linear scaling variable.  

> Not sure how I missed the input/out max in the datasheet, it's right there now that I look at it today... I really should get off the computer earlier.

I know what you mean - I swear the answers only come out of the datasheet after the 10th read through it!! I should (and am trying to) end my nights earlier, so I won't be getting too much into your maths that you've shared. But, the 75 Ω suggested burden resistor will keep the CT's output voltage below 1.65V for up to ~42A.

Also, the CT core will eventually become saturated at higher and higher currents, so the maximum output value through the secondary winding will eventually flatten out. I don't know what the maximum output at core saturation would be, but the "linear" output nature of the CT will stop being linear at or very near the 30A mark.  A current of 60A through the conductor being monitored will **not** register an output that is twice the output of a 30A current.

David00, 2021-07-08:
I am closing this as it will be addressed in an upcoming major software redesign that I'm currently working on.

closed 2021-07-08 by David00