David00/rpi-power-monitor #11: Voltage tuning - AC multiplier seems to have no effect

jx, 2020-08-14:
I purchased the kit from you, and bought the 9v transformer you suggested.

I'm having an odd issue on the voltage tuning step. https://github.com/David00/rpi-power-monitor/wiki/Voltage-Tuning

Steps 1-6, i've measured the voltages, and edited the config file. My values are:
GRID_VOLTAGE = 124.2
AC_TRANSFORMER_OUTPUT_VOLTAGE = 10.4

When I get to steps 7 and 8, this is the output I get:

```
2020-08-14 18:06:24 :

+---------+---------+---------+---------+---------+---------+---------+
|         |   CT0   |   CT1   |   CT2   |   CT3   |   CT4   |   CT5   |
+---------+---------+---------+---------+---------+---------+---------+
|  Watts  | 548.882 | 538.409 | 530.817 | 532.487 | 542.351 | 548.233 |
| Current |  2.682  |  2.629  |  2.591  |  2.601  |  2.648  |  2.678  |
|   P.F.  |  0.915  |  0.916  |  0.916  |  0.916  |  0.916  |  0.916  |
| Voltage | 223.561 |         |         |         |         |         |
+---------+---------+---------+---------+---------+---------+---------+
```

The wattage numbers look absurd given nothing is connected. The voltage is also wildly inaccurate. But, carrying on, I got the multiplier, and in my case it's 0.5343. I add that to the AC line at the bottom, and my output is the same. 

```
2020-08-14 18:07:50 :
+---------+---------+---------+---------+---------+---------+---------+
|         |   CT0   |   CT1   |   CT2   |   CT3   |   CT4   |   CT5   |
+---------+---------+---------+---------+---------+---------+---------+
|  Watts  |  557.58 | 547.525 | 539.154 | 540.746 | 550.063 | 555.691 |
| Current |  2.703  |  2.652  |  2.613  |  2.623  |  2.667  |  2.694  |
|   P.F.  |  0.916  |  0.917  |  0.916  |  0.915  |  0.916  |  0.916  |
| Voltage | 225.222 |         |         |         |         |         |
+---------+---------+---------+---------+---------+---------+---------+
```

I have absolutely no idea what could be wrong. I'm 99% sure I assembled the board right.

David00, 2020-08-14:
Hey Grey! I remember your order. It was kind of a while ago and there have been a couple of software updates since then - [one of which](https://github.com/David00/rpi-power-monitor/commit/84654d6d3f9d21b949d06b19ac241c3f757fa8b1) was directly related to the AC voltage calculation.  Let's first make sure you're running the latest software.  See the Wiki page [here](https://github.com/David00/rpi-power-monitor/wiki/Software-4.-Updating-the-Code). Since you're on the calibration section, I wouldn't worry about backup up `config.py` like the instructions say. 

The `git pull` command will either say `Already up to date` or it will have a few lines of output as it pulls and updates your local version of the software. If it says it's already up to date, there may be an issue here which I'll definitely look into further.

jx, 2020-08-14:
Already ahead of you. Did fresh pulls today. Let me blow it all away and try one last time.

jx, 2020-08-14:
Ok. I blew it all away, fresh pull, and the multiplier part works! So I'll close this as user error!

closed 2020-08-14 by jx

David00, 2020-08-14:
Awesome, glad the re-install fixed it!