David00/rpi-power-monitor #13: Incorrect Capacitor Listing in assembly/schematic

rayder13, 2020-08-28:
Capacitor C4 in the schematic and component list on the assembly page lists C4 as 100nf but the footprint and image of assembled pcb on the assembly page indicate that it is a 1nf capacitor. Also L1 isn't mentioned in the assembly page anywhere. I'm still pretty new to anything beyond basic mcu projects so it may just be obvious to everyone else but the assembly page image seems to look like a wire link? May be beneficial just to clarify that on the assembly page.

David00, 2020-08-28:
Hi rayder13, thanks for opening an issue. Can you clarify exactly where you're seeing discrepancies for capacitor C4? I just looked through the Hardware Assembly page and I don't see any problems - the only image where C4 is shown is a top down view and you can't see the markings on it. 

You're correct in observing that L1 is a wire link. The inductor L1 was in the prototype schematic just in case I found the PCB needing some additional signal filtering on the 3.3V supply.  It turned out that the circuit didn't need it, so a wire link is used in its place. It was one of those "just in case" sort of things. I'll update the Hardware Assembly page to mention this, but L1 will be removed from both the schematic and PCB in a future revision.

rayder13, 2020-08-28:
You may not be able to see the markings on the capacitor but the physical sizing of the 4 capacitors is noticeable. In the "Before you begin" section, the component table lists C4 as a 100nF Electrolytic which are the large box shaped caps, where the assembled pcb top down image shows C4 as the smallest cylindrical cap (In comparison to C6). And the op-amp circuit portion of the schematic also shows it as 100nF while C5 is 1nF.

David00, 2020-08-28:
Thanks for clarifying. Electrolytic capacitors are actually the can style ones in the picture. The yellow box shaped caps you're referring to are polyester/plastic film capacitors.
C4 is indeed an electrolytic (can style) 4x7mm 100nF / 0.1µF 50V capacitor.
C5 is also a electrolytic (can style) 4x7mm 1µF 50V capacitor.

They are the same form factor/footprint.

I took a look at the BOM that's linked on the Hardware 1 page and it seems to be missing the 100nF electrolytic capacitor. Now that I think about it, I used some 100nF capacitors I had on hand when I built my first few boards so they were never purchased/included in the BOM. I'll update the BOM to include them. 
Here is the specific component for C4 for reference:
https://lcsc.com/product-detail/Aluminum-Electrolytic-Capacitors-Leaded_CX-Dongguan-Chengxing-Elec-0-1uF-50V-4-7_C45664.html

David00, 2020-08-28:
I am marking this as closed via eaedc6a that includes the specific component for C4. I think since it was missing from the BOM it was contributing to your confusion over the capacitor types.  We can continue discussion though if you need more clarification. Thanks again for opening the issue - it lead me to notice the missing capacitor in the BOM.

closed 2020-08-28 by David00

rayder13, 2020-08-28:
No that's fine. I just assumed the BOM was from when you ordered so I figured the assembly page was incorrect. It was just a combination of an oversight on your behalf and a lack of understanding on my behalf. Either way if it wasn't me then somebody else at some point would have been scratching their head trying to work out what was going on lol.

rayder13, 2020-08-28:
Don't forget to update the link on the Acquiring the Components page. It still points to the original 1.1 BOM.

commit eaedc6ac (2020-08-28): Update BOM to include the 100nF capacitor for C4

Fixes #13