David00/rpi-power-monitor #15: Wiki instructions not working to run Docker containers as services

David00, 2020-09-09:
I'm opening this issue as I've received several reports via email about the influxdb part of the [Software 2. Running as a Service](https://github.com/David00/rpi-power-monitor/wiki/Software-2.-Running-as-a-Service) instructions in the project Wiki not working for some folks.

This issue will track the status of my investigation into the problem and the eventual resolution.

The problem is usually encountered when starting the software in terminal or regular mode, because both of these modes save to the database.  The traceback that Python provides is extremely verbose, so I will clean that up too, but if you see a `ConnectionRefusedError: [Errno 111] Connection refused` in your output, you are likely hitting this issue. You may also see the following from `sudo systemctl status influx`:

```
pi@raspberrypi:~ $ sudo systemctl status influx
● influx.service - Influx container
   Loaded: loaded (/etc/systemd/system/influx.service; enabled; vendor preset: enabled)
   Active: failed (Result: start-limit-hit) since Thu 2020-09-03 10:21:35 PDT; 5 days ago
  Process: 896 ExecStart=/usr/bin/docker start influx (code=exited, status=0/SUCCESS)
 Main PID: 896 (code=exited, status=0/SUCCESS)

Sep 03 10:21:35 raspberrypi systemd[1]: influx.service: Service RestartSec=100ms expired, scheduling restart.
Sep 03 10:21:35 raspberrypi systemd[1]: influx.service: Scheduled restart job, restart counter is at 5.
Sep 03 10:21:35 raspberrypi systemd[1]: Stopped Influx container.
Sep 03 10:21:35 raspberrypi systemd[1]: influx.service: Start request repeated too quickly.
Sep 03 10:21:35 raspberrypi systemd[1]: influx.service: Failed with result 'start-limit-hit'.
Sep 03 10:21:35 raspberrypi systemd[1]: Failed to start Influx container.
```

A quick workaround is to disable the InfluxDB service, then restart the Docker container manually:
`sudo systemctl disable influx`
`docker restart influx`

... but a more permanent fix along with revised Wiki instructions will be coming soon.

I will also be adding some error handling in the code to provide easier/more readable output instead of the multiple level traceback.

David00, 2020-09-11:
To resolve this issue, I've made some changes to a few places.  In addition to the new code, there are a few OS environment changes you'll have to make to apply this improvement:

1. Change your Docker container restart policies:

    ```
    docker update --restart always influx
    docker update --restart always grafana
    ```

2. Unlink, then remove the influx and grafana systemd files:

    ```
    sudo systemctl disable grafana.service
    sudo systemctl disable influx.service
    sudo rm /etc/systemd/system/grafana.service
    sudo rm /etc/systemd/system/influx.service
    ```

3. Update the restart timer and count in the power-monitor.service file:

    `sudo nano /etc/systemd/system/power-monitor.service`

    Make the contents of this file match the contents listed in step 2 of the [running as a service](https://github.com/David00/rpi-power-monitor/wiki/Software-2.-Running-as-a-Service) instructions in the Wiki.

4. Run `sudo systemctl daemon-reload` to update systemd with the new changes.

5. Get the new version of the code from this branch. **Back up your config.py file if you've already gone through the phase correction and calibration procedures!!**

    ```
    cd ~/rpi-power-monitor
    git config remote.origin.fetch "+refs/heads/*:refs/remotes/origin/*"
    git fetch
    git stash
    git checkout auto-phase-correction-experimental
    pip3 install -r requirements.txt
    ```

6. Replace the configuration settings in `config.py` if you went through the calibration process or if you are using a remote DB server.  **DB server connection settings were moved to config.py in this commit**

That _should_ be all that's needed... I want to keep this issue open and on a separate branch for a little while until we have a few people field test this.

speedyzerotwo, 2020-09-11:
Hi David,

Not quite working, I get this...

pi@raspberrypi:~/rpi-power-monitor $ git stash

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got 'pi@raspberrypi.(none)')
Cannot save the current index state

So

pi@raspberrypi:~/rpi-power-monitor $  git config --global user.email my_email_address
pi@raspberrypi:~/rpi-power-monitor $  git config --global user.name " Me"

and again

pi@raspberrypi:~/rpi-power-monitor $ git stash
Saved working directory and index state WIP on master: e98167a Add files via upload

pip3 install -r requirements.txt

Requirement already satisfied: pycairo>=1.11.1 in /home/pi/.local/lib/python3.7/site-packages (from PyGObject==3.30.4->-r requirements.txt (line 13)) (1.19.1)
requests 2.21.0 has requirement urllib3<1.25,>=1.21.1, but you'll have urllib3 1.25.10 which is incompatible.

running docker ps
CONTAINER ID        IMAGE               COMMAND                  CREATED             STATUS              PORTS                    NAMES
af31580b7223        influxdb            "/entrypoint.sh infl…"   2 weeks ago         Up 5 minutes        0.0.0.0:8086->8086/tcp   influx
3b69cb59cef3        grafana/grafana     "/run.sh"                2 weeks ago         Up 5 minutes        0.0.0.0:3000->3000/tcp   grafana

running sudo systemctl status power-monitor

gives...

● power-monitor.service - Raspberry Pi Power Monitor
   Loaded: loaded (/etc/systemd/system/power-monitor.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-09-11 06:37:58 BST; 4min 25s ago
  Process: 323 ExecStart=/usr/bin/python3.7 /home/pi/rpi-power-monitor/power-monitor.py (code=exite$
 Main PID: 323 (code=exited, status=1/FAILURE)

Sep 11 06:37:58 raspberrypi systemd[1]: power-monitor.service: Service RestartSec=1s expired, sched$
Sep 11 06:37:58 raspberrypi systemd[1]: power-monitor.service: Failed to schedule restart job: Unit$
Sep 11 06:37:58 raspberrypi systemd[1]: power-monitor.service: Failed with result 'exit-code'.

Hope this helps

Regards
Dave

From: David00
Sent: 11 September 2020 03:42
To: David00/rpi-power-monitor
Cc: speedyzerotwo; Manual
Subject: Re: [David00/rpi-power-monitor] Wiki instructions not working to runDocker containers as services (#15)

To resolve this issue, I've made some changes to a few places. In addition to the new code, there are a few OS environment changes you'll have to make to apply this improvement:
1. Change your Docker container restart policies:
2. docker update --restart always influx
3. docker update --restart always grafana
4. Unlink, then remove the influx and grafana systemd files:
5. sudo systemctl disable grafana.service
6. sudo systemctl disable influx.service
7. sudo rm /etc/systemd/system/grafana.service
8. sudo rm /etc/systemd/system/influx.service
9. Update the restart timer and count in the power-monitor.service file:
sudo nano /etc/systemd/system/power-monitor.service
Make the contents of this file match the contents listed in step 2 of the running as a service instructions in the Wiki.
10. Run sudo systemctl daemon-reload to update systemd with the new changes.
11. Get the new version of the code from this branch. Back up your config.py file if you've already gone through the phase correction and calibration procedures!!
12. cd ~/rpi-power-monitor
13. git config remote.origin.fetch "+refs/heads/*:refs/remotes/origin/*"
14. git fetch
15. git stash
16. git checkout auto-phase-correction-experimental
17. pip3 install -r requirements.txt
18. Replace the configuration settings in config.py if you went through the calibration process or if you are using a remote DB server. DB server connection settings were moved to config.py in this commit
That should be all that's needed... I want to keep this issue open and on a separate branch for a little while until we have a few people field test this.

David00, 2020-09-11:
Thanks for trying out the new code. The issue you ran into isn't explicitly stated, but I did some more testing and uncovered something I didn't notice earlier. I think the InfluxDB container simply needed more time to startup than the 2 seconds I was giving it.

I just pushed [bc54b2](https://github.com/David00/rpi-power-monitor/commit/77c2e00afca51eca982a08bb35cbdcee2bc545b2) which increases the wait time after the automated restarting of the influxDB container.

Can you try the following? We should make sure the version of your code works as-is before trying to use systemd to control it. The commands below will update your code to grab that update I just pushed, then stop the service, then start the software "manually".

```
cd ~/rpi-power-monitor
git pull
sudo systemctl stop power-monitor
python3 power-monitor.py
```

This should work without any problems.  Then, to test the features of this branch, you can stop the execution you just started with ctrl-c, then stop the influxDB container, and try to launch the power-monitor.py again:

Issue a `ctrl-c` first, then...
```
docker stop influx
cd ~/rpi-power-monitor
python3 power-monitor.py
```

Here's a sample of my terminal output after following the steps above (note that I have to manually type in `yes` in the 2nd line):

```
pi@raspberrypi4:~/rpi-power-monitor $ docker stop influx
influx
pi@raspberrypi4:~/rpi-power-monitor $ cd ~/rpi-power-monitor/
pi@raspberrypi4:~/rpi-power-monitor $ python3 power-monitor.py 
2020-09-11 07:15:13 : ... Trying to connect to database at: localhost:8086
2020-09-11 07:15:19 : It appears that your InfluxDB container is not running. Would you like me to try to restart it? Please enter yes or no: 
yes
2020-09-11 07:15:22 : ... restarting your docker container. Please wait... 
2020-09-11 07:15:27 : ... checking to see if the container is running now...
2020-09-11 07:15:27 : ... container successfully started!
2020-09-11 07:15:27 : ... Starting Raspberry Pi Power Monitor
2020-09-11 07:15:27 : Press Ctrl-c to quit...
```

This update should allow the end user to start the power-monitor software without worrying whether or not the InfluxDB docker container is running or not.  When the software is ran via systemd, permission to restart the influxDB container is assumed to be granted, so no user intervention will be necessary.

Thanks again,
David

speedyzerotwo, 2020-09-11:
Hi Dave,

I get the following

pi@raspberrypi:~ $ cd ~/rpi-power-monitor
pi@raspberrypi:~/rpi-power-monitor $ git pull
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Total 3 (delta 2), reused 3 (delta 2), pack-reused 0
Unpacking objects: 100% (3/3), done.
From https://github.com/David00/rpi-power-monitor
   d24e0ef..77c2e00  experimental-influxdb-container-restart -> origin/experimental-influxdb-container-restart
Already up to date.
pi@raspberrypi:~/rpi-power-monitor $ sudo systemctl stop power-monitor
pi@raspberrypi:~/rpi-power-monitor $ python3 power-monitor.py
/home/pi/.local/lib/python3.7/site-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.25.10) or chardet (3.0.4) doesn't match a supported version!
  RequestsDependencyWarning)
Traceback (most recent call last):
  File "power-monitor.py", line 16, in <module>
    from config import logger, ct_phase_calibration, ct0_channel, ct1_channel, ct2_channel, ct3_channel, ct4_channel, board_voltage_channel, v_sensor_channel, ct5_channel
ImportError: cannot import name 'ct_phase_calibration' from 'config' (/home/pi/rpi-power-monitor/config.py)
pi@raspberrypi:~/rpi-power-monitor $

Regards
Dave

From: David00
Sent: 11 September 2020 07:19
To: David00/rpi-power-monitor
Cc: speedyzerotwo; Manual
Subject: Re: [David00/rpi-power-monitor] Wiki instructions not working to runDocker containers as services (#15)

Thanks for trying out the new code. The issue you ran into isn't explicitly stated, but I did some more testing and uncovered something I didn't notice earlier. I think the InfluxDB container simply needed more time to startup than the 2 seconds I was giving it.
I just pushed bc54b2 which increases the wait time after the automated restarting of the influxDB container.
Can you try the following? We should make sure the version of your code works as-is before trying to use systemd to control it. The commands below will update your code to grab that update I just pushed, then stop the service, then start the software "manually".
cd ~/rpi-power-monitor
git pull
sudo systemctl stop power-monitor
python3 power-monitor.py
This should work without any problems. Then, to test the features of this branch, you can stop the execution you just started with ctrl-c, then stop the influxDB container, and try to launch the power-monitor.py again:
Issue a ctrl-c first, then...
docker stop influx
cd ~/rpi-power-monitor
python3 power-monitor.py
Here's a sample of my terminal output after following the steps above (note that I have to manually type in yes in the 2nd line):
pi@raspberrypi4:~/rpi-power-monitor $ docker stop influx
influx
pi@raspberrypi4:~/rpi-power-monitor $ cd ~/rpi-power-monitor/
pi@raspberrypi4:~/rpi-power-monitor $ python3 power-monitor.py 
2020-09-11 07:15:13 : ... Trying to connect to database at: localhost:8086
2020-09-11 07:15:19 : It appears that your InfluxDB container is not running. Would you like me to try to restart it? Please enter yes or no: 
yes
2020-09-11 07:15:22 : ... restarting your docker container. Please wait... 
2020-09-11 07:15:27 : ... checking to see if the container is running now...
2020-09-11 07:15:27 : ... container successfully started!
2020-09-11 07:15:27 : ... Starting Raspberry Pi Power Monitor
2020-09-11 07:15:27 : Press Ctrl-c to quit...
This update should allow the end user to start the power-monitor software without worrying whether or not the InfluxDB docker container is running or not. When the software is ran via systemd, permission to restart the influxDB container is assumed to be granted, so no user intervention will be necessary.
Thanks again,
David

David00, 2020-09-11:
@speedyzerotwo, I'm sorry, I put the wrong branch name in the instructions above. I just noticed it when looking through the logs in your comment. Not sure how that happened, but I edited my comment above to fix it.  Can you try everything again after doing a `git checkout experimental-influxdb-container-restart`?

After grabbing the branch, you'll still need to do the `pip3 install -r requirements.txt` as there is a new dependency to support these features.

speedyzerotwo, 2020-09-11:
Hi Dave,

Got this!

pi@raspberrypi:~/rpi-power-monitor $ git checkout experimental-influxdb-container-restart?
error: pathspec 'experimental-influxdb-container-restart?' did not match any file(s) known to git
pi@raspberrypi:~/rpi-power-monitor $

Regards
Dave
From: David00
Sent: 11 September 2020 16:00
To: David00/rpi-power-monitor
Cc: speedyzerotwo; Mention
Subject: Re: [David00/rpi-power-monitor] Wiki instructions not working to runDocker containers as services (#15)

@speedyzerotwo, I'm sorry, I put the wrong branch name in the instructions above. I just noticed it when looking through the logs in your comment. Not sure how that happened, but I edited my comment above to fix it. Can you try everything again after doing a git checkout experimental-influxdb-container-restart?
After grabbing the branch, you'll still need to do the pip3 install -r requirements.txt as there is a new dependency to support these features.

David00, 2020-09-11:
@speedyzerotwo It looks like you grabbed the question mark at the end of the line. Sorry, that was poorly placed!

`git checkout experimental-influxdb-container-restart`

speedyzerotwo, 2020-09-11:
No problem, I googled it, then realised.

I have seemed to have lost your previous instructions but guessing...

I tried this!! Can you resend the next steps?

Thanks Dave

pi@raspberrypi:~/rpi-power-monitor $ sudo systemctl stop power-monitor
pi@raspberrypi:~/rpi-power-monitor $ python3 power-monitor.py
Traceback (most recent call last):
  File "power-monitor.py", line 7, in <module>
    import influx_interface as infl
  File "/home/pi/rpi-power-monitor/influx_interface.py", line 6, in <module>
    from config import logger, db_settings
  File "/home/pi/rpi-power-monitor/config.py", line 15
    <<<<<<< Updated upstream
     ^
SyntaxError: invalid syntax
pi@raspberrypi:~/rpi-power-monitor $

From: David00
Sent: 11 September 2020 18:11
To: David00/rpi-power-monitor
Cc: speedyzerotwo; Mention
Subject: Re: [David00/rpi-power-monitor] Wiki instructions not working to runDocker containers as services (#15)

@speedyzerotwo It looks like you grabbed the question mark at the end of the line. Sorry, that was poorly placed!
git checkout experimental-influxdb-container-restart

speedyzerotwo, 2020-09-11:
OK So I copied my original saved config.py back over as the comments are git related (I’m not familiar with Git)

Then tried again and got this,

pi@raspberrypi:~/rpi-power-monitor $ python3 power-monitor.py
Traceback (most recent call last):
  File "power-monitor.py", line 7, in <module>
    import influx_interface as infl
  File "/home/pi/rpi-power-monitor/influx_interface.py", line 6, in <module>
    from config import logger, db_settings
ImportError: cannot import name 'db_settings' from 'config' (/home/pi/rpi-power-monitor/config.py)
pi@raspberrypi:~/rpi-power-monitor $

Sorry for all the mails I’m googling and mailing as I go

Regards
Dave

From: David Latter
Sent: 11 September 2020 18:17
To: David00/rpi-power-monitor
Subject: RE: [David00/rpi-power-monitor] Wiki instructions not working torunDocker containers as services (#15)

No problem, I googled it, then realised.

I have seemed to have lost your previous instructions but guessing...

I tried this!! Can you resend the next steps?

Thanks Dave

pi@raspberrypi:~/rpi-power-monitor $ sudo systemctl stop power-monitor
pi@raspberrypi:~/rpi-power-monitor $ python3 power-monitor.py
Traceback (most recent call last):
  File "power-monitor.py", line 7, in <module>
    import influx_interface as infl
  File "/home/pi/rpi-power-monitor/influx_interface.py", line 6, in <module>
    from config import logger, db_settings
  File "/home/pi/rpi-power-monitor/config.py", line 15
    <<<<<<< Updated upstream
    ^
SyntaxError: invalid syntax
pi@raspberrypi:~/rpi-power-monitor $

From: David00
Sent: 11 September 2020 18:11
To: David00/rpi-power-monitor
Cc: speedyzerotwo; Mention
Subject: Re: [David00/rpi-power-monitor] Wiki instructions not working to runDocker containers as services (#15)

@speedyzerotwo It looks like you grabbed the question mark at the end of the line. Sorry, that was poorly placed!
git checkout experimental-influxdb-container-restart

David00, 2020-09-11:
Yes, the steps are in one of the first comments above all our replies here:

https://github.com/David00/rpi-power-monitor/issues/15#issuecomment-690839138 (https://github.com/David00/rpi-power-monitor/issues/15#issuecomment-690839138)

David00, 2020-09-14:
I removed the comments from both myself and @speedyzerotwo because they were based from a typo I made in the instructions above and not necessarily specific to this issue/fix. The typo has been corrected thanks to @speedtyzerotwo's feedback and the above instructions should work for anyone who have already setup their Pi with the installation instructions in the README.

Please note, the primary installation instructions have been updated to setup your environment for this improvement, so if you are starting from scratch, please follow the installation instructions in the [README](https://github.com/David00/rpi-power-monitor/blob/master/README.md#installation). I will be merging the `experimental-influxdb-container-restart` branch into master soon and closing this issue as resolved.

closed 2020-09-15 by David00

David00, 2020-09-15:
The changes introduced in the commits mentioned above have been merged into master. You can now start the power monitor software without worrying about the status of the InfluxDB container. If the InfluxDB container is down, it will be automatically started prior to the power-monitor software starting.
If you use a remote InfluxDB server, the software will exit if the DB connection cannot be established.

vair327, 2020-09-16:
David

 

The power meter works perfectly. I am having a problem getting the program to work as a service. This is what I get when I check if the service is running.

pi@powermonitor:~ $ sudo systemctl status power-monitor

● power-monitor.service - Raspberry Pi Power Monitor

   Loaded: loaded (/etc/systemd/system/power-monitor.service; enabled; vendor pr

   Active: failed (Result: exit-code) since Mon 2020-09-14 22:07:36 PDT; 19h ago

  Process: 383 ExecStart=/usr/bin/python3.7 /home/pi/rpi-power-monitor/power-mon

Main PID: 383 (code=exited, status=1/FAILURE)

 

Sep 14 22:07:36 powermonitor systemd[1]: power-monitor.service: Service RestartS

Sep 14 22:07:36 powermonitor systemd[1]: power-monitor.service: Failed to schedu

Sep 14 22:07:36 powermonitor systemd[1]: power-monitor.service: Failed with resu

lines 1-9/9 (END)...skipping...

● power-monitor.service - Raspberry Pi Power Monitor

   Loaded: loaded (/etc/systemd/system/power-monitor.service; enabled; vendor preset: enabled)

   Active: failed (Result: exit-code) since Mon 2020-09-14 22:07:36 PDT; 19h ago

  Process: 383 ExecStart=/usr/bin/python3.7 /home/pi/rpi-power-monitor/power-monitor.py (code=exited, status=1/FAILURE)

Main PID: 383 (code=exited, status=1/FAILURE)

 

Sep 14 22:07:36 powermonitor systemd[1]: power-monitor.service: Service RestartSec=1s expired, scheduling restart.

Sep 14 22:07:36 powermonitor systemd[1]: power-monitor.service: Failed to schedule restart job: Unit influx.service not found.

Sep 14 22:07:36 powermonitor systemd[1]: power-monitor.service: Failed with result 'exit-code'.

~

~

~

 

It works fine if I start it manually.

 

I also look forward to instructions on the dash board.

 

Thanks again.

David00, 2020-09-16:
@vair327 I'm sorry, I'm not sure what happened to my revision of the [Software 2. Running as a Service](https://github.com/David00/rpi-power-monitor/wiki/Software-2.-Running-as-a-Service) page. It looks like the section for the `power-monitor.service` file had the old contents displayed.

Based on the logs you shared, you'll have to edit your `/etc/systemd/system/power-monitor.service` file and replace the contents with the following:

```
[Unit]
Description=Raspberry Pi Power Monitor
After=docker.service
Requires=docker.service

[Service]
Restart=always
RestartSec=1
StartLimitInterval=120
StartLimitBurst=5
User=pi
ExecStart=/usr/bin/python3.7 /home/pi/rpi-power-monitor/power-monitor.py

[Install]
WantedBy=multi-user.target
```

Then, run `sudo systemctl daemon-reload` and you should be able to start the software as a service normally with `sudo systemctl start power-monitor`.

vair327, 2020-09-16:
David

 

The changes worked perfectly.

 

Thanks

commit 93753807 (2020-09-15): Improved handling of influx container being down at the start of the software. Fixes #15.