David00/rpi-power-monitor #26: Documentation Clean Up

David00, 2021-01-03:
I've received feedback from a few people that the documentation can be difficult to follow along as it is sometimes unclear how to proceed after finishing a page. Thank you for all your feedback!

I'm opening this issue have something to reference commits to as I work towards improving the documentation.

If, at any time, you find something that is unclear or difficult to follow, please leave a comment here (even if the issue is closed in the future, feel free to reopen it), and I will address it.

clearlyunitended, 2021-01-09:
David,  I think it  would be worth-while mentioning that the CT's should be connected to the monitor board before clipping onto cables  - to avoid over-voltage on the ADC inputs.

David00, 2021-01-11:
Thanks @clearlyunitended, that is a great suggestion. I just updated this page in the Wiki:
https://github.com/David00/rpi-power-monitor/wiki/Final-Installation#installation-tips

David00, 2021-02-03:
All of the content files for the project documentation in the WIki is being moved to a branch titled `docs`.  This includes images, schematics, BOMs, and fabrication files.  I will be updating the links in the Wiki pages to reference the new branch, and then I will remove the docs folder from `master`. I understand this will break people's browser bookmarks, but it's needed to help keep things organized and easy to find.

corbolais, 2021-02-03:
In what regard should this be needed?

I've never seen a project use a doc branch for documentation. It's always a doc subdir.

Branches basically are intended for versioning the repository rather than structuring content.

I'd strongly suggest to revise this decision. On the long run it's going to be a hassle.
Just imaging having to switch branches back and forth for documentation and the rest of the repo.

David00, 2021-02-03:
There is no need to switch back and forth between branches for documentation because I do not intend for people to access documentation through the Git structure. Current documentation is hosted in the project WIki. The `docs` branch is used simply for hosting files that are linked from in the project Wiki.  

When people clone the project per the installation instructions, they only clone the master branch.  If documentation were on the master branch, they would be downloading images with no context, old/obsolete schematics, old fabrication files, etc.  This wastes both network bandwidth and disk space, because having these files on disk serves no purpose when the Wiki is hosted online.

corbolais, 2021-02-03:
Why link from the wiki to obsolete content at all?

`git clone --depth 1` lets you download only the latest revision, BTW.

Maybe I'm missing your point. Just let me know. I'm wondering why obsolete content needs to be there in the repo.

David00, 2021-02-03:
I think obsolete was too strong of a word. Here's an example of why I need to keep... _old_... stuff. This project deals with hardware which is naturally going to change over time through major and minor revisions. I might be working on v1.3 of the PCB and, but people are still using v1.2 hardware.  

When v1.3 is released, v1.2 doesn't become any less important - it just isn't the latest version anymore.  So, `git clone --depth 1` doesn't help to solve this because I'm not actually upgrading v1.2 files to v1.3 - I'm leaving v1.2 where it is, and adding new files for v1.3. People running v1.2 hardware may not be concerned with v1.3 schematics, BOMs, and fabrication files, just as people running v1.3 hardware aren't concerned with v1.1 or v1.2.  However, everyone needs access to the files relevant to their version of the project. This is why I am moving all content to a separate branch, so that when people clone `master`, they are only getting the application software needed to run the project and not all of the "extras".  

The fact that the Wiki is a separate object outside the code repo also factors into this. What I mean is, when you clone the project, you don't get the Wiki. The "docs" branch (or folder) doesn't actually contain the documentation... it contains documentation _resources_, like images, fab files, BOMs, schematics, etc. So, having all these _documentation resources_ in a docs folder on `master` without having the actual accompanying documentation just doesn't make sense and is a waste of disk space and network bandwidth for those trying to deploy this project.

I don't really care that you haven't seen a project use a separate branch for docs, to be honest. (No snarkiness intended - it just... really doesn't matter). I am using the tools I have available the way that I see fit - maybe GitHub isn't the best place for what I'm doing, but this is where the project lives for now. If you have suggestions for an approach that would make documentation more accessible, organized, and maintainable, I would love to hear it!

corbolais, 2021-02-05:
I see the trouble you're trying to solve here.

The way you're trying to handle it is just different from what others chose to, so, fair enough.

Another way I've seen alot with hardware projects is to create entire new repos for each version. Or one more is to create new branches or use tags and releases.

Personally, I'd try the route of a repo instead of a wiki and start from there.

closed 2021-07-08 by David00