David00/rpi-power-monitor #27: Monitor intake circuit or just sum the individual circuits?

haraldpaulsen, 2021-01-19:
Tried to find this in a FAQ or the documentation to no avail.

I have an apartment with a 40A intake / main circuit breaker. In addition I have 5 smaller circuits ranging from 10A to 40A.

I plan to order the kit, but should I get a 6 clamp version (ie, monitor the intake + 5 sub circuits) or just go for the 5 clamp version, and just sum the 5 sub circuits together to get the total power draw?

For all I know, the software already has a function to sum the circuits together :)

What would you people do?

RPI-PMK, 2021-01-19:
Personally I would gear up to monitor as many circuits as possible. If you have 5 circuits plus the main incomer you could either put CTs on all 6 feeds and then when you make you dashboard you can configure to have the main incomer compared with the sum of the 5 smaller circuits as a sense check. I have yet to work out exactly how I am going to use all the CTs but for me it is better to have too much information and not use it than to have too little. Hope that helps.

David00, 2021-01-19:
Hi, if monitoring your entire panel consumption is your goal, I'd actually recommend not doing this, and instead relying on a single CT on each of your mains (2 for North America, 1 for almost everywhere else).

The reason is because CTs are not perfect measurement devices. They are subject to a number of different errors. We try to correct for these as best as possible during the [calibration section](https://github.com/David00/rpi-power-monitor/wiki/Calibrating-for-Accuracy), but it is inevitable that there will be some sort of error.  It may not be much, but in theory, let's say you have a +/- 1% measurement error per CT, the single CT measuring the mains will be more accurate than summing the results of 5 CTs - all with their own +/- 1% errors.  As a cross check, I check the report I can get from my utility company regarding net consumption and production. After all, they're the ones billing me for usage, so it follows that they are the most accurate. If you don't have access to a report from your utility company then you can always go out and check on the meter yourself a few times, noting the time and the reading, and then checking the power monitor dashboard to compare.

Additionally, calculating total daily consumption and production through database queries is very CPU intensive. If you do want to try this, I suggest you modify the source code to sum the values from the individual CTs in Python before they are sent to the DB.

> I plan to order the kit, but should I get a 6 clamp version (ie, monitor the intake + 5 sub circuits) or just go for the 5 clamp version, and just sum the 5 sub circuits together to get the total power draw?

There is only 1 version of the project and it support 6 inputs, so you have the flexibility either way :)

closed 2021-01-25 by David00