David00/rpi-power-monitor #28: Kw reading disparity

Rangeman0, 2021-02-18:
Hi David, 

I recently built the kit from you and run all 6 CTs in my Australian 240v main switchboard. I got the 100A CT due to its wire size capacity but I would be better suited with a 60A to be a closer match to my power requirements - to this end I appear to be getting a disparity between the two dashboards I have installed in Grafana. The individual CT dashboard will show a constantly updated usage that is entirely correct (I have checked with clamp meters and a subpanel meter in my garage for relevant devices). When I swap dashboards to the Home power dashboard I appear to get a Kw usage for the day of almost exactly double my actual usage. I.e, if I consumed 10Kw in the day measured with my utility mains supply power meter or load tests of running the lights or AC my CT dashboard reflects this but the Home Power dashboard shows a 20Kw usage. I don't believe this to be a 100A CT range problem as I get almost exact readings to the utility meter when load testing.

I'm wondering if this is due to (somewhere in the Wiki) you discuss using a CT on your two legs of incoming power (I'm assuming you have 110v) which is different to the Australian standard of a single input of 240v. 

I've had a basic poke at the code behind the Home power dashboard but it is far beyond my knowledge. I'm assuming it's a set of code to 'double' your usage based on having two inputs in the US? 

All big assumptions on my side so I might just be doing something wrong. I'm happy to provide screenshots or such if that would help.

Not a big deal if it can't be done or if I'm doing something wrong as I'm extremely happy with everything and I'm very impressed at what it can show and how easy it was to build and install. It's just a convenience thing instead of dividing the readout on the Home Power dashboard in my head to see my daily usage. 

Thanks, 

Rowan

David00, 2021-02-18:
Hi Rowan,

Using this project on 240V is completely fine - in fact, you'll likely have better results than most of us North American folks because we have 60Hz electricity and most homes here have two phases (split phase) feeding our main electrical panels.

So, you said:
> The individual CT dashboard will show a constantly updated usage that is entirely correct 

That's a great sign because the Individual CT Dashboard is the raw data. The Home Power Dashboard is a higher level analysis of the raw data. I am assuming you've already gone through all 3 steps in the [calibration section](https://github.com/David00/rpi-power-monitor/wiki/Calibrating-for-Accuracy) of the project wiki.

> and run all 6 CTs in my Australian 240v main switchboard

By default, the _Home Power Usage_ plot in the Home Power Dashboard is the sum of all 6 input channels. So, if you're monitoring your main 240V input from the utility provider in addition to individual circuits on your panel, the _Home Power Usage_ plot will be showing the power on your mains **plus** the other 5 CTs. The same goes for the _Home Consumption_ statistic panel.

I believe this is the problem you're running into, so I'll get right to describing how to fix it. It involves minor code edits in `power-monitor.py`, but don't worry, they're easy.  

Find [this line](https://github.com/David00/rpi-power-monitor/blob/master/power-monitor.py#L476) in `power-monitor.py`.  This is the variable that is basically stored in the database and subsequently plotted in the _Home Power Usage_ plot.  See how it's defined as the sum of all the power calculations from each individual channel?

```
home_consumption_power = grid_0_power + grid_1_power + grid_2_power + grid_3_power + grid_4_power + grid_5_power + solar_power
```

Let's say - just for example purposes - that CT3 is the one that is on your main 240V incoming feed.  Your main incoming feed is how your entire home is powered, so it follows that the variable `home_consumption_power` should only contain the measurement from that single CT.  This means you'll need to change how `home_consumption_power` is defined by removing all of the other CTs that aren't measuring your 240V incoming main. The number in the `grid_#_power` variable represents the CT number.  So, you'll remove all of them that _aren't_ the specific CT that's on your mains. For example, if CT3 was measuring your main:

```
home_consumption_power = grid_3_power + solar_power
```

You can leave the `solar_power` variable because by default, it's zero, unless explicitly setup by following the net metering instructions.

You'll need to make a similar change in both the `home_consumption_current` and `net_consumption_current` variables a couple lines below the `home_consumption_power` variable above.

Make sure that you use a good text editor (like Notepad++) and also that the indentation level remains the same when making these changes because Python relies on indentations to interpret the code.

It's getting late for me now but feel free to provide feedback to my reply and I will see it tomorrow.

Lastly, thank you for opening this issue - it will lead to improvements in the Project Wiki setup instructions!

Rangeman0, 2021-02-18:
Hi David, 

Made the changes to the power-monitor.py as you suggested this morning while I had breakfast. Very easy to do. I suppose if I had dug deeper I may have stumbled across it but thanks for your very detailed and prompt response. 

I'll let it run while I'm at work and do some more testing this afternoon/tonight and see how the results go. I'm confident that this will indeed fix the problem (or my lack of attention). 

Also, general note you may wish to add - This is in no way related to your circuit but may help a future user. I originally used a Samsung dual voltage phone charger 9v/5vdc to power the pi while building and coding which eventually gave me undervoltage errors. 

This caused the pi to hang so I investigated and added a nightly reboot script I found online but still ended up with the pi freezing.

I swapped this Samsung to a USB C Google pixel charger and it has run fine ever since (still has the restart script running). It seems a high quality charger is required for consistency (and it also seems to have improved the Utility voltage reading consistency, which I didn't expect) 

Thanks again and I'll update you once I have more data collected over the next few days.

Rowan

closed 2021-03-02 by David00