David00/rpi-power-monitor #30: Local time and DayLightSaving offset

DuduNord, 2021-03-08:
Hello. Based on your experience, how will DayLight Saving will be handeled ? 
> if I understand weel, the python program does use the local time to store the date/time in the database.
> then Grafana will use this timing
I made a Cost-Of-Use dashboard and I had to substract (for some reason that is not really clear for the moment) my time difference with the UTC (here 1h because I am in France).
But because this delay will change in the PI/database, I will need to change in Grafana this offset to -2h.

Should we leave then the PI in UTC timezone and change the local zone in Grafana only ? This is what I read in some forum online.

David00, 2021-03-08:
Hi DuduNord, in general, data should be stored in the DB in UTC as a best practice.  

Grafana will use your local browser timezone setting by default with no customization needed... and your browser should use the timezone setting from your PC. Therefore, daylight savings should be reflected automatically in the dashboard when your PC time updates.

DuduNord, 2021-03-11:
Hi David.
Great !  Thanks. Can you just make a short comment then in your wiki (soft installation) to highlight this ?

closed 2021-07-08 by David00