David00/rpi-power-monitor #33: DIY kit now power

hollywoodhd, 2021-04-15:
When i connect my power monitor to the Raspberry PI the PI will not boot up and now power light.  I remove the monitor and it boots fine.  I have doubled check components and soldering not finding no problems. Any ideas?
[image]

hollywoodhd, 2021-04-15:
Found problem

closed 2021-04-15 by hollywoodhd

David00, 2021-04-15:
> Found problem

Would you mind adding a comment on what it was, so that anyone in the future checking through closed issues can rule it out? My guess would have been a short between C4 or C5 on the bottom side of the PCB.

hollywoodhd, 2021-04-16:
Ok

Yep that exactly what it was

hollywoodhd, 2021-04-16:
C4 leads on the bottom were bent and touching each other.  As soon as I bent them away for each other it worked prefect.