David00/rpi-power-monitor #35: Influx DB gets too big, producing communication issues

cgleggett, 2021-09-08:
I've been running the power monitor service for about 9 months, and recently I've seen a lot of communication issues with the Influx Db. The Influxd server was utilizing about 300% of the CPU, and the pi slowed to a crawl. I moved the influx directory, recreated a fresh power monitor db, and everything went back to normal. I think influx has an issue with large dbs on the pi.

I think there needs to be some kind of slimming process that thins out the old data - say keeping only 1 min or 5 min interval data for data older than 6 months.

David00, 2021-09-08:
This is most likely related to Docker logs taking up the disk space (it happened to my Pi too after running for about the same amount of time). The database should only be 1-2 gigabytes in size after 9 months.

I've amended the installation instructions to correct for this issue quite some time ago, but it was likely after you had deployed your Pi.

See step 5 in the [Software Installation section](https://github.com/David00/rpi-power-monitor/wiki/Software-0.-Installation) of the Wiki (the manual installation steps - not the custom Pi OS image steps).

As for data retention, InfluxDB supports this and I've been testing some retention policies and continuous queries on my Pi for some time now. They're a little bit involved to setup manually though, and writing documentation or writing a patch to do this for everyone is something I haven't had the time to sit down and do yet.

cgleggett, 2021-09-08:
Ahh - interesting. the containers dir is currently 3.5 Gb.

David00, 2021-09-08:
3.5GB sounds quite large because the actual power monitor data should be mapped to a volume living in `/opt/influxdb/` (assuming you used the `-v` flag in the `docker run` part of the setup instructions)

Let's see how large the Docker log files are on your Pi with the following command:

`sudo ls -lsht /var/lib/docker/containers/*/*.log`

cgleggett, 2021-09-08:
yes, it's indeed the docker log file:
```
> ls  -lsht
total 3.6G
3.6G -rw-r----- 1 root root 3.6G Sep  7 21:52 ad18fb27df1944fd734805a2ea1fab8d42b818df07cbc2d7d6c50270abc8ee50-json.log
4.0K -rw------- 1 root root 2.9K Sep  7 21:38 config.v2.json
4.0K -rw-r--r-- 1 root root 1.6K Sep  7 21:38 hostconfig.json
4.0K -rw-r--r-- 1 root root   13 Sep  7 21:38 hostname
4.0K -rw-r--r-- 1 root root   71 Sep  7 21:38 resolv.conf.hash
4.0K -rw-r--r-- 1 root root  174 Sep  7 21:38 hosts
4.0K -rw-r--r-- 1 root root   49 Sep  7 21:38 resolv.conf
4.0K drwx------ 2 root root 4.0K Nov  2  2020 mounts/
4.0K drwx------ 2 root root 4.0K Nov  2  2020 checkpoints/
```

I suppose the easiest thing to do would be to stop the docker service, wipe the log file, setup log rotation, and restart the service.

David00, 2021-09-08:
Yup! That'll do it.

David00, 2021-09-12:
I'm going to close this now, but feel free to reopen the issue if you see anymore disk space issues!

closed 2021-09-12 by David00

appleguru, 2021-11-03:
I'm having the same issue, though it seems to be compounded by the fact that influx is also no longer running... working on troubleshooting that now

appleguru, 2021-11-03:
Nope, data is still there.. just missing the couple of months where I didn't notice it was broken!

David00, 2021-11-03:
Thanks for looking through the existing issues before opening a new one! 

That's strange that your influx container is not showing up in the list of containers anymore. Are you running my custom Pi OS image, or did you setup everything using the manual instructions? You shouldn't be experiencing this issue with the custom Pi OS image, so I'm only asking to confirm that assumption.

As for data loss, as long as you created the container with the full command listed in the Wiki:

`docker run -d --restart always --name influx -p 8086:8086 -v /opt/influxdb:/var/lib/influxdb influxdb:1.8.3`

..,the influx data should be stored on the Pi in `/opt/influxdb`, and you can recreate the Docker container using the same command listed above, and it should start up like nothing happened.

I just saw your 2nd comment, so that's great it came back up! Bummer that it wasn't recording for the last couple months though, but now that you've resolved the oversized logging issue, you should be good to go!

appleguru, 2021-11-03:
RE your custom Pi OS image vs manual install, how can I tell? I honestly don't remember; I followed whatever instructions were available last ~January.

David00, 2021-11-03:
One way to check is just based on the hostname of the Pi. When you ssh to the Pi, the custom image will say `pi@powermon` whereas the stock Pi OS image will probably say `pi@raspberrypi`. My custom image will also have the directory `docker-compose` inside of `/boot/`, and the stock Pi OS image will not.

appleguru, 2021-11-04:
Looks like a stock pi image. Should I look at migrating to your image? Or not really worth it? (can take this is discussion elsewhere too, just let me know!)

David00, 2021-11-04:
Migrating to my image isn't really worth it unless you're running the GUI version of Pi OS, and the only value in doing a migration in this case would be to free up more resources for the power monitor software.

The image is really only meant to ease the setup process of all the dependencies for users that aren't really familiar with the terminal environment in Linux.