David00/rpi-power-monitor #39: Service occasionally fails: Failed with result 'start-limit-hit'

FreydNot, 2021-10-20:
Once every couple of weeks I notice my power data isn't being updated.  I can ssh into the Rpi successfully but the power-monitor service isn't running.  

Here is the error message

```
pi@powermon:~ $ sudo systemctl status power-monitor
● power-monitor.service - Raspberry Pi Power Monitor
   Loaded: loaded (/etc/systemd/system/power-monitor.service; enabled; vendor preset: enabled)
   Active: failed (Result: start-limit-hit) since Tue 2021-10-19 19:29:18 PDT; 19h ago
  Process: 23982 ExecStart=/usr/bin/python3.7 /home/pi/rpi-power-monitor/power-monitor.py (code=exited, status=0/SUCCESS)
 Main PID: 23982 (code=exited, status=0/SUCCESS)

Oct 19 19:29:18 powermon systemd[1]: power-monitor.service: Service RestartSec=1s expired, scheduling restart.
Oct 19 19:29:18 powermon systemd[1]: power-monitor.service: Scheduled restart job, restart counter is at 6.
Oct 19 19:29:18 powermon systemd[1]: Stopped Raspberry Pi Power Monitor.
Oct 19 19:29:18 powermon systemd[1]: power-monitor.service: Start request repeated too quickly.
Oct 19 19:29:18 powermon systemd[1]: power-monitor.service: Failed with result 'start-limit-hit'.
Oct 19 19:29:18 powermon systemd[1]: Failed to start Raspberry Pi Power Monitor.

```

I issue `sudo systemctl start power-monitor` and it starts up just fine.

Are there any changes we can make to `/etc/systemd/system/power-monitor.service` to avoid this failure?  Maybe increasing the number of retries or increasing the delay between retries?

For reference, here is what my `/etc/systemd/system/power-monitor.service` file looks like:
```
[Unit]
Description=Raspberry Pi Power Monitor
After=docker.service
Requires=docker.service

[Service]
Restart=always
RestartSec=1
StartLimitInterval=120
StartLimitBurst=5
User=pi
ExecStart=/usr/bin/python3.7 /home/pi/rpi-power-monitor/power-monitor.py

[Install]
WantedBy=multi-user.target

```

FreydNot, 2021-10-20:
It seems that `StartLimitInterval` has been replaced by `StartLimitIntervalSec`.  It's probably backwards compatible, so no worries.

From the man page (`man systemd.unit.5`)

>        StartLimitIntervalSec=interval, StartLimitBurst=burst
>            Configure unit start rate limiting. Units which are started more than burst times within an interval time interval are not
>            permitted to start any more. Use StartLimitIntervalSec= to configure the checking interval (defaults to
>            DefaultStartLimitIntervalSec= in manager configuration file, set it to 0 to disable any kind of rate limiting). Use
>            StartLimitBurst= to configure how many starts per interval are allowed (defaults to DefaultStartLimitBurst= in manager
>            configuration file). These configuration options are particularly useful in conjunction with the service setting Restart=
>            (see systemd.service(5)); however, they apply to all kinds of starts (including manual), not just those triggered by the
>            Restart= logic. Note that units which are configured for Restart= and which reach the start limit are not attempted to be
>            restarted anymore; however, they may still be restarted manually at a later point, after the interval has passed. From this
>            point on, the restart logic is activated again. Note that systemctl reset-failed will cause the restart rate counter for a
>            service to be flushed, which is useful if the administrator wants to manually start a unit and the start limit interferes
>            with that. Note that this rate-limiting is enforced after any unit condition checks are executed, and hence unit activations
>            with failing conditions do not count towards this rate limit. This setting does not apply to slice, target, device, and scope
>            units, since they are unit types whose activation may either never fail, or may succeed only a single time.
> 
>            When a unit is unloaded due to the garbage collection logic (see above) its rate limit counters are flushed out too. This
>            means that configuring start rate limiting for a unit that is not referenced continuously has no effect.
 
I'm going to experiment with these settings:
```
RestartSec=10 
StartLimitIntervalSec=300
StartLimitBurst=15

```
I think this will wait for 10 seconds between each delay attempt and stop trying to restart if there are more then 15 attempts inside of 300 seconds (5 minutes).  Seems like a reasonable place to start anyways.

FreydNot, 2021-10-20:
```
sudo journalctl -u power-monitor.service
-- Logs begin at Thu 2019-02-14 02:11:59 PST, end at Wed 2021-10-20 16:28:35 PDT. --
Oct 20 15:58:36 powermon systemd[1]: Started Raspberry Pi Power Monitor.
Oct 20 15:58:55 powermon python3.7[718]: 2021-10-20 15:58:55 : Could not create a backup of config.py file.
Oct 20 15:58:55 powermon python3.7[718]: 2021-10-20 15:58:55 : ... Trying to connect to database at: 10.11.12.74:8086
Oct 20 15:58:55 powermon python3.7[718]: 2021-10-20 15:58:55 : ... DB initalized.
Oct 20 15:58:55 powermon python3.7[718]: 2021-10-20 15:58:55 : ... Starting Raspberry Pi Power Monitor
Oct 20 15:58:55 powermon python3.7[718]: 2021-10-20 15:58:55 : Press Ctrl-c to quit...
```

No dice there, but I did very recently reboot the pi, so that could be why.

I am running a Pi 3 with 1GB of memory.
```
cat /proc/device-tree/model
Raspberry Pi 3 Model B Rev 1.2
```
I only run the powermon service on this pi.  The database and web server run on a different PC.

I seem to have plenty of free memory (at least after a recent reboot)

```
free -h
              total        used        free      shared  buff/cache   available
Mem:          926Mi       136Mi       563Mi        11Mi       225Mi       729Mi
Swap:          99Mi          0B        99Mi
```

```
top - 16:32:22 up 34 min,  1 user,  load average: 1.11, 1.07, 0.95
Tasks: 113 total,   2 running, 111 sleeping,   0 stopped,   0 zombie
%Cpu(s):  9.6 us, 18.5 sy,  0.0 ni, 71.9 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :    926.1 total,    563.4 free,    136.8 used,    225.9 buff/cache
MiB Swap:    100.0 total,    100.0 free,      0.0 used.    729.1 avail Mem
```

sohyt95, 2022-06-14:
I am also facing the same issue and tried to change the intervals as well. 
But it is still not working for me
[image]

David00, 2022-06-14:
Can you share the output of the following commands?

`sudo journalctl -u power-monitor.service -n 50`

`cat /proc/cpuinfo | grep -E "Hardware|Model|Revision"`

jmadden91, 2022-07-17:
I've been experiencing the same issue lately. I'm running the same hardware but with the db and webserver running on the pi. Was rock solid for about a year prior. 

> Jul 17 11:14:51 powermon systemd[1]: power-monitor.service: Service RestartSec=1s expired, scheduling restart.
Jul 17 11:14:51 powermon systemd[1]: power-monitor.service: Scheduled restart job, restart counter is at 2.
Jul 17 11:14:51 powermon systemd[1]: Stopped Raspberry Pi Power Monitor.
Jul 17 11:14:51 powermon systemd[1]: Started Raspberry Pi Power Monitor.
Jul 17 11:14:53 powermon python3.7[1322]: 2022-07-17 11:14:53 : Could not create a backup of config.py file.
Jul 17 11:14:53 powermon python3.7[1322]: 2022-07-17 11:14:53 : ... Trying to connect to database at: localhost:8086
Jul 17 11:14:59 powermon systemd[1]: power-monitor.service: Succeeded.
Jul 17 11:15:00 powermon systemd[1]: power-monitor.service: Service RestartSec=1s expired, scheduling restart.
Jul 17 11:15:00 powermon systemd[1]: power-monitor.service: Scheduled restart job, restart counter is at 3.
Jul 17 11:15:00 powermon systemd[1]: Stopped Raspberry Pi Power Monitor.
Jul 17 11:15:00 powermon systemd[1]: Started Raspberry Pi Power Monitor.
Jul 17 11:15:02 powermon python3.7[1323]: 2022-07-17 11:15:02 : Could not create a backup of config.py file.
Jul 17 11:15:02 powermon python3.7[1323]: 2022-07-17 11:15:02 : ... Trying to connect to database at: localhost:8086
Jul 17 11:15:08 powermon systemd[1]: power-monitor.service: Succeeded.
Jul 17 11:15:09 powermon systemd[1]: power-monitor.service: Service RestartSec=1s expired, scheduling restart.
Jul 17 11:15:09 powermon systemd[1]: power-monitor.service: Scheduled restart job, restart counter is at 4.
Jul 17 11:15:09 powermon systemd[1]: Stopped Raspberry Pi Power Monitor.
Jul 17 11:15:09 powermon systemd[1]: Started Raspberry Pi Power Monitor.
Jul 17 11:15:11 powermon python3.7[1324]: 2022-07-17 11:15:11 : Could not create a backup of config.py file.
Jul 17 11:15:11 powermon python3.7[1324]: 2022-07-17 11:15:11 : ... Trying to connect to database at: localhost:8086
Jul 17 11:15:17 powermon systemd[1]: power-monitor.service: Succeeded.
Jul 17 11:15:18 powermon systemd[1]: power-monitor.service: Service RestartSec=1s expired, scheduling restart.
Jul 17 11:15:18 powermon systemd[1]: power-monitor.service: Scheduled restart job, restart counter is at 5.
Jul 17 11:15:18 powermon systemd[1]: Stopped Raspberry Pi Power Monitor.
Jul 17 11:15:18 powermon systemd[1]: power-monitor.service: Start request repeated too quickly.
Jul 17 11:15:18 powermon systemd[1]: power-monitor.service: Failed with result 'start-limit-hit'.
Jul 17 11:15:18 powermon systemd[1]: Failed to start Raspberry Pi Power Monitor

nightcustard, 2022-07-18:
I started experiencing the same behaviour and error messages a month or so ago.  I'm using David's custom Pi OS image on a Pi 4.  On investigating the problem I found the load average (uptime) was typically over 6 and I was getting huge numbers of errors in the influxdb logs.  My database was about nine months old at this point.  To cut a long story short, I created a three month retention policy for the raw data and a never expire policy for the elaborated stuff (mostly costs).  Once the database was thinned, everything went back to normal (load average circa 1.2).  I'm not suggesting this is a solution, probably more of a sticking plaster but a large influxdb database has caused me problems before, hence deciding to bite the bullet and implement a modest retention policy.   I'd be interested to hear if you're seeing influxdb errors and a higher than normal average load.

David00, 2024-02-04:
I don't think this has been observed since the v0.3.0 image release, so I am going to close this issue. One attempt to reduce the load imposed by the database is that samples are now batched in memory and written to InfluxDB approximately every 30 seconds (as opposed to writing each and every point to the database)

closed 2024-02-04 by David00