David00/rpi-power-monitor #40: Log tariff alongside current data

nightcustard, 2021-11-24:
My electricity supply is dual rate (ie) I'm charged a day rate and a night rate. I've been looking into how to (fairly) accurately calculate how much a particular piece of equipment is costing me.   I need to apply the correct tariffs, which won't necessarily be the same as the current ones if I'm looking at historical data.   My other variable is the times of day when the tariffs cut in and out.
My researches so far suggest that the best way to do this would be to record the tariff data alongside the current measurements, making grafana display quite simple (which is a must as far as I'm concerned(!)).  Is this something that could be realised in the code or would I be better doing something in Grafana?

fruffygutss, 2021-11-28:
ooo yes I have the same question! following

David00, 2021-11-29:
Since the rates are going to inevitably change over time, you'll want to store the actual cost of the energy used in the database, like you were suggesting.

It's possible to do this with some modification to the Python code, OR, via InfluxDB through the use of [continuous queries](https://docs.influxdata.com/influxdb/v1.8/query_language/continuous_queries/). 

I'd suggest going down the InfluxDB CQ route just because the Python side of things doesn't actually have direct access to the _energy_ figures (without querying). (By energy, I mean the figure that represents `power * time`, denoted in kWh, which is what we typically pay for).

An influxDB CQ automatically runs every _X duration_ (whatever you set it to), and stores the output value from the period specified in a new table.  The continuous query would have to calculate the energy over the CQ interval, then apply the cost for that energy based on the time of the measurements, then store it to a dedicated measurement for the costs. The only thing I'm unsure of at this point is how to determine which rate applies to the energy value, since you can't select by time in the CQ.

nightcustard, 2021-11-29:
Thanks David, I'll investigate this.  My brief look so far confirms you can't select by time but maybe I can run two queries with the two rates and then select the outputs by time downstream.  That said, at this stage I don't even know how to run an InfluxDB query. It's a bit like attempting to write some doggerel in Welsh with only a Welsh phrasebook to help!  As the old phrase goes, it's a learning opportunity... cheers, Mike

David00, 2021-11-29:
I was hoping in the CQ you could maybe do two subqueries that return `tarrif1 * total kWh during tarrif1's time interval` and another one for `tariff2 * total kWh during tariff2's time interval`, but I'm not too sure about subqueries in a CQ.

To get an InfluxDB shell so you can run queries, you'd have to execute the following docker command on your Pi:

    docker exec -it influx influx

Definitely a learning opportunity! If you want to give it a go in Python and need some guidance on how the data makes its way into the DB, I can provide an overview for that too.  But, for the time being, the DB info in the Wiki (which provides an overview of the schema and even some sample queries) might be helpful:
https://github.com/David00/rpi-power-monitor/wiki/Software-5.-All-About-the-Database

nightcustard, 2021-11-29:
Many thanks for the pointers.  I'll certainly give it a go 😀

nightcustard, 2021-12-19:
Hi David - I've created a couple of CQs which are generating the cost data and it looks as if 'Flux' may be the way to go to select the right data over the time intervals I need. However, I got stuck trying to change the default influxdb config, which has flux-enabled = false.  I've created an influxdb.conf file in /etc/influxdb with appropriate permissions but I think I need to tell influxdb to use it on start up (which I haven't yet figured out how to do).  I'd be grateful for any suggestions!

David00, 2021-12-20:
Hi @nightcustard, that sounds interesting that Flux might be the way to go. I haven't yet looked into Flux but I know that's the direction that InfluxDB is moving towards. 

Since InfluxDB runs inside a Docker container, all of the config files are inside of the container itself. You'll need to get a shell inside the container, and then make the changes to the `/etc/influxdb/influxdb.conf` file inside the container. To do this, run the following command, which executes `bash` in interactive mode in the container named `influx`:

```
docker exec -it influx bash
```
Now that you're inside the container environment, you can navigate to /etc/influxdb and edit the influxdb.conf file there.  I think the only file editor available in the container is `vi`, which has a little bit of a learning curve, so definitely check out some "how to's" on YouTube if you're not familiar with it.

nightcustard, 2021-12-21:
Thanks David.  I'll have to pause on this one as I've come across an unrelated problem - I'm seeing lots of errors in my influx log and my Pi 4 is responding slowly (possibly after a power cut a couple of days ago). Normal service is resumed with another SD card and backup image.  I think I remember seeing something in the wiki about backing up and restoring influx databases but can't find it. Am I imagining this? If so, I'll raise another query :-)

nightcustard, 2022-01-06:
I'm looking into this issue again and am trying to decide which way to go - python or CQ.  What's tipping me back towards a python solution is that my power distribution system has a power output which only comes live when the cheap rate is active.  I can very easily hang a low voltage (circa 3v dc) wall wart on that supply and connect the output to an unused GPIO input on the Pi, thus giving me a '1' or '0' depending upon the active tariff.  It looks easy to add some code to the rpi-power-monitor programme to detect what state the GPIO input is in and it could get the relevant tariff from a tweaked config.py.   Is storing the tariff in the influx db any more complicated than adding a line to the  'infl.write_to_influx(' section?  I'm assuming there's more to it than that (it always is!) but I'll ferret away and see how I get on.  I might still revert back to CQs as any solution along those lines can be applied retrospectively, of course.

nightcustard, 2022-01-15:
I'm glad to be able to report success!  

Hardware-wise, I've got a small wall-wart power supply powered from the power breaker which is only active when the night rate is in force.  This is connected to GPIO17 via a simple buffer using a 2N7000 FET and a couple of resistors.

Software-wise, I've forked your code* and added appropriate GPIO input detection as well as the necessary additions to add the tariff data into the appropriate measurements.  This should all be visible in the 'Tariff_additions' branch in my repository ([https://github.com/nightcustard/rpi-power-monitor/tree/Tariff_additions]).

I've put the day & night tariff rates in the config.py file and they're picked up by power-monitor, which selects the appropriate rate according to the state of GPIO17.  I want to check the cost of the equipment monitored by a particular ct as well as the cost of the whole load. My reading of the influxdb query language is that you cannot reference data from more than one measurement in a query, so I've added a 'tariff' field key to both the 'home_load' and 'raw_cts' measurements, which is a little inefficient I suppose but it saves me having to learn how to use 'Flux'!

I then run a continuous query per ct and one for the 'home_load' total (eg)
`ct0_cost       CREATE CONTINUOUS QUERY ct0_cost ON power_monitor BEGIN SELECT last(tariff) * integral(power) / 3600000 INTO power_monitor.autogen.ct0_cost FROM power_monitor.autogen.raw_cts WHERE ct = '0' GROUP BY time(10m) END'

'home_load_cost CREATE CONTINUOUS QUERY home_load_cost ON power_monitor BEGIN SELECT last(tariff) * integral(power) / 3600000 INTO power_monitor.autogen.home_load_cost FROM power_monitor.autogen.home_load GROUP BY time(10m) END`

I can then display the results in Grafana :-))

Many thanks again for pointing out the way.  BTW, I've also transferred over to an 'endurance' type SD card as you suggested and I'm also running the Pi from a mains UPS, so I've added 'nut-client' to achieve an orderly shutdown before the UPS cruds out.

* I'm rather unsure of the protocol involved in forking other people's code, so if I've breached said protocol it was of course unintentional and please let me know!

closed 2022-01-15 by nightcustard