David00/rpi-power-monitor #41: Influxdb error log: 'cannot allocate memeory' & 'fatal error: out of memory'

nightcustard, 2021-12-26:
This isn't strictly an RPi-power-monitor issue, so I hope you'll forgive me for raising this issue here in the hope you'll tell me it's a me-only problem (ie) there isn't a general InfluxDB memory allocation bug or something equally intractable at work here!

Whilst trouble-shooting influxdb backup and restore, I noticed the docker influxdb log is recording quite a large number of 'cannot allocate memory' errors and a few 'fatal error: out of memory' events.
 
I'm running a new build (following the Wiki instructions for a manual install) with a couple of minor changes - I've installed 32 bit Bullseye and I've had to tweak one of the startup commands to reference Python 3.9 (I can't remember which one at the moment). 
My computing hardware is a Pi 4 with 2GB RAM

I get at the logged errors using "docker logs influx 2>&1 | grep -i 'fatal\|cannot allocate memory' " 

The docker log is 8MB (the container has only been running a couple of days or so).  I've attached it plus an extract to illustrate the sort of error messages I'm getting.

The errors only started appearing some hours after the initial start up.  The reason I started off with a new build was that I noticed similar errors in the logs on my long running power monitor build and assumed it had become corrupted by a recent power outage.   Unfortunately, this now looks like it wasn't the case.

A quick look on the web threw up this little gem:
[https://community.influxdata.com/t/fatal-error-out-of-memory/16103](url)
which made me wonder whether what I'm seeing may be inherent with only having 2GB RAM to play with.  The proposed solution changes indexing to disk from RAM and this doesn't sound good in terms of SD wear.
Any thoughts?

Regards, 
Mike

[Influxdb errors.txt](https://github.com/David00/rpi-power-monitor/files/7777066/Influxdb.errors.txt)
[influxdb.log](https://github.com/David00/rpi-power-monitor/files/7777071/influxdb.log)

David00, 2021-12-27:
That's an interesting find regarding RAM vs disk indexing. Do you have a dashboard that is looking at periods of time longer than 1 week?  These can be very intensive because the raw data is quite high resolution.  (I have some changes coming soon that will bring continuous queries and retention policies to the project, which essentially automatically down sample data for you so that you can run queries on data that spans weeks, months, and even years.)

I understand you saw the cannot allocate memory message in the Influx container's logs, but do you know what process was eating up all the memory? 2GB is plenty (for the server/CLI only image).

That's good to hear that my custom image seems to have resolved the problems. There are a lot of variables at play especially when switching to a new major OS release.

nightcustard, 2021-12-30:
Thanks David

I hadn't twigged long timeframe Dashboard queries would hit the processor
(duh!).  However, I'm not sure in this case this is the cause as most of my
dashboard comprises time histories and only one panel is calculating energy
usage over a 24hr period.

When I started looking into CQs, I noticed the Pi's ssh response times had
slowed right down, sometimes taking several seconds for a simple 'ls' type
command.  The software build at this time was the manually created build
I'd put together using your instructions and had been in operation since I
built the interface. As the build had been running such a long time, I'd
assumed the SD had possibly become corrupted and reverted to an image I'd
made of the original build a couple of months previously.  This started off
fast, but ssh response times fell off again and a look at the logs showed
the errors had started again.  At this point I made a fresh build but made
the rookie error of using an updated version of the OS (ie) Bullseye.  This
again started off with no errors but very rapidly started accumulating them
after a short period.  Dashboard queries were typically at this stage
limited to 'last 30s'.

I then did what I should have done and used your OS image, and the Pi has
been running error free for over three days now.
That said, I didn't establish what had been eating up the memory - that's a
task for later, once I've managed to recover the months of data on the
original SD card and sideloaded it into the (new) database.  I haven't
succeded yet - I got sidetracked by these logged errors.  Now that appears
resolved, I'll get on with troubleshooting that!

Anyway, thanks again for the great support and I'm looking forward to the
CQ and retention policy update when it appears!
Mike

closed 2021-12-31 by nightcustard