David00/rpi-power-monitor #42: Backup & Restore

nightcustard, 2021-12-31:
Hi David

I'm having a few problems with backing up and restoring the power_monitor database and I wonder whether you have any insights which may help?

After a fair amount of browsing I attempted the following backup and restore process:

I created an internal docker directory '/backups':

$ docker exec -it influx bash
# mkdir /backups
# exit

Create backup:
$ docker ps (gets influxdb container ID)
$ docker exec -it <container ID> influxd backup -portable /backups

Copy the backed up databases to a network share mapped in fstab:
$ docker cp <container ID>:/backups /home/pi/influxdata/backup<backup date>

Restore the data:
Copy the backup to the /backups container folder:
$ docker cp /home/pi/influxdata/backup<backup date> <container ID>:/backups

Start a bash session in the influx docker container and then restore the backed up database into a new database called pm_bak.
$ docker exec -it influx bash 
# influxd restore -portable -db "power_monitor" -newdb "pm_bak" /backups/backup<backupdate>
# exit

Sideload into the power_monitor database and drop the temporary pm_bak database:

Open up an influxdb shell:
$ docker exec -it influx influx

> USE pm_bak
> SELECT * INTO power_monitor..:MEASUREMENT FROM /.*/ GROUP BY *
> DROP DATABASE pm_bak
> exit

I think this should work as it's almost verbatim from instructions on the Influxdb website.  However, the sideload command ('Select *' etc.) results in errors in the influx docker logs and terminates with a 'ERR: no data received'.

$ docker logs influx 2>&1 | grep -i 'error'
ts=2021-12-31T15:06:11.114892Z lvl=warn msg="Error while freeing cold shard resources" log_id=0YlFfLY0000 service=store error="shard is disabled" db_shard_id=36
ts=2021-12-31T15:06:17.807313Z lvl=info msg="Cannot retrieve measurement cardinality" log_id=0YlFfLY0000 service=store db_instance=pm_bak error="engine is closed"
ts=2021-12-31T15:06:19.276092Z lvl=warn msg="Error while freeing cold shard resources" log_id=0YlFfLY0000 service=store error="shard is disabled" db_shard_id=25
plus several more lines, ending with
fatal error: out of memory

It's possible I have a corrupt database but I've also tried the same process with a 'clean' database and I get the same types of errors, so I'm pretty sure I'm doing something stupid. Permissions, possibly?

How do you do your backups?

Help!

nightcustard, 2022-01-02:
Further to the above, I've found by deleting some of the more recent shards, the 'out of memory' errors cease, and I can successfully restore individual shards.  I'm working my way through the shards from oldest to newest but it looks at the moment as if the problems detailed above are probably due to several corrupt shards.

David00, 2022-01-04:
Hi Mike, your backup process looks right to me. On my Pi, I have created a local folder on the pi in  `/opt/influxdb` and mapped it to `/var/lib/influxdb` inside the InfluxDB container.  Then, when I do a backup, I can do it in one line with:

`docker exec -it influx influxd backup -portable /var/lib/influxdb/backups/`

(where `influx` is the name of my influxdb container).

I have not experienced the out of memory issues that you've been having, but that it interesting to hear it may be shard related.

nightcustard, 2022-01-06:
That's a useful idea!  
I can confirm I've got some corrupt shards - there's a couple which, if restored singly, cause the docker influx container to crash several seconds after the restore aborts and it's then a race to restart the container and delete the offending shard before the container is shut down again.  As a result, I've 'archived' the older shards and my current build contains shards from the beginning of November.  I'm now looking into small UPS as I think power outages are the most likely culprit.
Well, if nothing else, your power monitor project has caused me to get familiar with docker (which I'd heard of but never used) and influxdb (which I'd not heard of).  Of course, I'm the owner of a very useful power monitoring system :-)

closed 2022-01-06 by nightcustard

David00, 2022-01-06:
I'm glad to hear you got to the bottom of it and that my project caused you to learn Docker and Influx! You're probably right about power outages being the culprit. I recommend getting one of the endurance microSD cards while you're still thinking of rebuilding your setup... my standard Samsung Evo card failed after just over a year and I've been hearing of others cards doing the same in about the same time.

nightcustard, 2022-01-08:
Thanks David - I've bought and installed an 'endurance' SD card.  I didn't even know they existed!

FreydNot, 2022-01-28:
This procedure would be a great addition to the documentation here:
https://github.com/David00/rpi-power-monitor/wiki/Software-5.-All-About-the-Database#database-backups