David00/rpi-power-monitor #43: Default voltage value

400bird, 2022-01-09:
Yesterday, I lost power to my AC-AC adapter for a few hours. During that time the current sensors and Pi were still churning along gathering data. (The Pi is powered by a POE hat and the GFCI breaker on the AC-AC tripped)

Is it possible for the voltage value to default to 120 if the measured voltage is low (<20)?

David00, 2022-01-09:
Yes, you can do this with a small code modification.  In [`power-monitor.py`, line 458](https://github.com/David00/rpi-power-monitor/blob/master/power-monitor.py#L458), the voltage reading is extracted from the overall results for the current polling cycle. Immediately after this, you can check the value and set it to 120 if it's less than 20, like this:
```
            # solar_power = results['ct3']['power']
            # solar_current = results['ct3']['current']
            # solar_pf = results['ct3']['pf']
            voltage = results['voltage']
            if voltage < 20:        # <--- Add these lines
                voltage = 120       # <--- Add these lines
```

Make sure the indentation is consistent when you edit the file, otherwise you'll get a syntax error and it won't start.  After saving the change, restart the power monitor with `sudo systemctl restart power-monitor`, and then verify that it's running with `sudo systemctl status power-monitor`.

400bird, 2022-01-09:
# If you are monitoring solar/generator inputs to your panel, specify which CT number(s) you are using, and uncomment the commented lines.
            solar_power = 0
            solar_current = 0
            solar_pf = 0
            solar_power = results['ct0']['power']
            solar_current = results['ct0']['current']
            solar_pf = results['ct0']['pf']
            voltage = results['voltage']
            if voltage < 20 :             #added  to default voltage when AC AC conv inop
                voltage = 120             #default voltage value

            # Set solar power and current to zero if the solar power is under 20W.
            if solar_power < 20:
                solar_power = 0
                solar_current = 0
                solar_pf = 0

> 

I was thinking/hoping the code would be that simple.
But, it didn't work for me. I still get 0 volts and 0 watts if I unplug the 9vac plug.

[image]

David00, 2022-01-09:
The values are going to the database, so you'd see them in Grafana if you're plotting the voltage.

In terminal mode, you'd have to change the value gets printed. See line https://github.com/David00/rpi-power-monitor/blob/master/power-monitor.py#L566 - its pulling the actual value from the results dictionary. To get the hard-coded value to print in terminal mode, you'll need to overwrite the value in `results['voltage']` where you made the same edit earlier, like this:

`results['voltage'] = 120`

400bird, 2022-01-10:
Ok, I hadn't thought about where terminal mode pulled voltage. I'll continue to use Grafana.

I was originally using Grafana go monitor, and it's doing the same thing, wattage goes to 0 when the adaptor is unplugged.

I've got a work trip this week and will be out of town and will need to resume testing this later.

I'd have to double check, but I wonder if the watt calculations were done further up in the code, then we're changing voltage after watts are calculated.

David00, 2022-01-10:
Yes, the watt calculations are done further up in a different function and they're using the voltage as measured.  I'm on mobile and away from home so I will share the edits when able.

closed 2023-01-10 by David00