David00/rpi-power-monitor #44: Is this project also avaible without the 9V AC transformer?

YoManGaming, 2022-01-19:
hello there, i live in the Netherlands and i want to use this project to watch my home power usage. Is it also possible to use this project without the 9V AC transformer (because it cant be shipped to here)?

David00, 2022-01-19:
Hello! Yes, you can use it without the 9V AC transformer, but you'll only be able to measure current from the CTs.  

To fully utilize the features of the project, such as real power (Watts), energy (kWh), and power factor,  you'll need to acquire a 9V AC transformer that works with your grid.  The connector is a 2.1x5.5mm barrel jack. Feel free to send a link to a 9V AC transformer and I can confirm that it will work with the project or not.

YoManGaming, 2022-01-20:
Hey David, will this one work? https://www.amazon.com/inShareplus-Voltage-Transformer-100-240V-Connector/dp/B01GD4ZOI4/ref=mp_s_a_1_2_sspa?keywords=9v+ac+transformer&qid=1642661311&sr=8-2-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExS0JPMjFMMldYQkdFJmVuY3J5cHRlZElkPUEwMzYyNzQ3MThYTThVNUlaUzJWTiZlbmNyeXB0ZWRBZElkPUEwMDI5MjczMzY3TVlIUjIyTUdHMyZ3aWRnZXROYW1lPXNwX3Bob25lX3NlYXJjaF9hdGYmYWN0aW9uPWNsaWNrUmVkaXJlY3QmZG9Ob3RMb2dDbGljaz10cnVl

David00, 2022-01-21:
No, that one won't work. It puts out 9V DC. It needs to put out 9V AC.

YoManGaming, 2022-01-23:
Does this work? https://www.amazon.com/Reliapro-ADU090150A2231-Adapter-Transformer-Straight/dp/B00B886CWS/ref=mp_s_a_1_3?keywords=9v+ac+transformer&qid=1642962819&sr=8-3

David00, 2022-01-23:
Yes - that's the exact same one I have available in my shop:
https://power-monitor.dalbrecht.tech/product/9v-ac-transformer-north-america-only/

Will it fit your outlets in the Netherlands? I thought they have the standard EU 2-prong 230V outlets over there.

YoManGaming, 2022-01-24:
I didn't think about that one😅. Yes, we have the standard 230V here. Then ill use the project only with the CT's. Can i just set the get_board_voltage() to the voltage im using?

David00, 2022-01-24:
> Can i just set the get_board_voltage() to the voltage im using?

If you did that, you'd only be able to get apparent power in Volt-Amperes (VA), which isn't really in line with what most utility companies charge you for. They charge you for the amount of real power you consume, which takes into account the phase relationship between the current and voltage waves in the electrical system.  

You'd need one of these:
https://www.globalsources.com/AC-AC-adapter/AC-AC-adapters-1088369069p.htm

Perhaps you can find one within the Netherlands or within a European market? It is difficult for me to search for them as the results are mostly ads that are relevant to the North American 120V grid.

1v052, 2022-01-24:
> If you did that, you'd only be able to get apparent power in Volt-Amperes (VA), which isn't really in line with what most utility companies charge you for. They charge you for the amount of real [...]

You would only need to order 500 pieces ;)

Jokes aside, you should try to look on site slike eBay or Conrad electronics.
I found one on Ebay Germany which you could order, unfortunately the offer only runs for another 2hrs.
If it doesnt sell you could simply contact the seller and offer to buy it.

https://www.ebay.de/itm/313837031207?hash=item4912250f27:g:-r0AAOSwUvldEk-O

YoManGaming, 2022-01-24:
> If you did that, you'd only be able to get apparent power in Volt-Amperes (VA), which isn't really in line with what most utility companies charge you for. They charge you for the amount of real [...]

The problem is that i don't want to use this to monitor my home usage (like i said earlier). I didn't say this because i already knew how to use the code on the boat. It is needed for a project on a solar boat. The problem is then that i need to run it on batteries. I can run the raspberry pi on power banks, but measuring voltage is an other problem. You got a solution for that?

David00, 2022-01-25:
> hello there, i live in the Netherlands and i want to use this project to watch my home power usage. 

> It is needed for a project on a solar boat.

Which is it?

It sounds like you need a DC monitoring system for your boat. This project is for AC power systems only and is intended for home use, not marine use (though if your boat has a DC to AC inverter to supply 120V or 230V AC, you can use it to measure the AC output of the inverter).

> You got a solution for that? edit: can i send everything (the PCB and the CT's) back if my project doesn't work?

Unfortunately, no.

Closing the issue as this has gone off topic but feel free to reopen if needed.

closed 2022-01-25 by David00