David00/rpi-power-monitor #45: Phase calibration does not work even with the correct kernel version

chenhoulin6204, 2022-02-09:
So, I've followed the setup instructions, the only difference is I remade the circuit on a breadboard and used some bigger capacitors.
The current transformer I used is YHDC SCT006, the burden resistor is 22 Ohms.

The issue is, when I am doing the phase calibration, even with the correct kernel version (4.19.118-v7l+), the results are still larger than 1.x. I have attached the results from your program's debug mode and phase calibration mode, please help!
(The sampling rate reported by the debug mode is around 28 KSPS, is that normal? I don't know... I am using Raspberry Pi 4B with 4GB ram)
[phase_calibration_problem.zip](https://github.com/David00/rpi-power-monitor/files/8029266/phase_calibration_problem.zip)

The load I used was an electric heater and its power factor is 1.00, confirmed by a store-bought power meter.

I have also attached photos of my setup, I am clueless what to do next... Any help is much appreciated, thanks a lot.
[image]
[image]

chenhoulin6204, 2022-02-09:
Hi,

1. The R0 is here:

I am only using one CT, so R1-R3, R8-R9 are not connected. R0 is at 22Ohms. The LMV321 is outputting 1.65V but my problem was I connected one of the CT’s feet to GND, I have now correctly connected it to the Vbias.
2. I’ve used a load resistor to make the AC input now 10V, but it is still not working well. R4 is 9.9k and R5 is 99.8k Ohms as measured so I think the values are correct, but when I measure the AC voltage between R4 R5 and GND, I got 0V instead of the expected 1V, I suspect either the ADC is drawing too much current or something wrong with the GND_break? I am currently using a 4.7µF electrolytic capacitor to ground it to GND.

The problem now has changed, the phase calibration does not give me 2.x anymore but the results are inconsistent, the value fluctuates. Also in terminal mode, the unconnected CTs are showing weird results, although I am able to complete the accuracy calibration now, something still seems wrong. I’ve attached the updated initial test and phase calibration results, could you take a look please? Thanks a lot for your time!

From: David00
Sent: 09 February 2022 06:19
To: David00/rpi-power-monitor
Cc: chenhoulin6204; Mention
Subject: Re: [David00/rpi-power-monitor] Phase calibration does not work evenwith the correct kernel version (Issue #45)

Hi @chenhoulin6204, thank you for providing the debug plot while opening this issue. Based on the "Initial Test" debug plot, it appears that there are a couple of problems with your breadboard circuit.
1. The board is only capturing the positive half of the A/C sinewave. The LMV321 chip should be putting out 1.65V on pin 4, and I suspect this is not happening in your setup (or LMV321's pin 4 output is not tied to each CT input properly). Actually, I just noticed right before submitting this comment - where are your resistors R0 - R3, R8 - R9? The CT inputs need to be bridged together through a resistor that's connected to the Bias-V rail so that the "negative" halves of the AC wave can be captured.
2. The AC input voltage is clipped at both the top and bottom extremes (the ADC range is from 0 - 1023 and you can see the top and bottoms of the voltage wave are being clipped at both bounds). This is either caused by using too large of an input AC voltage for this circuit, or, the values for R4 and R5 are wrong. The AC input should be between 9-12V AC as measured by your multimeter, not as printed on the label. Most transformers will put out a higher voltage without any load on them, and the PCB is essentially no load at all.
Breadboards are tricky, but it looks like you did some really clean work!

David00, 2022-02-10:
Can you add the attachments on GitHub directly? They don't come through via email.

> The problem now has changed, the phase calibration does not give me 2.x anymore but the results are inconsistent, the value fluctuates. Also in terminal mode, the unconnected CTs are showing weird results, although I am able to complete the accuracy calibration now, something still seems wrong.

Don't worry about the values until the circuitry is fixed.  You'll still be very susceptible to noise on the breadboard though, and you can already see that noise in the original debug plot on the other CT channels that you don't have connected to anything.  Not much you can do about that on a breadboard.

chenhoulin6204, 2022-02-18:
Hi, sorry for the delay. I can complete the accuracy calibration now, but I still think the results are dodgy, since the phase calibration value is inconsistent.

The debug test result and phase calibration result are attached, what could be the problem please? Is it because of the noise? Could using shorter cables or better cables solve this problem? Or is it the capacitors I used are of poor quality? I have no idea...

Please educate me, thanks a lot
[new_results.zip](https://github.com/David00/rpi-power-monitor/files/8094113/new_results.zip)
!

David00, 2022-02-18:
The new `initial_test.html` looks pretty good now actually.  The only thing I'd recommend is trying with a higher powered resisitve load so that the current samples coming in are more in the middle of the range of the sensor.

If you don't have a larger resistive load, you can use the loop trick to multiple the amount of current that the sensor sees:
https://github.com/David00/rpi-power-monitor/wiki/Phase-Correction-Procedure#current-multiplication-example

closed 2022-03-19 by chenhoulin6204