David00/rpi-power-monitor #52: How to correct measurements for 240v circuits, i.e. air conditioner, range, dryer, etc

pablomarquesmendez, 2022-06-08:
David

first of all thank you for this project.

How should I go about If I want to measure 120v and 240v circuits with the same pi hat?

Are there specific settings on the config.py if, for example, I am measuring a single 15Amps breaker on CT-0 and a 60Amps 240v circuit on CT-1.    

Regards

Pablo

David00, 2022-06-08:
Hi Pablo, there are a couple of ways you can do this for 240V circuits, depending on the appliance and your tolerance for accuracy.

1) You can double the measurement from a single sensor to get an reasonable approximation for a 240V circuit, or
2) You can use another current sensor to measure the other phase of the 240V circuit, and then sum the values together before sending them to the database.

Option 1 is easier and is fully supported. Option 2 requires some code modification to sum the measurements before adding to the database.

For option 1, see this part in the wiki:
https://github.com/David00/rpi-power-monitor/wiki/Solar-Integration#doubling-a-cts-measurement-for-240v-circuits

For option 2, the general idea is to add the values from the two sensors together, similar to how I've done it here in [this line](https://github.com/David00/rpi-power-monitor/blob/master/power-monitor.py#L476), and then create a new measurement in the database.  Getting a new measurement written to the database ([this function](https://github.com/David00/rpi-power-monitor/blob/04c88ec8a1b08afdbbc6567e98f0f2f100bc2749/influx_interface.py#L168)) currently isn't very easy or user friendly, and this is something I've been thinking of adding lately. So, if you want to try option 2, I can make some changes to make it easier to do this.

pablomarquesmendez, 2022-06-08:
David

Thank you for the quick reply.

I tried option 1 and it is not giving me good results.
I edited power-monitor.py and modified it following the instructions:
    results = {
        'ct1' : {
            'type'      : 'consumption',
            'power'     : real_power_1 * 2,
            'current'   : rms_current_ct1 * 2,
            'voltage'   : rms_voltage_1,
            'pf'        : power_factor_1
        },

Even after doing the phase correction step and entering the ct_phase_correction in config.py, it is reading huge amp numbers.
With the stove running (pure resistive load), the program reports 45Amps. I measured it with my multimeter and it is really taking about 10.7 Amps.

What could be the issue here?

Thank you

David00, 2022-06-09:
The calibration process takes two steps - one is the phase calibration, and the other is accuracy calibration.  The phase calibration corrects the timing between samples, and the accuracy calibration corrects the amplitude of the readings to match the sensor value you have.  Did you do [step 3](https://github.com/David00/rpi-power-monitor/wiki/Calibrating-for-Accuracy#step-3-final-accuracy-calibration) of the calibration process?

If not, you'll have to remove the `* 2` from the power and current results, and then do the accuracy calibration for the channel in question.

I came to this conclusion because the [software assumes](https://github.com/David00/rpi-power-monitor/blob/master/config.py#L51) you're using a 100A sensor, and since I believe you ordered all 60A sensors, your actual readings should be around 60% of what you're seeing now.  And 60% of 45A is 27A, and if that was the doubled measurement from CT1, then we can half it to get pretty close to your handheld meter (27A / 2 = 13.5A). But, this is just ballpark based on an assumption (and the uncalibrated CT1 is being doubled, therefore increasing the error) so the accuracy calibration step will really bring the reading closer.

closed 2022-09-28 by David00

JeromeThiv, 2022-11-24:
Hi David, 
I followed option1 for my 240v circuit of the water heater. 
[image]

I tought that the result would simply be a multiplication by 2 but the 1.6kw consumption became a 2.15kw :

[image]

When I zoom, i see that the measure are oscillating between 1.6 and 3.2, with an average of 2.15 :  
[image]

Did i miss something ?

David00, 2022-11-25:
Hi Jerome, if you're on the latest version of the software (v0.2.0), you'll have to do a `pip install .` to get the values in config.py to reflect in the measurements. So, try:

`cd ~/rpi_power_monitor && pip install .`

Then, restart the software and you should see the changes take effect.

JeromeThiv, 2022-11-26:
oops yes thank you. I was only restarting the systemcl process.