David00/rpi-power-monitor #53: Start4.elf is not compatible

sohyt95, 2022-06-13:
[image]

Please do help me with this issue as per the photo has shown, i have tried multiple methods with the newest imager and even with the imager that was downloaded from the guide's link. 
I am still having the same issue over and over again.

David00, 2022-06-13:
Hi @sohyt95, this is a known issue with the latest bootloaders that are shipping on the newer Pi's.  

Please try with the Pi OS Lite image here:
https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-04-07/2022-04-04-raspios-bullseye-armhf-lite.img.xz

... and follow the manual setup instructions here:
https://github.com/David00/rpi-power-monitor/wiki/Software-0.-Installation#manual-installation-steps

The main problem is the newer bootloader (and kernel) that's shipping with the newer Raspberry Pis.  My custom Pi OS image is not compatible with this newer bootloader. This is a kernel issue with the latest Raspberry Pi units and we have not found a solution to this issue yet:
https://github.com/raspberrypi/linux/issues/3381

Installing the software via the manual installation instructions will work, but you'll have half the sampling rate due to the issue mentioned above.  I am investigating a way to downgrade the bootloader or fix the SPI issue, but both potential solutions are rather complicated.

sohyt95, 2022-06-13:
thank you for the promptly reply, after I follow as per the instruction I am back to the same issue again

David00, 2022-06-13:
Whoops, I should have mentioned that you should skip step 14 of the manual install instructions (the `sudo rpi-update` command).  
This command installs an older kernel that triggers the message in your screenshot.

David00, 2022-11-13:
This is resolved in the version 0.2.0 release.

https://github.com/David00/rpi-power-monitor/releases/tag/v0.2.0

closed 2022-11-13 by David00