David00/rpi-power-monitor #54: Non Linear calibration?

vulture2600, 2022-08-02:
Hey David, 
First of all, this is an awesome project and my HAT is off to you! (pun very...VERY much intended).

I do have a question about final calibration. If noise induced on the CT shows a value above zero when not even clamped around a wire that has current flowing, wouldn't that imply a non-linear scaling factor? Maybe in the future you could make it possible to graph several values compared to a clamp meter and scale accordingly?
Even 3-5 points on a graph would make this quite a bit more accurate.

Thanks again for all you work!

David00, 2022-08-04:
Hi @vulture2600, thanks!

The challenge with noise coming in is that it could be coming from several different places. Most often, I think, is that the CT wire acts as an antenna and picks up surrounding fields.  It gets a little bit complicated for _how_ those values make their way up the stack and get presented to you, primarily because the calculations are all done using a RMS technique on a batch of 2000 samples from each channel. So, it could be one big spike that happened in a batch of samples, or a steady stream of noise that's always there in every sample. 

When the noise is caused by big spikes, adjusting the sampling technique to shrink the batch size and build a moving average would help reduce the impact of spikes.  If it's a steady stream of noise, it should be calibrated out with a scaling factor. But, in either case, having a hard cut off for low readings has worked as a "software" filter.  Here's an example in the code that currently only checks the `solar_power` variable:
https://github.com/David00/rpi-power-monitor/blob/master/power-monitor.py#L461

You could implement the same thing by checking the `grid_#_power` variables and setting them (and the amperage & PF) to 0 if they fall under a threshold of your choosing.

What are you thinking regarding graphing points compared to a clamp meter? I don't think I follow what you're asking with that point.

vulture2600, 2022-08-05:
Hey David,
Thanks for the response.

To clarify my question, If I clamp the CT around a grounded wire, I should show zero amps. Induced noise makes it show a non zero number, around an amp or so. My clamp meter should also show zero. When I clamp the CT around another wire along with my clamp meter, I can calibrate the scaling factor so that they show the same amperage. The problem is when I turn on more devices, like a few air conditioners, the amperage on my clamp meter will show 20a for example, but the readings from the CT might be a few amps off in either direction. My proposal then, is to have the user take several samples at known values with a clamp meter and enter them in, and then have the software calculate scaling factors as a non linear curve instead of a hard value. It might be difficult for most people to get a sample above 20 or 30 amps, as a lot of things would all have to be running at the same time. Those with electric water heaters or dryers might have an easier time with higher current loads available and easily accessible. 
So to clarify:
1) user reads 0 amps with clamp meter, user enters 0. 
2) user reads 3 amps, user enters 3.
3) etc etc as high up as you can go, maybe like 5 data points.
4) software calculates a scaling curve that roughly follows the data points and extrapolates up to 50 or 100 amps.

Does that make more sense?

vulture2600, 2022-08-05:
David,
This is my clamp meter:
https://www.homedepot.com/p/Klein-Tools-400-Amp-Digital-Clamp-Meter-AC-Auto-Ranging-with-Temp-CL220/312649913?ITC=AUC-127738-23-12140

I'm going to "borrow" a Fluke from work and see how far off they are from each other.

David00, 2022-08-07:
Yes, that makes sense now. I've tried to stay away from extrapolating smaller measurements up because that can introduce errors that multiply in magnitude the higher up you go.  Also, the sensors themselves are not exactly linear.  They're usually fairly linear between 5% and 100% of the rating, but outside those bounds, they're not.

The "[loop trick](https://github.com/David00/rpi-power-monitor/wiki/Phase-Correction-Procedure#current-multiplication-example)" of tricking the sensor (and your handheld meter) into seeing more current than is actually being consumed helps get the measured current into the sensor's linear zone and get a good calibration.

That is a good meter to use, and I'm interested to see how the Fluke compares!

David00, 2022-09-28:
Hey @vulture2600, how'd your testing with the Fluke go?

vulture2600, 2022-10-01:
I never actually tried the Fluke. Maybe this time around i will!

closed 2022-11-13 by David00