David00/rpi-power-monitor #55: Buster vs. Bullseye.

bert269, 2022-09-07:
I was trying to upgrade the unit to Bullseye and followed the instruction to in PiPower again, but it fails to monitor.
It was really a while ago and I could not remember the exact details.

Do you have any plans to upgrade the software to be able to run under Bullseye, with the latest Docker and Grafana?

Thank you.

David00, 2022-09-10:
The power monitor software should run on Bullseye, but the problem is the SPI sample rate is cut in half, which impacts the accuracy of this project. See this issue in the official Raspberry Pi repository:
https://github.com/raspberrypi/linux/issues/3381

So, until the sample rate problem is solved, I can't recommend running this on Bullseye with a 5.x kernel.

David00, 2022-11-09:
Hey @bert269, we ended up resolving the SPI transfer rate issue in the `spidev` Python library, so I'm preparing to release an updated Bullseye image for this project. If you're still on the Buster release, you should be safe to perform a `sudo apt update && sudo apt upgrade` on your system. 

You'll also need to update `spidev` and disable the Pi CPU governor with the following:

```bash
pip install -U spidev
sudo su
echo "force_turbo=1" >> /boot/config.txt
```

Once that's done, reboot your Pi, and it should be good to go. You may want to stop the service, then generate a debug plot (just to validate the sample rate), and then restart the service:

```bash
sudo systemctl stop power-monitor.service
cd ~/rpi-power-monitor
python3 power-monitor.py debug "speed check"
```

Take note of the "sample rate" in the output when generating the power monitor debug - it should be around 28 KSPS on a Pi 4.

>Finished Collecting Samples. Sample Rate: 28.41 KSPS

While this upgrade is safe to perform, you should **not** execute an upgrade for the power monitor code unless you're in the process of setting up the power monitor.  The main reason is that the channel numbers recently changed, so instead of using 0 - 5, it now uses 1 - 6.  More to come on the upgrade procedure from v0.1.x to v0.2.0.

closed 2022-11-09 by David00

bert269, 2022-11-09:
Thanks David.

The "sudo apt update && sudo apt upgrade" did not install Bullseye on the pi4. It only updated the packages. I am still on Buster.
Anyway, I did do the "speed check", but I only got around 16:
pi@piPower:~/rpi-power-monitor $ python3 power-monitor.py debug "speed check"
sample count is 10000
2022-11-09 07:10:03 : Finished Collecting Samples. Sample Rate: **16.64 KSPS**
2022-11-09 07:10:03 : Building plot.
2022-11-09 07:10:07 : Chart created! Visit http://192.168.1.241/speed_check.html to view the chart. Or, simply visit http://192.168.1.241 to view all the charts created using 'debug' and/or 'phase' mode.

Then I went a step back and tried the pip install -U spidev, hopefully to resolve the speed issue, but it failed with:
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools>=61.0
    Could not find a version that satisfies the requirement setuptools>=61.0
I guess it is expecting Bullseye, but it still found Buster.

Please let me know when you have posted the Bullseye image on your website.

Appreciated.

David00, 2022-11-09:
Right - Bullseye refers to the base Ubuntu image version, but the update && upgrade will install a newer kernel (similar to Bullseye's default kernel).

The setuptools message implies you'll need to upgrade pip with one of the following:

`pip install -U pip` 
`pip3 install -U pip`

Then, you should have a newer setuptools version, which should let you install the newer spidev version 3.6. 

Does your power monitor installation use the CT0 - CT5 channel names in the code and InfluxDB? I'm wondering because this will pose a challenge for those wanting to upgrade the power monitor code to the latest version and keep all their old data. I'll probably have to put together an upgrade guide for this scenario.

corbolais, 2022-11-09:
You could do `apt dist-upgrade` if you wanted to upgrade (providing that the repositories in /etc/apt/sources.list[.d/] are updated).

David00, 2022-11-13:
Hi folks, version 0.2.0 was released this weekend! There's a new custom OS image built with the latest Pi OS (Buster) image, which is part of the `release-v0.2.0` branch.

https://github.com/David00/rpi-power-monitor/tree/release-v0.2.0