David00/rpi-power-monitor #56: Panel plugin not found: timeseries

askewdread, 2022-09-15:
first off, great project! absoultley love it

have just set it up following the wiki, and using the individual CT dashboard im getting the following error on the CT listings
Panel plugin not found: timeseries

do you know how i can fix it? have never used grafana before and google doesnt seem to answer it

David00, 2022-09-15:
Hi @askewdread, thanks!

How did you install the power monitor? Custom OS, or by following the manual setup steps?  

In either case - can you share the output from the Terminal of:

`docker ps -a`

askewdread, 2022-09-16:
i used your image, after editing the panel i was able to change it to stat and its showing data now, but it doesnt seem to like the timeseries one
CONTAINER ID   IMAGE                   COMMAND                  CREATED        STATUS         PORTS                                       NAMES
160df37c6a80   grafana/grafana:7.3.7   "/run.sh"                36 hours ago   Up 4 minutes   0.0.0.0:3000->3000/tcp, :::3000->3000/tcp   grafana
164cc6b5a43d   influxdb:1.8.3          "/entrypoint.sh infl…"   36 hours ago   Up 4 minutes   0.0.0.0:8086->8086/tcp, :::8086->8086/tcp   influx

David00, 2022-09-16:
Ok, thanks. I'll need to update the Grafana version that's included in my custom image.

To fix this, you can remove the grafana container, and launch a container with a newer image, with the following commands. (Run one by one)

```
docker stop grafana
docker rm grafana
 docker run -d --restart always --name grafana -p 3000:3000 grafana/grafana:8.1.8
```

David00, 2022-09-19:
Oops! Not intentionally, but after looking back at it, yes, unfortunately it would remove the dashboards. The reason is because the custom image doesn't map a local folder to the container (I need to change this).

I would recommend removing the container again, and re-creating it, this time with a mapped volume, so you can update your Grafana version without impacting the dashboards.  But, doing this will remove your dashboards again if you've already re-added them, so you should export them first.

```
docker rm grafana
docker run -d --user 0 --restart always --name grafana -p 3000:3000 -v /opt/grafana:/var/lib/grafana grafana/grafana:8.1.8
```

This will keep your local dashboards and settings stored on the Pi outside of the Docker environment in `/opt/grafana`.

Also, the default dashboards that are included in my custom image are here:
https://github.com/David00/rpi-power-monitor/wiki/Software-3.-Building-your-Dashboard#dashboard-options

askewdread, 2022-09-19:
haha excellent thanks for that

closed 2022-09-20 by askewdread