David00/rpi-power-monitor #58: Calibration issues

askewdread, 2022-09-19:
when i set this up i missed a bunch of the calibration steps and just trying to get them done but something doesnt seem to be working correctly... its using your custom image on a pi4 and the kernel version appears to be the correct 4.19 one

my issue is CT1 is the main feed coming into the house (in NZ, so on 240v single phase), and for some reason its showing that CT3 which is our swimming pool is using more power than what is actually coming into the house which suggests somethings not quite right

trying to do the automatic calibration following the wiki comes up with the following suggestions on what to enter in the phase adjustments but they dont seem to be what they should be

CT1 (Main incoming feed) 2.54774229
CT2 (Hot water) 4.74891988
CT3 (Swimming pool) 4.1974763

[image]

David00, 2022-09-19:
Did you stop the power monitor service (`sudo systemctl stop power-monitor.service`) before running the phase calibration?

askewdread, 2022-09-19:
yeah,  i had missed that the first time but those numbers are with it stopped

askewdread, 2022-09-19:
[CTphase-correction-result.zip](https://github.com/David00/rpi-power-monitor/files/9600990/CTphase-correction-result.zip)

attached

David00, 2022-09-19:
Thanks. It looks like CT1 is fine and doesn't really need to be phase corrected - so you can leave that value as 1.0.  

CT2 and CT3 are being clipped (notice the flat top/bottom of the wave) so the phase correction algorithm must be having a hard time finding the peak, hence the higher than expected values.

If you're using the loop trick to multiply the current during calibration, try removing 1 loop, then run the phase correction mode again, and inspect the HTML output plot to see if the clipping is gone.

askewdread, 2022-09-19:
this is testing CT3, and its suggesting 3.21662917
[CT3-phase-correction-result.zip](https://github.com/David00/rpi-power-monitor/files/9601078/CT3-phase-correction-result.zip)

David00, 2022-09-19:
CT3 looks pretty equal too, so I'd try it out with 1.0 for the phase correction and see how it goes.

askewdread, 2022-09-19:
ok cool thanks for that.. im waiting for a meter to arrive with the amp loop so will see how that goes :)

thanks for all the help!

askewdread, 2022-09-20:
all looks a lot better after using the new meter to adjust the CT's, making a lot more sense now :)

thanks!

closed 2022-09-20 by askewdread