David00/rpi-power-monitor #61: Remote influx

cis2131, 2022-09-23:
Is it possible to send data to a remote influx server?.

David00, 2022-09-23:
Yes, simply replace 'localhost' with the IP or URL of your remote server in config.py.  If you're running it on a different port, or have credentials setup on your server, you can add them here too.

https://github.com/David00/rpi-power-monitor/blob/master/config.py#L21

David00, 2022-09-23:
Also I should note that the remote Influx server should be 1.8.X.  For this to work with the newer Influx 2.X software, I'll need to make some changes.

David00, 2022-09-28:
Hey @cis2131, did this work for you?

cis2131, 2022-09-28:
Sorry for not getting back to you.. 

I am currently using 1.8, so that is not a problem. But at some time I would like to upgrade. Is that a big problem?.
My plan is to monitor usage from powerstrips in server racks, so i can get an alarm if usage is to high.
I currently use grafana for monitoring everything else, that's why I would like to add it to my own influx.

cis2131, 2022-09-28:
Thanks for quick reply.
I have just read post #63, it answers my question.

David00, 2022-09-28:
No worries! I'll go ahead and close this now but feel free to reopen or add additional comments in the future.

closed 2022-09-28 by David00