David00/rpi-power-monitor #63: Latest influx ?  Latest Grafana ?

beadon, 2022-09-28:
influx is moving to a v2.0, however it doesn;t appear that the raspi 3B+ is supported just yet.  Know anything different ?  

Grafana too is up to docker container 8.2.6 - going to give this a try.

David00, 2022-09-28:
Hi @beadon, I believe InfluxDB 1.8 will stay around indefinitely as v2.0 is significantly different.  I am interested in moving to v2.0 eventually but it requires learning their Flux query language (and converting all DB interactions to use Flux).  I just don't have that much time to take on that big of a change now.

Updating Grafana is perfectly fine, just make sure to export your dashboards first if you've customized them, because the current build of my custom OS image does not map the Grafana container's storage to the Pi itself, so removing the container to upgrade it will remove all the Grafana data. 

Note that I just added the volume to the run command to step 10 on the [manual installation steps](https://github.com/David00/rpi-power-monitor/wiki/Software-0.-Installation#manual-installation-steps) page in the wiki, so when you recreate the Grafana container, it will now map to the local disk and your dashboards and settings will persist between upgrades:

`docker run -d --restart always --name grafana -p 3000:3000 -v /opt/grafana:/var/lib/grafana grafana/grafana:8.1.8`

For anyone else reading this in the future that isn't familiar with Docker, to upgrade Grafana:

* Backup your dashboards by going into the Grafana UI, opening the dashboard, going to the dashboards settings (gear icon - top right), and clicking on the JSON Model tab. Copy the entire JSON model out to simple text editor and save it.
* SSH to the Pi, and run the following commands, one by one (and you can change the Grafana version number at the end to one of the latest versions found [here](https://hub.docker.com/r/grafana/grafana/tags).  Just note that Grafana is known to make breaking changes between major versions, so you may need to update your dashboards to their latest formats if you jump between v8 and v9 for example.

```
docker stop grafana
docker rm grafana
docker run -d --restart always --name grafana -p 3000:3000 -v /opt/grafana:/var/lib/grafana grafana/grafana:8.2.6
```

vulture2600, 2022-10-01:
Hello David,
I had to rebuild my entire Pi because my old SD card got corrupt. Now I'm using a 64GB usb stick for a fresh install of bullseye 64bit.  I got stuck at building the grafana container. "docker ps" shows the grafana process "Restarting (1) 41 seconds ago". Any ideas?

David00, 2022-10-01:
Hi @vulture2600 , unfortunately that is prone to happen to most (all?) microSD cards when it's subjected to constant writes.

Let's check the docker logs for the container to see if it gives us a hint as to why it's restarting.

Try this: `docker logs grafana -n 50`, which will show you the last 50 lines of the logs from the Grafana container.  If you attach the output to your next comment, I'll take a look.

vulture2600, 2022-10-01:
Hey David,
thanks for the quick reply!

This is the output  shown:
GF_PATHS_DATA='/var/lib/grafana' is not writable.
You may have issues with file permissions, more information here: http://docs.grafana.org/installation/docker/#migrate-to-v51-or-later
mkdir: can't create directory '/var/lib/grafana/plugins': Permission denied
GF_PATHS_DATA='/var/lib/grafana' is not writable.
You may have issues with file permissions, more information here: http://docs.grafana.org/installation/docker/#migrate-to-v51-or-later
mkdir: can't create directory '/var/lib/grafana/plugins': Permission denied
GF_PATHS_DATA='/var/lib/grafana' is not writable.
You may have issues with file permissions, more information here: http://docs.grafana.org/installation/docker/#migrate-to-v51-or-later
mkdir: can't create directory '/var/lib/grafana/plugins': Permission denied

David00, 2022-10-01:
It looks like I had somehow removed the `--user 0` part of the `docker run` command.

Try to remove your Grafana container and rebuild with:

`docker run -d --user 0 --restart always --name grafana -p 3000:3000 -v /opt/grafana:/var/lib/grafana grafana/grafana:8.2.6`

(I just edited the comment above, and the installation instructions to reflect this).

vulture2600, 2022-10-01:
That worked! Ill continue here with the rest of the set up shortly. Thanks again!

closed 2022-10-03 by beadon