David00/rpi-power-monitor #65: Cost Calculation

Rangeman0, 2022-10-16:
With the 'Today's Cost' panel on the Home Power dashboard, is it possible to also add a daily network connection charge to be added?

I'm intending to factor in the connection charge per day and create a duplicate of the Today's Cost panel which would then have the variables of 'cost_ per_kWh' and 'cost_per_day' summed across a month. 

My particular bill date is in the middle of the month so this would be able to give a more realistic prediction of the upcoming bill.  

I'm able to create a variable to supply the relevant cost but, I am at a loss how to program the query on Grafana (If it is even possible)

Thanks!

nightcustard, 2022-10-16:
Could you possibly do this by creating a continuous query which calculates the cost of the energy used over a time interval (eg) 10 minutes and adding the daily connection charge divided by 144 (1440/10) and then display the continuous query summed over the month?
Hopefully I'm not missing the point ....

David00, 2022-10-17:
It is tricky to do multi-step operations in Grafana. I'd recommend using either a subquery or a continuous query to help prepare the data before Grafana tries to pull it.

Check out this page for subqueries:
https://www.influxdata.com/blog/tldr-influxdb-tech-tips-january-26-2017/

And this page for CQs:
https://docs.influxdata.com/influxdb/v1.8/query_language/continuous_queries/

I'd assume you might even be able to create a CQ using a subquery - I haven't tried it though.

Hopefully this helps - I can't say I fully understand what you're trying to do, though.

Rangeman0, 2022-10-22:
I got quite close with:

SELECT integral("power") * [[cost_per_kWh]]/3600000 + [[cost_per_day]] *30 FROM "net" WHERE $timeFilter

I don't know how to change the *30 (days) into the interval of the time picker. If I use $__Interval I just end up with an error in the query.  

Nightcustard - I tried going down that road but ended up finding it easier to work per day - I did explore your suggestion quite a lot but got too confused as the numbers get super tiny. 

David - I'm intending to make a copy of your panel 'Today's Cost', but I would like to see a 'Month's cost' as the duplicate panel. In Australia we pay a daily connection charge - I'm working on $0.8664 per day as my 'cost_per_day' variable with the intention of using the time picker to choose the days of my billing cycle (midnight to midnight) and then get a display of what the bill will end up at. 

ie; aprox $25 / month + electricity use (your 'cost_per_kWh' variable) for a total of $xxx.xx 

Using my query as above, I'm off by 10% (which is added to the bill as tax, but I'm not sure that Grafana will be able to add this automatically) so it seems correct, but only when the time picker has a month of 30 days selected. As some months are longer / shorter this will be inaccurate across a longer stretch. 

Any thoughts on the $__Interval code (to replace *30(days) after by day variable)?

Rangeman0, 2022-10-22:
I did some googling and went with the average days per month of 30.4. 

I also figured out Grafana's percentage calculator as you can't do 0.10 for 10%.   

My Cost per month panel code is 
SELECT integral("power") * [[cost_per_kWh]]/3600000 + [[cost_per_day]] *30.4 *1.1 FROM "net" WHERE $timeFilter 

This could also be adjusted to quarterly with a *91.2 after the cost per day variable.

This gets me within $1 of the actual bill charged (on average) which is close enough unless someone comes up with a more accurate code. 

Thanks for your input - David you can close unless someone posts a better solution.

David00, 2022-11-09:
Thank you for sharing what ended up working for you - hopefully this helps future readers! I'm also glad to hear that the estimated costs are close to your real costs!

closed 2022-11-09 by David00