David00/rpi-power-monitor #67: Phasecal below 1.xxx

ejg3855, 2022-10-30:
I am having some trouble with phasecal values and my CT leads, I have checked the obvious with the system running as a service and that seemed to improve it a bit but still seeing values over 1.xx

I did purchase all the hardware from you and I think I have all the resistors installed in the correct locations. 

I am seeing values recommended at :

2022-10-28 23:42:35 : Wave 1/3 results:
2022-10-28 23:42:35 :  Best PF: 0.036897026844769185 using phasecal: 1.5187898946333451
2022-10-28 23:43:10 : Wave 2/3 results:
2022-10-28 23:43:10 :  Best PF: 0.031783049466366946 using phasecal: 3.2677097966075936
2022-10-28 23:43:44 : Wave 3/3 results:
2022-10-28 23:43:44 :  Best PF: 0.030424070733198246 using phasecal: 4.492907124990358
2022-10-28 23:43:44 : Please update the value for ct1 in ct_phase_correction in config.py with the following value: 3.09313561
2022-10-28 23:43:44 : Please wait... building HTML plot...
2022-10-28 23:43:47 : file written to CT1-phase-correction-result.html

Also where is it writing these html files to I can't seem to find them in any of my pi directories.

David00, 2022-10-30:
HI @ejg3855, what model Pi are you using?

Did you install using my custom OS image, or by following the manual instructions?

Can you generate and then share a debug plot? Run `python3 power-monitor.py debug "Test"`, and then grab the chart by going to `http://<your Pi's IP address>/Test.html`.

The HTML plots go into `/var/www/html`, but it's easiest to get them through a web browser just by navigating to your Pi's IP address.

ejg3855, 2022-10-30:
@David00 thanks for the quick reply.

I get an error with the debug test

`pi@powermon:~/rpi-power-monitor $ python3 power-monitor.py  debug "Test"
2022-10-30 17:38:42 : Finished Collecting Samples. Sample Rate: 30.75 KSPS
2022-10-30 17:38:42 : Building plot.
Traceback (most recent call last):
  File "power-monitor.py", line 692, in <module>
    plot_data(samples, title, sample_rate=sample_rate)
  File "/home/pi/rpi-power-monitor/plotting.py", line 37, in plot_data
    ct0 = samples['ct0']
KeyError: 'ct0'
pi@powermon:~/rpi-p`

David00, 2022-10-30:
It looks like you're running an older version of the software - can you try to update? Instructions here:

https://github.com/David00/rpi-power-monitor/wiki/Software-4.-Updating-the-Code

David00, 2022-10-30:
Thanks, can you also attach the `CT1-phase-correction-result.html` file?

I think this situation happens when the PF is already incredibly good, which we can determine by looking at the phase correction result plot.

ejg3855, 2022-10-30:
lets try this:

[CT1-phase-correction-result.html.pdf](https://github.com/David00/rpi-power-monitor/files/9896745/CT1-phase-correction-result.html.pdf)

David00, 2022-10-30:
I can't quite zoom in to that like I can with the HTML file - can you attach it? Just go to File -> Save in you web browser when you're looking at the plot, then you'll get the actual HTML file.

ejg3855, 2022-10-30:
github doesnt seem to support html attachements. 
[CT1-phase-correction-result.zip](https://github.com/David00/rpi-power-monitor/files/9896781/CT1-phase-correction-result.zip)

David00, 2022-10-30:
Ok, thanks! In that specific plot, there is essentially no data for CT1. Was the load turned on when you ran that phase correction, with the sensor measuring only the hot wire in the load's power cable (or the hot wire in the electrical panel?)

ejg3855, 2022-10-30:
lets try this.
[take 2 ct1 and ct2.zip](https://github.com/David00/rpi-power-monitor/files/9896926/take.2.ct1.and.ct2.zip)

David00, 2022-10-30:
Hmm, the current wave is looking very strange:

[image]

What is the load that you're measuring? Can you try testing on different input channels to see if the current waveform looks the same?

ejg3855, 2022-10-30:
It's a heat gun, was the only highish resistance load I have. I've done the test on 1/2 and posted the results above. I can try again again with 3/4. I have the solid cat5 wiee connected to the red from the CT lead.

David00, 2022-10-30:
Ok. The fan in the heat gun might be messing with things slightly, but I don't believe it would be producing a current waveform like what we're seeing. I saw that channel 2 had the same issue.

While checking 3 and 4, can you also share a picture of your assembled PCB? I should be able to check to see if it's assembled right.

ejg3855, 2022-10-30:
[ct3-4.jpg.zip](https://github.com/David00/rpi-power-monitor/files/9897049/ct3-4.jpg.zip)

i was thinking exactly that and included a photo

David00, 2022-10-31:
The board looks good - I don't see any issues there.  It might just be heat gun - do you have any other high power sources to test with? If not, I can run the phase calibration for that sensor and share the value for you to plug in.

ejg3855, 2022-10-31:
I have 2 200a sensors and 2 60a CT sensors. What other things do you suggest for high draw? I have a old school 200amp battery charger?

David00, 2022-10-31:
For the calibration part, a large halogen light bulb is ideal, because it's pretty high current draw, and also purely resistive.  Your heat gun is good for measuring a high current, but the motor in it throws some inductive characteristics into the current waveform.  But still, it shouldn't be reporting a wave form like what we're seeing, so it must be something else.

Can we try reducing the clock speed in `common.py`? Let's change `spi.max_speed_hz = 1750000` to `spi.max_speed_hz = 1000000`, and the generate a new debug plot with it measuring your same heat gun load?  I'm suspecting that the ADC might be faulty and reducing the clock speed might help determine that.

ejg3855, 2022-10-31:
reduced the sample rate and attached.
[Test.zip](https://github.com/David00/rpi-power-monitor/files/9905317/Test.zip)

David00, 2022-10-31:
Thanks - that's really strange that all channels - except for the AC voltage channel - are showing the same type of distorted waveform. I think the ADC in your kit might be faulty. Can you shoot me an email at github@dalbrecht.tech and I'll send out a replacement?

ejg3855, 2022-11-05:
Replaced the ADC not seeing much change. 

Would it benefit to rebuild
[Test.zip](https://github.com/David00/rpi-power-monitor/files/9944737/Test.zip)
 the pi image? 

I am just trying to create a logical path before I go and wholesale change the board.

David00, 2022-11-06:
Before changing the board, have you tried all the sensors too? We haven't really looked into the possibility of a faulty sensor yet so I'm thinking that might be a good easy step before replacing the board.

ejg3855, 2022-11-06:
I can try again. I'm suspect of my LAN cable connections at this point. Not sure what else it could be.

ejg3855, 2022-11-07:
I confirmed the sensor were corrected to the LAN properly. then I removed one of the leads and soldered it right to the board on CT5. I think I got the pins correct from looking at the bare extra board.

[image]

`thon3 power-monitor.py phase

Which CT number are you calibrating? Enter the number of the CT label [1 - 6]: 6

#------------------------------------------------------------------------------#
# IMPORTANT: Make sure that current transformer ct6 is installed over          #
#            a purely resistive load and that the load is turned on            #
#            before continuing with the calibration!                           #
#------------------------------------------------------------------------------#

Continue? [y/yes/n/no]: y
2022-11-07 18:36:06 : Please wait while I read ct6 and calculate the best PHASECAL value. This can take a few minutes, so please be patient.
2022-11-07 18:37:28 : Wave 1/3 results:
2022-11-07 18:37:28 :  Best PF: 0.8174227659750949 using phasecal: 1.4888637335882218
2022-11-07 18:38:49 : Wave 2/3 results:
2022-11-07 18:38:49 :  Best PF: 0.8011874789909631 using phasecal: 2.3067227440403673
2022-11-07 18:40:11 : Wave 3/3 results:
2022-11-07 18:40:11 :  Best PF: 0.6948684263975088 using phasecal: 4.537836196240262
2022-11-07 18:40:11 : Please update the value for ct6 in ct_phase_correction in config.py with the following value: 2.77780756
2022-11-07 18:40:11 : Please wait... building HTML plot...
2022-11-07 18:40:15 : file written to CT6-phase-correction-result.html`

David00, 2022-11-08:
Hmm - is that with a different sensor, or the same one? The current sensor trace in the CT6-phase-correction is still showing that same odd trace, so if it's a different sensor, it's not the sensors.

Another option - can you try with a different load (like an incandescent lightbulb) and use the [loop trick](https://github.com/David00/rpi-power-monitor/wiki/Phase-Correction-Procedure#current-multiplication-example) to multiply the current flowing through the sensor, and then generate a debug plot? I'm suspecting it might be the test load that you're using.

ejg3855, 2022-11-08:
more promising results with my percolator.

`2022-11-08 16:56:02 : Wave 1/3 results:
2022-11-08 16:56:02 :  Best PF: 0.999297609996258 using phasecal: 1.3488501525493075

2022-11-08 16:57:23 : Wave 2/3 results:
2022-11-08 16:57:23 :  Best PF: 0.999032581248988 using phasecal: 1.4482316150707089

2022-11-08 16:58:44 : Wave 3/3 results:
2022-11-08 16:58:44 :  Best PF: 0.9989602901343453 using phasecal: 1.4647194142493882
2022-11-08 16:58:44 : Please update the value for ct6 in ct_phase_correction in config.py with the following value: 1.42060039`

[CT6-phase-correction-result.zip](https://github.com/David00/rpi-power-monitor/files/9963459/CT6-phase-correction-result.zip)

David00, 2022-11-08:
Yep, that looks much better! The current waveform in the plot looks good now, and the phase-correction-result plot looks great too.  So, this might have just been due to your heat gun!

ejg3855, 2022-11-08:
Im still meeting a challenge with he Cat5 connector at the moment. Working on doing a bit of diag on it.

David00, 2022-11-08:
They can be tricky... here's a note in the Wiki about the cat5 connector... hopefully it helps!

https://github.com/David00/rpi-power-monitor/wiki/Hardware-Assembly#assembling-the-current-transformer-cat5e-cable

ejg3855, 2022-11-08:
It seems to be all working now, thanks for the assistance. You want the
extra parts back?

David00, 2022-11-08:
Great! You're welcome, and no, feel free to keep them :)

closed 2022-11-08 by David00