David00/rpi-power-monitor #68: CI / CD Pipeline Proposal

beadon, 2022-11-12:
I would like to offer a CI / CD pipeline to build this.  

First cut proposal -

GitHub Actions are set to build based on gitflow workflow, release/feature branch, and tagging.  The details of this are here:  https://www.atlassian.com/git/tutorials/comparing-workflows/gitflow-workflow

-- Benefits --
- Speedy building and release process
- Permit scale of the project to be increased without developers running into one another.  These good practices help perform the tedious tasks of build, test, release.   
- Consistency in all releases
- Sharpen our skillsets
- Codify a location where tests can be performed (CI) so no regressions occur
- Allow work to be distributed amongst any existing or future contributors

Famous last words, but it shouldn't be too difficult!  :)

David00, 2022-11-13:
I'm all for it!  I agree that it won't be too difficult. The hardest part might be coming up with valuable tests to run (and that a runner can perform). A lot of the "gotchas" are going to be specific things about the Pi hardware itself.

David00, 2022-11-25:
It looks interesting - I think the biggest benefit from that would be having authorized users be able to interact with the derek bot for issue and PR management. I don't see too much benefit for the CI/CD process though, aside from maybe the changelog generation.