David00/rpi-power-monitor #75: (PR Welcome) Move calibration/custom settings out of config.py to a TOML config file

David00, 2022-11-25:
There are two problems that stem from having user-inputted config settings into [`config.py`](https://github.com/David00/rpi-power-monitor/blob/master/rpi_power_monitor/config.py):

1.  The recent refactor that converted the power_monitor into a module (#62) allows the code to be installed with Pip via the included setup.py file.  When pip installs the code, it compiles the python source into bytecode, which means the module will no longer use the contents of config.py - instead, it uses the compiled version of config.py from the latest pip installation. Essentially, changing a value in config.py requires reinstalling with pip. 

2. The config.py file is a git-tracked part of the codebase, and making changes to your local version means git will want you to do something with those changes before you can interact with the upstream codebase again. The current upgrade instructions rely on the end user to use git stash to store their changes, which is unnecessarily complicated for those unfamiliar with git.

Specifically, the following user-facing variables should be moved to TOML configuration file:
* GRID_VOLTAGE
* AC_TRANSFORMER_OUTPUT_VOLTAGE
* db_settings
* CT_PHASE_CORRECTION
* ACCURACY_CALIBRATION

The necessary code to read from the TOML and apply the settings upon startup will also need to be introduced and tested.

The TOML file should also have a version defined in it to help control compatibility with future code changes.

David00, 2022-12-07:
Also, add a cutoff threshold for users to specify at what value they want to ignore any readings.

David00, 2023-01-16:
I've started working on this issue.

Cube3SGM, 2023-02-07:
Hello, I have a similar problem related to the config.py when setting up for the calibration.

After modifying the config.py (236v and 10.1v) I am unable to successfully complete "pip install" I receive the message
Defaulting to user installation because normal site-packages is not writable
ERROR: You must give at least one requirement to install (see "pip help install")

I have replied in #83 but realised that this is a closed discussion, not sure if any one would see it!  Sorry if I have confused things by cross posting!

Any help appreciated, Regards Simon

Cube3SGM, 2023-02-09:
Hello David,

Really appreciate the fast response, you were absolutely spot on, I had missed the period. Sorry for the silly mistake.

Unfortunately I have run into another issue when setting it to run as a service.  I will give a quick description of the issue here but if you would like me to start another thread then let me know.

Created the service file, modified it as per instructions (Note. The Execstart statement ExecStart=/usr/bin/python3 /home/pi/rpi-power-monitor/power-monitor.py  Was changed to ExecStart=/usr/bin/python3 /home/pi/rpi_power_monitor/power_monitor.py)     (Hyphens and underscores)
Enable and reboot went without a problem

When I run    sudo systemctl status power-monitor  I get the following message.

Loaded: loaded (/etc/systemd/system/power-monitor.service; enabled; vendor present:enabled)
Active: failed (Result: resources) since (date and time)
Process: 382 ExecStart=/usr/bin/python3 /home/pi/rpi_power_monitor/rpi_power_monitor/power_monitor.py (code=exited, status=0/SUCCESS)
Main PID: 382 (code=exited, status=0/SUCCESS)
CPU: 4.566s

Feb 09 17:34:23 rpipowermonitor systemd[1]: power-monitor.service: Failed to schedule restart job:Unit docker.service not found
Feb 09 17:34:23 rpipowermonitor systemd[1]: power-monitor.service: Failed with result 'resources'
Feb 09 17:34:23 rpipowermonitor systemd[1]: power-monitor.service: Consumed 4.566s CPU time.

I would love to help with testing the 0.3 beta but as you can tell I am pretty new to this so I am nervous I will cause you more headaches than help! Regards Simon
P.S.  I am using your custom PI OS image on a Raspberry 3.

David00, 2023-02-09:
Hi @Cube3SGM, no problem!

What you described above appears to be a documentation issue! I had forgotten to remove the docker dependencies from the "running as a service" page in the Wiki. I just made the changes to that page, but essentially, just edit `/etc/systemd/system/power-monitor.service` again, and remove these two lines:

	After=docker.service
	Requires=docker.service

I'm sure you'll do just fine in testing the 0.3 beta! Don't worry about causing more headaches than help - I want the documentation to be easy to follow for beginners too, so if you go through the steps and have trouble or something doesn't make sense, feel free to let me know and I'll do my best to help you out and also see if any improvements can be made.

If you want to get started on the v0.3.0 beta (from the current v0.2.0 custom OS image), have a look at the upgrade instructions on the `v0.3.0-beta` release page here:
https://github.com/David00/rpi-power-monitor/releases/tag/v0.3.0-beta

Cube3SGM, 2023-02-11:
Hi David,
All is working now on the bench and test loads... Fantastic
One small issue I had when configuring Grafana,  Section 3 Building your Dashboard.  In dashboard option, it points to two dashboard with screenshots and JSON files. This is where I went to load a dashboard into Grafana as instructed but the Links return a 404 page.  I subsequently realised that in your default OS image they are pre-loaded. So I am fine, its just a heads up.  Now I will prepare another SD card and have a run at 0.3 beta.   Thanks for your help. Regards Simon

David00, 2023-02-11:
>  All is working now on the bench and test loads... Fantastic

Awesome!

> One small issue I had when configuring Grafana, Section 3 Building your Dashboard. In dashboard option, it points to two dashboard with screenshots and JSON files. This is where I went to load a [...]

Thanks for letting me know. I just fixed those links. 

> Now I will prepare another SD card and have a run at 0.3 beta. Thanks for your help. Regards Simon

Thank you for checking it out! I'd love to hear your feedback on it in the discussion thread: https://github.com/David00/rpi-power-monitor/discussions/88

David00, 2023-03-29:
This has been completed and integrated into v0.3.0, which will be releasing soon.

commit b832cf9b (2023-04-02): Major refactor to resolve #75 and #76. See comments...
- Introduced a new configuration file (config.toml) to simplify the configuration and setup of the power monitor. It is kept outside of version control, so see the docs for the config file details.
- Introduced a new command line interface with a more standard method of passing flags and values
- Simplified the sampling process to remove the need for a complex phase correction.
- Applied a standard calibration constant to improve the out-of-the-box accuracy of all sensors from my shop (and further reducing the need for calibration)
- Impro [... cut]
closed 2023-03-29 by David00