David00/rpi-power-monitor #80: Voltage readings VERY low after inputting voltages during voltage tuning

emersonsc, 2022-12-02:
My voltage readings are WAY off. I measured the voltage per the voltage tuning guide. How ever, at step 8, voltage is MUCH lower than it should be reading. I used the v0.2.0 image.

Here is my config.py
# Using a multimeter, measure the voltage of the receptacle where your 9V AC transformer will plug into.
# Enter the measured value below.
GRID_VOLTAGE = 120.9

# Using a multimeter, measure the output voltage of your AC transformer. Using the value on the label is
# not ideal and will lead to greater accuracy in the calculations.
AC_TRANSFORMER_OUTPUT_VOLTAGE = 10.51

Here are my readings:

pi@shopenergymon:~/rpi_power_monitor $  python3 ~/rpi_power_monitor/rpi_power_monitor/power_monitor.py terminal
2022-12-01 20:37:16 : Could not create a backup of config.py file.
2022-12-01 20:37:16 : ... Starting program in terminal mode
2022-12-01 20:37:16 : ... DB initialized.
2022-12-01 20:37:16 : ... Starting Raspberry Pi Power Monitor
2022-12-01 20:37:16 : Press Ctrl-c to quit...
2022-12-01 20:37:19 :
+---------+--------+-------+-------+-------+-------+-------+
|         |  ct1   |  ct2  |  ct3  |  ct4  |  ct5  |  ct6  |
+---------+--------+-------+-------+-------+-------+-------+
|  Watts  | 0.128  | 0.119 | 0.132 | 0.129 | 0.145 | 0.151 |
| Current | 0.022  | 0.022 | 0.023 | 0.022 | 0.024 | 0.024 |
|   P.F.  | 0.286  | 0.268 | 0.285 | 0.287 |  0.3  | 0.312 |
| Voltage | 20.309 |       |       |       |       |       |
+---------+--------+-------+-------+-------+-------+-------+
2022-12-01 20:37:22 :
+---------+--------+-------+-------+-------+-------+-------+
|         |  ct1   |  ct2  |  ct3  |  ct4  |  ct5  |  ct6  |
+---------+--------+-------+-------+-------+-------+-------+
|  Watts  |  0.26  | 0.266 |  0.27 | 0.241 | 0.254 | 0.285 |
| Current | 0.032  | 0.032 | 0.032 | 0.032 | 0.033 | 0.033 |
|   P.F.  | 0.406  | 0.408 | 0.417 | 0.372 | 0.384 | 0.423 |
| Voltage | 20.297 |       |       |       |       |       |
+---------+--------+-------+-------+-------+-------+-------+
^Cpi@shopenergymon:~/rpi_power_monitor $

emersonsc, 2022-12-02:
BTW just for poops and smiles I did the math, put in 7.1585 for the voltage calibration, and the numbers didnt. change.

David00, 2022-12-02:
Unfortunately, you have to do a pip install when the config.py file is updated. Try:

`cd ~/rpi_power_monitor ` followed by

`pip install .`

emersonsc, 2022-12-02:
config.py
# Using a multimeter, measure the voltage of the receptacle where your 9V AC transformer will plug into.
# Enter the measured value below.
GRID_VOLTAGE = 120.9

# Using a multimeter, measure the output voltage of your AC transformer. Using the value on the label is
# not ideal and will lead to greater accuracy in the calculations.
AC_TRANSFORMER_OUTPUT_VOLTAGE = 10.51

ACCURACY_CALIBRATION = {
    'ct1': 1,
    'ct2': 1,
    'ct3': 1,
    'ct4': 1,
    'ct5': 1,
    'ct6': 1,
    'AC': 5.00579,
}

Output:
+---------+---------+------+------+--------+------+------+
|         |   ct1   | ct2  | ct3  |  ct4   | ct5  | ct6  |
+---------+---------+------+------+--------+------+------+
|  Watts  |   -0.0  | -0.0 | -0.0 | -0.003 | -0.0 | -0.0 |
| Current |   0.0   | 0.0  | 0.0  | 0.002  | 0.0  | 0.0  |
|   P.F.  |    0    |  0   |  0   | -0.014 |  0   |  0   |
| Voltage | 121.016 |      |      |        |      |      |
+---------+---------+------+------+--------+------+------+
2022-12-01 21:24:03 :
+---------+---------+------+-------+------+------+------+
|         |   ct1   | ct2  |  ct3  | ct4  | ct5  | ct6  |
+---------+---------+------+-------+------+------+------+
|  Watts  |  0.006  | -0.0 | 0.006 | -0.0 | -0.0 | -0.0 |
| Current |  0.002  | 0.0  | 0.002 | 0.0  | 0.0  | 0.0  |
|   P.F.  |  0.022  |  0   | 0.022 |  0   |  0   |  0   |
| Voltage | 121.085 |      |       |      |      |      |
+---------+---------+------+-------+------+------+------+
2022-12-01 21:24:06 :
+---------+---------+-----+-----+-----+-----+-----+
|         |   ct1   | ct2 | ct3 | ct4 | ct5 | ct6 |
+---------+---------+-----+-----+-----+-----+-----+
|  Watts  |   0.0   | 0.0 | 0.0 | 0.0 | 0.0 | 0.0 |
| Current |   0.0   | 0.0 | 0.0 | 0.0 | 0.0 | 0.0 |
|   P.F.  |    0    |  0  |  0  |  0  |  0  |  0  |
| Voltage | 120.402 |     |     |     |     |     |
+---------+---------+-----+-----+-----+-----+-----+
2022-12-01 21:24:09 :
+---------+---------+--------+--------+--------+--------+--------+
|         |   ct1   |  ct2   |  ct3   |  ct4   |  ct5   |  ct6   |
+---------+---------+--------+--------+--------+--------+--------+
|  Watts  |  -2.047 | -2.418 | -2.791 | -3.016 | -2.201 | -3.203 |
| Current |  0.903  | 0.904  | 0.906  | 0.905  | 0.904  | 0.903  |
|   P.F.  |  -0.019 | -0.022 | -0.025 | -0.027 | -0.02  | -0.029 |
| Voltage | 122.023 |        |        |        |        |        |
+---------+---------+--------+--------+--------+--------+--------+

emersonsc, 2022-12-02:
Is it ok for the AC calibration number to be that high, or is there something going on?

David00, 2022-12-02:
It is not normal for the AC calibration to be that high - it should normally be pretty close to accurate just by entering the output voltage and the receptacle reading. Can you double check the values of R4 and R5 on the PCB, and also share a full screenshot of a [debug plot](https://github.com/David00/rpi-power-monitor/wiki/Software-1.-Overview#debug-mode) with the 9V AC transformer plugged in? I'll want to see the sample rate on the plot (or the text printout from the terminal), and the scale of the right axis.

emersonsc, 2022-12-02:
Raspberry Pi 4b 2gb
Sandisk Extreme 32gb Micro Sd

Sample Rate: 28.52 KSPS

[image]

emersonsc, 2022-12-02:
[image]
[image]

emersonsc, 2022-12-02:
I even tried with the other 9v ac adapter you sent me, and while the voltages seem to be more stable, they are still low, around 25v before doing changing the accuracy calibration values.

David00, 2022-12-02:
Thanks for the detailed screenshots. I see a problem in the debug screenshot.  See how the AC voltage wave is flat on the bottom?  

This tells me something is wrong with the bias voltage supply, which is this part of [the schematic](https://github.com/David00/rpi-power-monitor/wiki/Schematic):

[image]

So, something in the part of the schematic above isn't working properly.  It looks like your bias voltage is not at 1.65V, but instead somewhere around 0.80V or so, so your board isn't catching the full negative cycle of the AC wave and can't calculate the voltage properly.  This will impact all CT input channels as well.  This is suspiciously close to half of what it should be.

Things to check:

- The voltage on your [Pi's 3.3V rail](https://pinout.xyz/pinout/3v3_power#), both with and without the power monitor PCB installed.  As long as it's reading 3.3V without the PCB installed, your Pi is fine, and you can continue checking the next things. If it reads 3.3V with the PCB off, and not 3.3V with the PCB on, then something is up with the assembly of the PCB, which we'll walk through below. Continue to check...

- The polarity on all the electrolytic capacitors look good, but make sure there aren't any bridges on between the leads on the bottom side of the PCB, specifically between C4 and C5
- Double check the joints on R6 and R7 (they look a little bit empty)
- Check the DC voltage going into pin 5 on the left half of the ADC, circled below in red. This is where the bias voltage is supplied to the ADC, and it's supposed to read 1.65V (DC).  
[image]

- I don't see any bridges between U1, so that's good, but make sure they have good contact with the pads underneath.

emersonsc, 2022-12-02:
went through and filled any "empty" spots. I soldered from the bottom so I didnt see some of the voids on the top at the time.

After that, I powered the pi back up, checked the 3v3, and all is well there. Pin 5 on the ADC is a different story. Its bouncing all over the place from .6v to 1.1v but seems at times it settles around .77-.81 before it bounces around. It does that consistently though.

emersonsc, 2022-12-02:
I tried the other MCP3008, same story as far as low voltage, how ever its not as jumpy, staying solid around .7v

David00, 2022-12-02:
Ok. If the rest of the points look good then I think U1 might be bad. I will send a replacement kit with 2 U1s so you can try swapping it out on this board if you want. It might be tricky to get in there and replace the one part, though, so your call!

emersonsc, 2022-12-02:
Yeah ill need 2 U1s for sure. I went to put the other board together and found a pin missing on the 2nd U1 chip

emersonsc, 2022-12-02:
[image]

emersonsc, 2022-12-07:
CT5 HTML File [https://drive.google.com/file/d/1h7GqVUOMhqaTIErfXIs1zOj_70kuuxXj/view?usp=sharing](url)

CT6 HTML File [https://drive.google.com/file/d/1scTwQm49c_73F4bHfU_DpqOyzAxaTg8v/view?usp=sharing](url)

emersonsc, 2022-12-07:
> https://drive.google.com/file/d/1h7GqVUOMhqaTIErfXIs1zOj_70kuuxXj/view?usp=sharing

awesome thank you! It was over the 1.x so i was like awwww crap what did i do wrong now lol

David00, 2022-12-07:
Yeah the phase correction algorithm is a little finicky sometimes, which is one of the reasons why I want to get rid of it and go with streamlined sampling and calibration as a part of #76.

I'm glad the replacement IC worked out for you!

emersonsc, 2022-12-07:
Ok, one last anomaly, and other than that everything is great, even got the dashboards up and showing data!

is it normal to show a negative wattage, but positive amperage when a circuit is "off" or theres nothing plugged into any of the outlets?

[image]

[image]

David00, 2022-12-07:
Awesome! In a way, yes, it's normal in the sense that it could be noise entering the sensor wires, especially if they're ran parallel close to any high voltage wires.  You could add a small check to the code to ignore values less than X. If you don't have any sources of production (solar / generator / etc), then X could be zero, since you should never have a negative reading (unless the sensor is clipped on backwards).

There currently isn't support for having a hard cutoff threshold to ignore values, but it would be trivial to add. I'll make it a part of the issue I mentioned above when I migrate config.py to a config.toml approach.  But, until then, you could modify the following lines:

https://github.com/David00/rpi-power-monitor/blob/08cea42f25f161a686b169607587bae769c09440/rpi_power_monitor/power_monitor.py#L534-L548

... to something like this:

```python
# Prepare values for database storage 
grid_1_power = results['ct1']['power']
grid_2_power = results['ct2']['power']
grid_3_power = results['ct3']['power']
grid_4_power = results['ct4']['power']
grid_5_power = results['ct5']['power']
grid_6_power = results['ct6']['power']
 
grid_1_current = results['ct1']['current']
grid_2_current = results['ct2']['current']
grid_3_current = results['ct3']['current']
grid_4_current = results['ct4']['current']
grid_5_current = results['ct5']['current']
grid_6_current = results['ct6']['current']

FILTER_THRESHOLD = 5 # in Watts
if grid_1_power) < FILTER_THRESHOLD:
    grid_1_power = 0
    grid_1_current = 0

if grid_2_power) < FILTER_THRESHOLD:
    grid_2_power = 0
    grid_2_current = 0

# ... and so on ... for the rest of the CTs
```

emersonsc, 2022-12-10:
I'll have to try and see what I get. I've got a couple reporting negative wattage values which is not playing nice with MQTT & home assistant

closed 2022-12-14 by emersonsc