David00/rpi-power-monitor #81: Voltage Tuning

atzimot, 2022-12-08:
I'm tring to do the voltage tuning and i have in terminal values below 0

GRID_VOLTAGE = 235
AC_TRANSFORMER_OUTPUT_VOLTAGE = 8.94

What do I have to do ?
I have done all steps from the wiki

[image]

David00, 2022-12-08:
Hi atzimot, is this from a DIY board or the factory assembled PCB?

Can you run and attach the output from a debug plot?
https://github.com/David00/rpi-power-monitor/wiki/Software-1.-Overview#debug-mode

Also, make sure that the software is not running in the background with `sudo systemctl status power-monitor`, and if it is running, stop it with `sudo systemctl stop power-monitor`.

atzimot, 2022-12-09:
Hi David,
  Is a factory  assembled PCB.
  The service was stopped when I have done the tuning.
  I have attached the debug plot
 I don't have any ct attached right now to the PBC

[image]

atzimot, 2022-12-09:
also I have used your prebuild raspbbery image

emersonsc, 2022-12-09:
Have you only done the initial voltage check and input for the voltage, or have you changed other values in the config.py? 

Also, after you changed the values in config.py, did you run the following:
 cd ~/rpi_power_monitor
 pip install .

lastly, can you copy and paste the contents of your config.py

David00, 2022-12-09:
@ atzimot, was your AC power supply plugged into the PCB when you ran the debug plot above? The PCB looks like it is not seeing much come in on the AC voltage input.  Can you share a photo of the label on the AC power supply that you're using?

atzimot, 2022-12-10:
I have changed the AC power supply and now the debug is like this.
Now the AC has the value 9.54V 

[image]

Here is the AC data

[image]

But the values are the same under 0

[image]

David00, 2022-12-10:
The problem appears to be that your power supply puts out 9V DC, not 9V AC.  It needs to be an AC output.

atzimot, 2022-12-10:
Thanks for the info
What is the minimum consumtion for this transformer ?
I'm asking becuase I have found transformers but they have different consumtions 100, 200, 400 mA

emersonsc, 2022-12-10:
Jameco Reliapro 9V AC to AC
Output: 9V AC, 500mA

David00, 2022-12-10:
> What is the minimum consumtion for this transformer ?

Any of the ones you've listed should work fine. There is very little consumption from the AC transformer input.

closed 2023-01-10 by David00