David00/rpi-power-monitor #83: power_monitor.py not importing from config.py

jmadden91, 2022-12-22:
Hi David, 

I'm having an issue since updating to release 0.2. Power_monitor.py does not seem to be importing from config.py, but at the same time I can't work out where the 125v default seems to be coming from. As you can see below, I've updated my config.py GRID_VOLTAGE to 1000v, but the program still shows 125v.

[image]

[image]

If I make the adjustment in line 35 of power_monitor.py it works correctly, however I assume its also not importing any of the calibration data either..

David00, 2022-12-22:
Did you re-install the module with pip after updating config.py? I know it's not intuitive, but it currently needs to be reinstalled after every modification to config.py for the changes to take effect:

    cd ~/rpi_power_monitor
    pip install .

I consider this a functional issue and have plans to resolve it via issue #75.  

I also just updated the "Calibrating for Accuracy" page to reflect this step, as it was missing in step 6 of the Final Accuracy Calibration part.

jmadden91, 2022-12-22:
Ah, thats fixed it. Perfect, thanks for your help!

closed 2022-12-22 by jmadden91

Cube3SGM, 2023-02-07:
Hello,  I have a similar problem related to the config.py when setting up for the calibration.

After modifying the config.py (236v and 10.1v) I am unable to successfully complete "pip install"  I receive the message
Defaulting to user installation because normal site-packages is not writable
ERROR: You must give at least one requirement to install (see "pip help install")

Any help appreciated, Regards Simon