David00/rpi-power-monitor #84: Rebooted Pi - Negative values , 0 values

ejg3855, 2023-01-09:
I had to reboot my pi and the power for the daughter board, and since then I have lost all semblance of reasonable data. This is not the first time I have unplugged the device but its the first time its thrown bizarre data.

[image]

David00, 2023-01-09:
Can you generate a debug plot and share a screenshot of it here?

https://github.com/David00/rpi-power-monitor/wiki/Software-1.-Overview#debug-mode

David00, 2023-01-10:
Thanks. Two things - the sample rate was a little bit low, which suggests you might have grabbed the debug plot with the power monitor still running in the background, OR, your Pi updated over the reboot and bumped up your kernel version.

The second thing is that I don't see any data coming in from any CTs. Only from the AC voltage input.   

Can you do a `pip freeze | grep spidev`, and see if it shows `spidev==3.6`? If it's not 3.6, you'll need to update with `pip install spidev==3.6`.

This won't solve the issue of no data, but it will help the sample rate, at least.  For the issue of no data, can you double-check your CT wiring and connections?

ejg3855, 2023-01-16:
havent done the above, but i didnt plug the cable back in. I have data but its negative.

David00, 2023-01-21:
I'm not sure I understand. Are you saying that the problem was that you did not plug the cable back in?

ejg3855, 2023-01-21:
That was the lack of data problem, I am not sure why all the values are
suddenly negative tho.

closed 2023-03-25 by David00