David00/rpi-power-monitor #85: Looking for influxdb files in Docker

bmist45, 2023-01-20:
**My rpi_power_monitor is up and running fine!  I want to explore the influx database with an eye towards setting up a backup routine to save the db on my Synology Diskstation.  Im having a problem accessing the influx database on my Pi.  Here's a sample of what I'm seeing:**

Linux rpipowermonitor 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l......
.............
pi@rpipowermonitor:~/rpi_power_monitor/rpi_power_monitor $ python3 power_monitor                                                                                        .py terminal
2023-01-20 12:05:14 : ... Starting program in terminal mode
2023-01-20 12:05:14 : ... DB initialized.
2023-01-20 12:05:14 : ... Starting Raspberry Pi Power Monitor
2023-01-20 12:05:14 : Press Ctrl-c to quit...
2023-01-20 12:05:17 :
+---------+---------+---------+-------+---------+--------+---------+
|         |   ct1   |   ct2   |  ct3  |   ct4   |  ct5   |   ct6   |
+---------+---------+---------+-------+---------+--------+---------+
|  Watts  | 815.736 | 427.683 |  2.44 | 352.578 | -0.603 | 147.872 |
| Current |  10.076 |  4.762  | 0.107 |  4.865  | 0.048  |  1.462  |
|   P.F.  |  0.674  |  0.745  | 0.183 |  0.577  | -0.102 |  0.814  |
| Voltage |  120.09 |         |       |         |        |         |
+---------+---------+---------+-------+---------+--------+---------+
>

**When i try to check on the docker status I get this:**

pi@rpipowermonitor:~ $ docker ps
-bash: docker: command not found
pi@rpipowermonitor:~ $

**I also don't see anything in the /opt/  directory.  and when I try this I don't get anything:**
pi@rpipowermonitor:~ $ docker exec -it influx influx
-bash: docker: command not found

**What am I missing?**

David00, 2023-01-20:
I recently changed my project to not use Docker at all - which reminds me, I need to update the Wiki's part on the database info!

Influx is now installed natively within the OS instead of running as a container via Docker. To access it, simply use:

`influx -database power_monitor -precision rfc3339`

I will update the Wiki ASAP. Thanks for bringing this to my attention!

David00, 2023-01-20:
The [Database Info](https://github.com/David00/rpi-power-monitor/wiki/Software-5.-All-About-the-Database) wiki page has been updated.

JeromeThiv, 2023-01-20:
Why did you choose to not use docker ?

Jérôme Thivierge

bmist45, 2023-01-20:
Thanks for the quick response David!

David00, 2023-01-20:
Anytime, @bmist45!

@JeromeThiv, I stopped using Docker because of difficulties with my image build process using Qemu. I could not get the Docker daemon to run when emulating the image with Qemu, so I could not pull the InfluxDB and Grafana images. My first custom OS image (v0.1.0) included Docker, but it relied on a startup script to download the Grafana and InfluxDB images if they didn't already exist... which assumes a reliable and moderate speed internet connection.  

So, long story short, removing Docker allows anyone to download my custom v0.2.0 image, flash it to a microSD, and start using my project without ever needing access to the internet from the Pi itself.

closed 2023-01-20 by David00