David00/rpi-power-monitor #87: v0.1.0 to v0.2.0 upgrade path

David00, 2023-02-05:
TBD

David00, 2023-02-05:
Here is the migration process. This is broken down into two main steps, across two comments (this one, and the next one).  This comment/step is for backing up your existing power monitor data.  The next comment is for executing the upgrade, installing native InfluxDB, and running the import.

## Influx Backup Procedure
On the existing power monitor:

1. Stop the power monitor service and the Grafana container:

    ```
    sudo systemctl stop power-monitor.service
    docker stop grafana
    ```

1. Ensure that the InfluxDB Docker container has a mapped folder to the Pi's local storage:

    ```
    docker container inspect -f '{{ .Mounts }}' influx
    ```

    
    Sample output:
    ```
    [{bind  /opt/influxdb /var/lib/influxdb   true rprivate}]
    ```

    (Where /opt/influxdb inside the container is mapped to /var/lib/influxdb on your Pi. If your output matches the above, you're good to go.)
    

1. Start the backup:

    ```
    docker exec -it influx influxd backup -portable -database power_monitor /var/lib/influxdb/backups/powermon_migrate
    ```

    Monitor the output for errors.  If the backup does not complete properly, see below. If the backup finishes successfully, move on to step 4.

    ### 3B. Backup Failure
    If you see any mention of `download shard # failed copy backup to file`, or if the backup fails, this is potentially due to a bug in the early 1.8.x versions of Influx.  To try to get around it, you can remove the InfluxDB container and recreate it with version 1.8.10 where the issue is supposedly fixed.

    ### Ensure that step 2 showed a mapped folder - otherwise, you'll lose the data in your InfluxDB container if you run the next commands!
    
    >WARNING: Only run the following commands if the backup failed AND you have a mapped folder according to step 2.
    >If your backup was successful, do not run this, and move on to step 4.
    If your backup failed but you do not have a mapped folder, see [this question and answer](https://stackoverflow.com/questions/28302178/how-can-i-add-a-volume-to-an-existing-docker-container) for how to add a mapped folder to an existing container.

    ```
    !!! ONLY RUN THESE IF YOUR BACKUP FAILED AND YOU HAVE A MAPPED FOLDER !!!

    docker stop influx
    docker rm influx
    docker run -d --restart always --name influx -p 8086:8086 -p 8088:8088 -v /opt/influxdb:/var/lib/influxdb influxdb:1.8.10

    Then, try to rerun step 3.
    ```
    
1. Create a tar archive of the backup files as root:

    ```
    sudo su
    tar -C /opt/influxdb/backups -cvzf /home/pi/powermon_migrate.tar.gz powermon_migrate
    exit
    ```

1. Optionally, export the tar file from the Pi so that you can keep it safe. If you have another Raspberry Pi on your local network, you can use SCP to get the file onto the other Pi (replacing your IP addresses as necessary). 

    ```
    sudo scp ~/powermon_migrate.tar.gz pi@192.168.0.30:~/
    ```
    ... where `192.168.0.30` would be the IP address of your second Raspberry Pi, and the password prompt is your login/ssh password for the other Pi.
    
    Other options for moving files off your Pi are WinSCP, FileZilla, or good old flash drives.  If you have another computer on your local network that has a Linux shell (like WSL on Windows), you can use SCP to copy _from_ the Raspberry Pi to your Linux environment, by just changing the order of the above command.  So, from your other Linux computer, run the following, and change the IP address below to match what your existing Pi is assigned to:

    scp pi@192.168.10.40:~/powermon_migrate.tar.gz ~/

The password prompt (assuming you're not using SSH keys) will be for the login of your Raspberry Pi, which might still be the default value of `raspberry`.

---

This concludes the Docker container InfluxDB backup - you should now have a file named powermon_migrate.tar.gz inside of ~/.

David00, 2023-02-05:
I am currently testing the process on my existing power monitor deployment. I will edit this comment with the upgrade and migration procedure when finished.

David00, 2023-02-06:
Reserved to continue from above

5ft24dave, 2023-02-06:
db migrate is failing.  did the backup as above, everything going smooth, estracts then bombs out

`Backup found in: /home/pi/powermon_migrate.tar.gz

Extracting backup to /opt/influxdb/power-monitor-influx-import. This can take up to 20 minutes (or more). Please wait. You will not see any output while it is extracting.
Backup extracted in 35.9306263923645 seconds.
Traceback (most recent call last):
  File "/home/pi/db_migrate.py", line 573, in <module>
    start()
  File "/home/pi/db_migrate.py", line 504, in start
    validate_rps()
  File "/home/pi/db_migrate.py", line 520, in validate_rps
    existing_rps = client.get_list_retention_policies()
  File "/usr/local/lib/python3.9/dist-packages/influxdb/client.py", line 794, in get_list_retention_policies
    rsp = self.query(
  File "/usr/local/lib/python3.9/dist-packages/influxdb/client.py", line 458, in query
    results = [
  File "/usr/local/lib/python3.9/dist-packages/influxdb/client.py", line 459, in <listcomp>
    ResultSet(result, raise_errors=raise_errors)
  File "/usr/local/lib/python3.9/dist-packages/influxdb/resultset.py", line 25, in __init__
    raise InfluxDBClientError(self.error)
influxdb.exceptions.InfluxDBClientError: database not found: power_monitor
`

5ft24dave, 2023-02-07:
> Thanks for trying - I just updated `db_migrate.py` to fix this.  You
> can try again from step 3:

Working perfect now!  Data in my DB started June 4, 2022. Took 4 hours and 20 minutes!

I'm running a 500GB m.2 drive in a USB3.1 carrier.   Boots right up!  Running flawlessly.  Thank you

Get BlueMail for Android

David00, 2023-02-07:
Excellent.  So now that you have the right data, you'll want to upgrade the power monitor software itself, and re-enable the service file.  I haven't yet decided on the best way to do it, but one way would be to do the following, for the v0.2.0 (November 2022) release:

```
cd ~/
wget https://github.com/David00/rpi-power-monitor/archive/refs/tags/v0.2.0.tar.gz
tar -xzf v0.2.0.tar.gz
cd rpi-power-monitor-0.2.0/
pip install .
```
Next, you'll need to manually update the new `config.py` file in `~/rpi-power-monitor-0.2.0/rpi_power_monitor/config.py`.

The important thing to note is that the CT channel labels changed between versions, so as you're updating the settings, everything from ct0 in the old file should be applied to ct1 in the new file. Specifically, just these four variables:

* `ct_phase_correction` (name changed in the new version to) `CT_PHASE_CORRECTION`
* `accuracy_calibration` (name changed in the new version to)  `ACCURACY_CALIBRATION`
* GRID_VOLTAGE
* AC_TRANSFORMER_OUTPUT_VOLTAGE

Finally, you'll want to update the service file to point to the new version of the code.

`sudo pico /etc/systemd/system/power-monitor.service`

If you followed the steps verbatim above, then you can replace the `ExecStart` line with the following:

`ExecStart=/usr/bin/python3 /home/pi/rpi-power-monitor-0.2.0/rpi_power_monitor/power_monitor.py`

Then, run the following to apply the changes, start the service, and check its status:

```
sudo systemctl daemon-reload
sudo systemctl start power-monitor.service
sudo systemctl status power-monitor.service
```

That's it! (Well, except for Grafana still running in a Docker container), but that's not too big of a deal.  Once you get your dashboards manually backed up, it's easy to remove.

## Updating Grafana

If you have customized dashboards, you'll need to back them up manually, or else they will be removed. The default dashboards to go with this project area available on the [Dashboard Wiki page](https://github.com/David00/rpi-power-monitor/wiki/Software-3.-Building-your-Dashboard). 

To backup your dashboards, go into each dashboard, click the settings wheel at the top, and then find the JSON model button on the left. Copy the entire text contents of the JSON model out and save it to a text file for future import.

1. Once you've stashed your dashboard JSON models in a safe place, or chosen to lose them, stop the Grafana container, and then remove it:

```
docker stop grafana
docker rm grafana
```

2. Native Grafana Installlation

```
wget -q -O - https://packages.grafana.com/gpg.key | sudo apt-key add -
echo "deb https://packages.grafana.com/oss/deb stable main" | sudo tee -a /etc/apt/sources.list.d/grafana.list
sudo apt-get update
sudo apt-get install -y grafana

# Then, remove docker with:
sudo apt purge docker
```

## Old Data Removal

The final thing to do, which is completely optional, is to remove your old high resolution power monitor data. As long as the migration script went well, all of your old high-res data has been downsampled and stored into new measurements, meaning your old data is safe to remove from Influx. You should retain the original backup archive just in case, though!

To remove your old data, we'll go into the Influx CLI and execute a few drop statements. The following will remove all of the high resolution data that is older than 30 days from the time you execute the command:

```
influx -precision rfc3339 -database power_monitor
DELETE from "rp_raw" WHERE time <= now() - 30d
DELETE from "home_load" WHERE time <= now() - 30d
DELETE from "net" WHERE time <= now() - 30d
DELETE from "solar" WHERE time <= now() - 30d
DELETE from "voltages" where time <= now() - 30d
exit
```

This concludes the upgrade process to get your environment fairly close to the v0.2.0 image.  However, I am right on the brink of releasing v0.3.0, which contains a lot of improvements over v0.2.0, so stay tuned for that one (a beta is already out, but there are still a few issues I need to fix). 

The upgrade path from v0.2.0 to v0.3.0 will be much easier.

5ft24dave, 2023-02-11:
I can confirm that 0.2.0 works without issues under the V6.1.11 Kernel on the pi

bobstanl, 2023-02-16:
Dashboard Changes?

Hi David
Have had problems bringing up the USB-SSD but it is now working and I can SSH
Decided to keep the uSD that contains the old system as it was and start fresh on the SSD.
The new image is based on "Raspberry-Pi-OS-Lite_rpi_power_monitor-0.2.0+release"
Aside: I was able to advance configure using the Imager "gear" on your image to change names, passwords, wifi, etc. Can forward the way they did it if you are interested.
The db_migrate.py  seemed to run OK last night on my old data backup from the uSD card.
Both influx and grafana service are reported running.
Power-monitor is not running yet as RPi in not back in outside box yet.
Grafana is working at correct IP and acted new, requiring new password.

Using the dashboard that existed, I wanted to check the old data in grafana but all I get on any time period is No Data!

The influxdb "save and test" shows "datasource is working. 19 measurements found".

Do the dashboards need to be changed?

I did not see that in the above.

David00, 2023-02-16:
> Aside: I was able to advance configure using the Imager "gear" on your image to change names, passwords, wifi, etc. Can forward the way they did it if you are interested.

Excellent. I'm not sure that I can implement the credential and Wi-Fi customizations into the image build process, but sure, I'd be curious to see!

The `db_migrate.py` script moves all the high resolution data (which the dashboard looks at by default) into the downsampled measurements that the dashboard does not yet look at.  The script does not carry over any of the high resolution data, which is why the dashboards appear to be empty.

All of your old data should be in the database, but you'll have to add a new panel to the dashboard that looks at the downsampled measurements.  In the "show and tell" discussion for dashboards, I've shared a panel that shows the last 60 days worth of production and consumption data. You should be able to deploy the panel I've shared in [this comment](https://github.com/David00/rpi-power-monitor/discussions/77#discussioncomment-4239861), but you don't need to do any of the `SELECT` commands that are shown.  You should just be able to set your timezone variable as described and then import the panel JSON.

Once you start the power monitor service, you should start to see data populate again in the default dashboard.

bobstanl, 2023-02-16:
I still don't understand.

My ideal is that I can query any of the old data (that hasn't been deleted by a uSD failue) for any desired day or number of hours.

A reason for displaying the data in a similar way is for debugging. Is this the same power and currents that I got before? If not, is it software? A hardware fault? Did I have the same current spike when the spa pump turned on last year or is todays turn-on spike higher? Do I have a new pump problem?

I remember my neighbor asking a question about solar output, so I could show him my solar for a day in June and compare it to a day in the winter.

I personally prefer to use bigger SSD's and keep all historical data rather than throw it away. This is why I am so annoyed by failing uSD cards.

You said: "The script does not carry over any of the high resolution data, which is why the dashboards appear to be empty."

I have been meaning to ask what is meant by "high resolution data". Will I still see the same current excursions that I could see before?

Can a dashboard be created to see the old data as before?

David00, 2023-02-16:
> My ideal is that I can query any of the old data (that hasn't been deleted by a uSD failue) for any desired day or number of hours.

The data has been downsampled by the `db_migrate.py` script into 5 minute intervals. (Your backup will still contain all your original data though, untouched). By "high resolution" I am referring to the data that goes into the database once every second or two. The new docs cover this whole subject a little bit (but after a quick review, there is some more info I need to add to a few sections on this page):

https://david00.github.io/rpi-power-monitor/docs/v0.3.0-beta/database-info.html#overview

> Will I still see the same current excursions that I could see before?

Yes, when you start the power monitor again, the dashboard that you're used to should continue to function just like before, but the one thing that changed in v0.3.0 is that the power monitor software now configures InfluxDB to automatically downsample the high resolution data into the 5 minute averages.  The downsampling happens in the background and is basically a completely different copy of the high resolution data.

For example... with updates coming in every second, there swould be about 300 data points every 5 minutes (per channel!) in the "high res" data.  InfluxDB will now automatically average all 300 of those points together and store the resulting average in a new retention policy and measurement.  The averaged data will be stored forever, so you can always query it in the future.  The "high res" data will be deleted after 30 days, which means, the home power dashboard (as currently configured) can only go back 30 days.  

The retention policies (RPs) are kind of like different folders that hold data inside individual measurements.  The "high res" data goes into the default RP named `autogen`.  So, the provided dashboard queries against the `autogen` RP.

If you look at one of the continuous query (CQ) creations, you can get an idea of what it's doing:

```
CREATE CONTINUOUS QUERY cq_home_power_5m ON power_monitor BEGIN SELECT mean(power) AS power, mean(current) AS current INTO power_monitor.rp_5min.home_load_5m FROM power_monitor.autogen.home_load GROUP BY time(5m) END
```

* `SELECT mean(power) AS power, mean(current) AS current` ... `FROM power_monitor.autogen.home_load`
* `INTO power_monitor.rp_5min.home_load_5m`
* `GROUP BY time(5m)`

I've highlighted the key points above.  So, for each `GROUP` of `5m`, it's taking the `mean` power and `mean` current, `FROM` the database named `power_monitor`, RP named `autogen`, and measurement named `home_load`, and storing the result `INTO` the database named `power_monitor`, inside the RP named `rp_5min`, into the measurement named `home_load_5m`.

bobstanl, 2023-02-16:
My problems are more existential. We cannot assume there is any valid data in the database. There is a lot of disk space being taken by something, but is it valid data?

Tried loading in the Overview-Panel.JSON as you suggested. Seems to load but the dashboard shows no panel. When I look at the json model version of the importe dashboard, it has only 48 lines, where the original file has 198 lines. No obvious error messages.

Remember this database came from the faulty uSD card that went through a series of operations. On the new V0.1 uSD, the system ran and I could, by randomly picking dates, see that some days had data and some did not. You said there was no easy way to look for gaps "Missing is tough - without writing a custom python script to compare the data, I don't think there's an easy way.

Problem now is my original technique of changing dates on the "Home Power Dashboard" no longer works because of the v0.2 changes.

I know Jan 23 only has a few days of data, and I think a bit of Dec 22 is gone. Jan 22 is gone, but I think other dates might have data. It was unusual that the "db_migrate.py" marched through EVERY day, since late Nov 21, with no errors. Did it convert anything on those days that the old "Home Power Dashboard" said No Data?

If there any way to find out if there is valid data in some date, say in July 22?

bobstanl, 2023-02-16:
Potential excess logging in db_migrate.py?

Sorry to be such a helpless pain in influxdb, all the small tricks I know so far have not shown me any results in query.

Looking at the big picture, how much data is on this new SSD drive with "Raspberry-Pi-OS-Lite_rpi_power_monitor-0.2.0+release" and my 2.1G powermon_migrate.tar.gz?

The whole du seems excessive at 13G, so I spent some time examining the drive. Here are the results:

List of sizes of top directories of root
Notes 
/home has ~2G, this is my PM data backup
/opt has ~2G, presumably has this data imported
What is on /var???
root@powermon:/# du -h --max-depth=1
2.0G	./opt
56K	./tmp
15M	./root
1.2M	./run
0	./sys
2.0G	./home
4.1M	./etc
16K	./lost+found
0	./proc
50M	./boot
0	./dev
4.0K	./mnt
4.0K	./media
4.0K	./srv
7.3G	./var
1.7G	./usr
13G	.

Note /var has very large log
root@powermon:/var# du -h --max-depth=1
133M	./lib
4.0K	./opt
36K	./tmp
6.9G	./log
24K	./backups
16K	./spool
4.0K	./mail
275M	./cache
4.0K	./local
12K	./www
7.3G	.

Note usr has large lib, maybe OK?
root@powermon:/usr# du -h --max-depth=1
735M	./lib
4.0K	./games
4.0K	./src
115M	./sbin
244M	./bin
103M	./local
29M	./include
1.6M	./libexec
490M	./share
1.7G	.

/var/log ts huge!
root@powermon:/var/log# du -h --max-depth=3
8.0K	./nginx
3.2G	./journal/5afa955048e549beaed1b5c169d97e04
3.2G	./journal
4.0K	./private
4.0K	./influxdb
4.0K	./runit/ssh
8.0K	./runit
124K	./grafana
156K	./apt
6.9G	.

daemon.log is excessive 1.8G, does this fold into system logs, maybe several times?
root@powermon:/var/log# ls -alh
total 3.8G
drwxr-xr-x   9 root     root            4.0K Sep 21 17:23 .
drwxr-xr-x  12 root     root            4.0K Nov 11 23:50 ..
-rw-r--r--   1 root     root             721 Feb 14 17:44 alternatives.log
drwxr-xr-x   2 root     root            4.0K Feb 14 18:07 apt
-rw-r-----   1 root     adm              32K Feb 15 23:17 auth.log
-rw-r--r--   1 root     root               0 Sep 21 17:23 bootstrap.log
-rw-rw----   1 root     utmp               0 Sep 21 17:23 btmp
-rw-r-----   1 root     adm             1.9G Feb 15 23:35 daemon.log
-rw-r-----   1 root     adm             7.0K Feb 15 01:12 debug
-rw-r--r--   1 root     root             83K Feb 14 18:08 dpkg.log
-rw-r--r--   1 root     root             24K Nov 11 23:53 faillog
drwxr-xr-x   2 grafana  grafana         4.0K Feb 15 00:04 grafana
drwxr-xr-x   2 influxdb influxdb        4.0K Oct 11  2021 influxdb
drwxr-sr-x+  3 root     systemd-journal 4.0K Sep 21 17:23 journal
-rw-r-----   1 root     adm             228K Feb 15 01:12 kern.log
-rw-rw-r--   1 root     utmp            286K Feb 15 16:03 lastlog
-rw-r-----   1 root     adm             225K Feb 15 01:12 messages
drwxr-xr-x   2 root     adm             4.0K Feb 15 00:00 nginx
drwx------   2 root     root            4.0K Sep 21 17:23 private
drwxr-xr-x   3 root     root            4.0K Sep 21 17:06 runit
-rw-r-----   1 root     adm             1.9G Feb 15 23:35 syslog
-rw-r-----   1 root     adm             2.7K Feb 15 01:12 user.log
-rw-rw-r--   1 root     utmp             16K Feb 15 16:03 wtmp

Here are a few lines from the end of the daemon.log file. Maybe help with debugging too much logging in db_migrate.py?

Feb 15 23:55:00 powermon influxd-systemd-start.sh[680]: ts=2023-02-16T07:55:00.305666Z lvl=info msg="Executing query" log_id=0g15OJ20000 service=query query="SELECT mean(power) AS power, mean(current) AS current INTO power_monitor.rp_5min.net_5m FROM power_monitor.autogen.net WHERE time >= '2023-02-16T07:50:00Z' AND time < '2023-02-16T07:55:00Z' GROUP BY time(5m)"
Feb 15 23:55:00 powermon influxd-systemd-start.sh[680]: ts=2023-02-16T07:55:00.306292Z lvl=info msg="Finished continuous query" log_id=0g15OJ20000 service=continuous_querier trace_id=0g2JMLKG000 op_name=continuous_querier_execute name=cq_net_power_5m db_instance=power_monitor written=0 start=2023-02-16T07:50:00.000000Z end=2023-02-16T07:55:00.000000Z duration=0ms
Feb 15 23:55:00 powermon influxd-systemd-start.sh[680]: ts=2023-02-16T07:55:00.306362Z lvl=info msg="Continuous query execution (end)" log_id=0g15OJ20000 service=continuous_querier trace_id=0g2JMLKG000 op_name=continuous_querier_execute op_event=end op_elapsed=0.904ms
Feb 15 23:55:00 powermon influxd-systemd-start.sh[680]: ts=2023-02-16T07:55:00.306618Z lvl=info msg="Continuous query execution (start)" log_id=0g15OJ20000 service=continuous_querier trace_id=0g2JMLKW000 op_name=continuous_querier_execute op_event=start
Feb 15 23:55:00 powermon influxd-systemd-start.sh[680]: ts=2023-02-16T07:55:00.306642Z lvl=info msg="Executing continuous query" log_id=0g15OJ20000 service=continuous_querier trace_id=0g2JMLKW000 op_name=continuous_querier_execute name=cq_solar_5m db_instance=power_monitor start=2023-02-16T07:50:00.000000Z end=2023-02-16T07:55:00.000000Z
Feb 15 23:55:00 powermon influxd-systemd-start.sh[680]: ts=2023-02-16T07:55:00.306833Z lvl=info msg="Executing query" log_id=0g15OJ20000 service=query query="SELECT mean(real_power) AS power, mean(current) AS current INTO power_monitor.rp_5min.solar_5m FROM power_monitor.autogen.solar WHERE time >= '2023-02-16T07:50:00Z' AND time < '2023-02-16T07:55:00Z' GROUP BY time(5m)"
Feb 15 23:55:00 powermon influxd-systemd-start.sh[680]: ts=2023-02-16T07:55:00.307316Z lvl=info msg="Finished continuous query" log_id=0g15OJ20000 service=continuous_querier trace_id=0g2JMLKW000 op_name=continuous_querier_execute name=cq_solar_5m db_instance=power_monitor written=0 start=2023-02-16T07:50:00.000000Z end=2023-02-16T07:55:00.000000Z duration=0ms
Feb 15 23:55:00 powermon influxd-systemd-start.sh[680]: ts=2023-02-16T07:55:00.307352Z lvl=info msg="Continuous query execution (end)" log_id=0g15OJ20000 service=continuous_querier trace_id=0g2JMLKW000 op_name=continuous_querier_execute op_event=end op_elapsed=0.740ms
(END)

David00, 2023-02-16:
> Tried loading in the Overview-Panel.JSON as you suggested. Seems to load but the dashboard shows no panel. When I look at the json model version of the importe dashboard, it has only 48 lines, where the original file has 198 lines. No obvious error messages.

That sounds right - the Overview-Panel.json is just a single panel that goes into a dashboard.

> Problem now is my original technique of changing dates on the "Home Power Dashboard" no longer works because of the v0.2 changes.

I was concerned with this impacting this functionality, but I think it can be reasonably solved with an update to the Grafana dashboard.  You can still technically pull data by changing the date in Grafana, but only for the previous 30 days.  If you look at a date beyond that, the _existing_ dashboard won't show any data, because the data gets downsampled and stored elsewhere permanently, and the original data gets deleted as it "ages out".

If you duplicate the panels and edit the queries in each one, and change the query to use the new retention policy that holds the downsampled data, your technique of changing the dates in Grafana will work on the edited panels.  The only difference is the data it's pulling is "lower resolution", and by that I simply mean each data point represents a 5 minute average, and not a 1 or 2 second average.

Here's how you can edit a panel to tell it to query from the down sampled data:

[image]

By making that change, you can then query as far back as your data goes AND also query over long intervals, like several years at a time, if you wanted to. It is not possible to do such a long-term query on the high-resolution data, so there's actually a ton of functionality unlocked by using the down-sampled data set.  For example, you can compare monthly power output year over year, or even track the annual costs of your EV charger, etc.

You can even override the time frame for a single panel so that the date range selector doesn't interfere with what it shows:

[image]

---

>It was unusual that the "db_migrate.py" marched through EVERY day, since late Nov 21, with no errors. Did it convert anything on those days that the old "Home Power Dashboard" said No Data?
>
>If there any way to find out if there is valid data in some date, say in July 22?

The db_migrate script simply stepped through every day, issued a query for each day, and inserted the result into the database. It did not make any checks to see if the data was legitimate or even existed.

You can query manually in InfluxDB to see if there is data there.  Just remember that if you're looking at the new version of the database, any data beyond 30 days from now is going to be stored in the `rp_5min` retention policy.

I've put a table together here that shows the retention policies and measurement names:

https://david00.github.io/rpi-power-monitor/docs/v0.3.0-beta/database-info.html#structure

Here's a quick example from the influx shell:

```
select mean(power) from rp_5min.home_load_5m where time <= '2022-07-07 00:00:00' and time >= '2022-07-06 00:00:00'
```

---

Excellent diagnosis of the large log files! It does look like the InfluxDB logging can be turned down significantly. We certainly don't need or want detailed results of all the InfluxDB actions, especially on a Pi.  I'll look into that separately.

bobstanl, 2023-02-16:
> I made the following changes to InfluxDB's config file at /etc/influxdb/influxdb.conf:
> [logging]
>
>   level = "warn"
>
> [http]
>   log-enabled = false
>
> This seems to have reduced the verbosity of the logging going to /var/log/daemon.log.
>
> I will include this as the standard config when I build the v0.3.0 OS image.

On this issue, just had a thought. These huge files will go away after some number of days, right!This may only be a potential bug for someone with a smaller uSD and a large data set.
Asides:1. Added explanatory comments and file on RPi Imager Advance Options to  Grafana password frequency and SD Card fail · Issue #36 · David00/rpi-power-monitor

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
Grafana password frequency and SD Card fail · Issue #36 · David00/rpi-po...

Hi David! Grafana (v7.1.3) is bugging me much more with password requests than I remember when first installing ...
 |

 |

 |

2. Am working on a new post: "Stop the bus! Can I get off at 0.1.5?" Should I post it to this thread?

ivynetca, 2023-11-22:
I'm confused about the clean up of old high res data. Your directions above include:
DELETE from "rp_raw" WHERE time <= now() - 30d.
There are no rp_raw measurements in my database, but cts_raw (with 3 field keys) is still there. Is this just a typo?
I havent started the new power_monitor up yet. Which measurements will be generated - rp_raw? 
Thanks - Jeff

David00, 2023-11-22:
> I'm confused about the clean up of old high res data. Your directions above include: DELETE from "rp_raw" WHERE time <= now() - 30d. There are no rp_raw measurements in my database, but cts_raw [...]

Thanks for pointing this out! Yes, I do believe this is a typo, and it should be `DELETE from "raw_cts" WHERE time <= now() - 30d;`. I will update the instructions above.

The measurements (and retention policies) that will be generated in v0.3.0 are documented here, but feel free to follow up if you have further questions :)

https://david00.github.io/rpi-power-monitor/docs/v0.3.0/database-info.html#structure