David00/rpi-power-monitor #96: AC Voltage off

shingle01, 2023-03-20:
This might be a case of but the pre-assembled kit stupid, but living in Australia I found a self build more economical.

I have the 0.2 image rpi image installed.

9VAC supply attached, no CT clamps connected.  I've added the initial mains and transformer voltages to the config and ran:` pip install .`

No AC voltage correction value yet but the readings are 10% off and wildly erratic vs the kill'o'watt output.

[image]

No problems with the 3.3VDC supply.  The Voltage on PIN5 of ADC is a steady 1.89VDC.

Below is a debug output.  Strangely, despite CT1 being discounted I have a trend.

[image]

[image]

When AC supply is disconnected:
[image]

David00, 2023-03-20:
Hi @shingle01, I can tell from the debug graph that something isn't quite right with the bias_v output because all of the channels should be "idling" right around 511 in the plot.  Also, the trace that is showing up for CT1 looks very similar to a capacitor charge/discharge cycle.

I would suggest double-checking everything in the "Bias Voltage Reference" part of [the schematic](https://raw.githubusercontent.com/David00/rpi-power-monitor/docs/_includes/resources/Schematics/latest.png), and confirming the value (and capacitor type) for the filtering capacitors C0 - C3, C4, C5,

shingle01, 2023-03-22:
Thanks for the Info.

I replaced C5 with a polarized cap (store originally gave me non-polarized).  No Luck.

Added some more solder to the legs of U1, FIXED!

Thanks for the help.

[image]

[image]

David00, 2023-03-23:
Ok, great! I'm going to close this now, but feel free to reply in the future.

closed 2023-03-23 by David00