David00/rpi-power-monitor #105: 3.0 update still gives negative wattage readings

emersonsc, 2023-05-28:
I recalibrated everything, made sure all of the cts are installed correctly. My two 60a clamps show negative wattage when the breakers are turned off or when there is no load. My circuit for my air compressor, for example, shows shows -2.7w when the compressor isnt running, but shows 2400w when running. I made sure the calibration was done correctly, and everything reads as it should.

jmadden91, 2023-05-29:
Might not address the true problem, but if its reading correctly when you are drawing current, you could use the watts_cutoff_threshold, to remove this. Eg, setting this to watts_cutoff_threshold=1 will zero it out. [https://david00.github.io/rpi-power-monitor/docs/v0.3.0/configuration.html#current_transformerschannel_](url)

David00, 2023-05-29:
It could be some noise coming in via the sensor wires.  If you disconnect the CT from the PCB, does the -2.7W go away? If so, that would confirm the noise suspicion, and you can try to move the sensor wires further away from any high voltage lines and/or change the path they're ran.

Thanks @jmadden91 for suggesting that. I was going to do the same. Here's the link to the relevant section in the docs:

https://david00.github.io/rpi-power-monitor/docs/v0.3.0/configuration.html#watts_cutoff_threshold

emersonsc, 2023-06-12:
Changing the value to 1 or even 2 didnt change anything, im still getting negative values reported. Its ONLY on the 60a CTs i have. My 20a, 30a, and 100a sensors dont give negative values with there is no load.

[config.txt](https://github.com/David00/rpi-power-monitor/files/11716336/config.txt)

vulture2600, 2023-07-02:
I am also getting this issue, but only when my central AC is on. I have 4 CT's, one on each main leg and one on each sub leg.
When the AC is off, and theres minimal load on the sub leg, it shows positive wattage. When the AC is ON, and theres minimal load on sub leg goes negative.

emersonsc, 2023-09-11:
> I am also getting this issue, but only when my central AC is on. I have 4 CT's, one on each main leg and one on each sub leg. When the AC is off, and theres minimal load on the sub leg, it shows positive wattage. When the AC is ON, and theres minimal load on sub leg goes negative.
> 
> Edit: I'm noticing that the amps is still positive but the watts is negative.

I took an still having this issue with any circuit that has no load. It's very strange. I have tuned all of the clamps per the instructions but still have this issue

David00, 2023-09-11:
My suspicion is still noise coming in via the CT lead, especially because AC units draw a large current (which produces a larger magnetic field).

What value are you seeing reported? In the config.txt file you shared above, I see a `watts_cutoff_theshold` value of 2, which will only take effect if the readings are between -1.99 through 1.99.  If  you see anything outside this window, you'll have to increase the setting for watts_cutoff_threshold, or, you can set `enabled = false` for the channels you're not using.

closed 2024-02-04 by David00