David00/rpi-power-monitor #106: "two_pole" configuration incorrenctly doubles current

cgleggett, 2023-06-05:
When a CT is marked as "two_pole" in the configuration file, ie the CT is metering one leg of a two pole breaker, both the power and current values are doubled. I don't think this is correct - it should be just the power that's doubled.

David00, 2023-06-06:
Hi @cgleggett, thanks for opening this issue for discussion.  

The `two_pole` setting is essentially saying: If you had a sensor on each leg of a North American 240V circuit, what would the total power and the total current look like? 

The `two_pole` setting assumes that the device connected to the 240V circuit distributes the amperage evenly over both legs, which is why the amperage is doubled. This is usually the case for solar inverters and EV chargers.

There may be an opportunity here to improve the docs surrounding the `two_pole` setting.

cgleggett, 2023-06-06:
Right, but if you've say, 20A running through one leg of a 240V circuit with no neutral, like an inverter, you have 20A running through the other. And while the power would be 2 x 120 x 20, the current is still 20A. If you're looking at the current of the circuit in say the grafana plot, the value will be twice what you would expect, though the power is correct.

David00, 2023-06-07:
I see what you mean now! Thanks for explaining further.  I'll have to look into how I can fix this in a way that makes sense for all types of grids.

David00, 2024-02-04:
This has been resolved now - thanks for reporting the issue!

commit f02b9f5c (2024-02-04): Fix #106 by removing doubling of amperage for 2-pole channels
closed 2024-02-04 by David00