David00/rpi-power-monitor #108: V1.2 PCB Resistor Values and Positions

misterjtc, 2023-07-29:
So I bought a DIY kit a really long time ago (April 2021) and I am just getting to putting it together now.

The documentation is a bit confusing as I only have 4X 22 ohm resistors in my kit and therefore am short 2 based on the docs.

The parts list also doesn't show a 10k ohm resistor but its shown in the assembly guide.

Anyways, it looks like instead of the 2X 22 ohm resistors, I ended up with 2X 10 ohm resistors which I have installed in R0 and R1 positions. Is this correct or do I need to get 2X 22 ohm resistors and install them in R0 and R1?

David00, 2023-07-29:
Hey @misterjtc, that's great you're now getting around to putting it together!  Quite a bit has changed since then - for the better!

One thing to note is that the latest v0.3.0 software has the channel numbers renamed from CT0 - 5 to CT1 - 6.  If your order included a paper printout with some instructions to get started, it is likely referring to the old CT0-5 numbering scheme. The [Assembly page in the Wiki](https://github.com/David00/rpi-power-monitor/wiki/Hardware-Assembly) has the latest assembly instructions.

In 2021, certain CT sensors required specific resistor values.  Since then, I have expanded my custom sensor lineup so that they all require the same 22 Ω resistor. This is why the latest docs only mention the 22 Ω resistor.

I checked your order, and the two 100A PZCT-02 sensors each need a 10 Ω resistor.  So, if you installed the two 10 Ω resistors in R0 and R1, you'll want to use the PZCT-02 sensors on channels CT1 and CT2, which correspond to the orange and green pairs in a cat5e cable.

Feel free to reach back out here if you have any more questions about the assembly, or shoot me an email at github at dalbrecht.tech.

misterjtc, 2023-07-29:
Hey @David00, thanks a lot for the quick and detailed reply!

I am glad I asked because I am working on the cat cable right now! This makes everything more clear, thank you.

closed 2023-07-31 by misterjtc