David00/rpi-power-monitor #109: Add updating Grafana data source when using a remote InfluxDB instance

misterjtc, 2023-07-31:
Maybe I missed it in the docs and keeping in mind that I don't know much about Grafana, but I think a comment about updating your datsources in Grafana should be added when using a remote db. It didn't take me too long but I was defintely confused when I finished everything and the dashboard reported no data.

beadon, 2023-08-01:
Could you edit the readme to propose changes ?  That sounds like a good idea.

David00, 2023-08-02:
Thanks @misterjtc! I have been meaning to do this since releasing the new documentation. I will update it today - time permitting.

David00, 2023-08-08:
Hi guys, I just pushed a change that addresses this issue. My CI/CD is going to build and deploy the change to the live site automatically, so would you like to review it before closing this issue?

https://david00.github.io/rpi-power-monitor/docs/v0.3.0/advanced-usage.html#add-remote-datasource-to-grafana

closed 2023-08-15 by David00