David00/rpi-power-monitor #110: RPI Power Monitor Case

misterjtc, 2023-07-31:
I am wondering if anyone has designed a case for this system. If there is something that works, can someone share a link to the design?

If not, maybe I can contribute to the project by designing one.

beadon, 2023-08-01:
I have not yet seen a case.  If you do design it, that would be a nice addition.

misterjtc, 2023-08-01:
Are dimensions of component locations available for the board? This would be helpful but not required as I can take measurements of my unit (DIY) if needed.

beadon, 2023-08-01:
To get exact PCB dimensions you can refer to the Gerber files.  To get a visual, you can use a local PCB inspection tool like Altium Viewer (Autodesk Fusion 360 is another option).  Or you can upload them to a manufacturing partner, like OSH Park.  

These will allow you to get the exact dimensions, and test the fit of your 3d model.

Since raspberrypi cases are so popular, I recommend starting with one as a base, and then extending the capabilities.  You will find that existing cases also permit stacking additional boards outside the case and/or permit access to the GPIO PINs.  These existing cases may be good enough for what you are looking for.

If you find one that works, please let us know !  I'm interested to see the pictures and what you've designed.

misterjtc, 2023-08-01:
Thanks for all of the info @beadon, this is very helpful.

I have an existing case in mind that I just used for a pi with a poe+ hat. 

Using this as a base:
https://www.thingiverse.com/thing:3723561

I am thinking to make something like this:
https://www.thingiverse.com/thing:3724434

I'll post updates when I get some time to work on it.

beadon, 2023-08-02:
Neat , that looks very reasonable.  

I was thinking some pre-built ones would get you to the heights you need.  This (link following) type of case would allow you to just put in screws at higher "height".  Very simple acrylic layers as layer separators :

https://geekworm.com/products/raspberry-pi-cluster-case-with-fan-kit-1-5-layer-multi-layer-acrylic-case?variant=20348313010234  

Saves doing the CAD and 3d-printing work if you don't mind the sides being open.

misterjtc, 2023-08-21:
Where can I find the gerber files? For someone reason I thought I saw them when I looked some weeks ago ago but now there isn't much in the repo. Did something change? or am I just confusing them it with some other location!?

David00, 2023-08-21:
The gerber for the DIY PCB is in the Wiki, on this page:
https://github.com/David00/rpi-power-monitor/wiki/Schematic

The gerber for the factory assembled board is not publicly available at this time.

misterjtc, 2023-08-21:
> The gerber for the DIY PCB is in the Wiki, on this page: https://github.com/David00/rpi-power-monitor/wiki/Schematic
> 
> The gerber for the factory assembled board is not publicly available at this time.

Thanks! Not sure how I missed that...

David00, 2023-08-21:
No problem! :)

misterjtc, 2023-09-10:
So its not ready for prime time, but I thought I would post an update. I have a first case prototype which is actually pretty good for MKI but defintely still needs some work.

I have put provisions for a 40mm fan but am unsure how to power it in the current state as the hat is using all of the gpio pins.

Some pictures are attached.
[image]
[image]
[image]
[image]
[image]

David00, 2023-09-10:
That looks awesome, @misterjtc! I like the hex vents on top and the clean recessed lines around the ports!  The no additional hardware is a huge plus for tinkerers - it's awesome to just print something and use it without needing additional parts. 

The fan power would likely have to be soldered onto the board's 5V and ground pins.  Something like this would allow for a quick disconnect:
https://www.adafruit.com/product/4872

I could also breakout the 5V and ground to a connector or screw terminal in a future PCB revision...

Awesome work!

misterjtc, 2023-10-19:
So I thought I would post a quick update as I’ve been through a few more prototypes and I think I am pretty close to having a design worth sharing.

Some key updates:

1. I am satisfied with the level of difficulty to put the case together. It may not be 100% intuitive but with a little instruction I would consider it easy to put together. I will provide additional details on this at a later time.
2. Fortunately I have an unhealthy Noctua fan addiction so I had a fan and a bunch of cables lying around which I cannibalized (the cables) and soldered directly to the gpio header and it appears to work well. I took the approach mentioned above by @David00 having a quick disconnect.
3. @David00 let me know how you want to share the files? I can submit a pull request to have the STLs hosted within this repo or we can use something like Thingiverse, Printables, etc (or all of the above). The original design is hosted on OnShape and I will share a link to the original design files when I am satisfied with everything.
4. Lastly, it would be great to have some volunteers to print the case and see how they like it. I am looking for feedback!

[image]
[image]

David00, 2023-10-19:
Hi @misterjtc, that looks great - I would love to dust my 3D printer off and give it a try!

You're certainly welcome to submit a PR with the STLs.  I'm thinking of a new branch simply named `cases`, with a structure like so:

```
-- diy-kit
    |__ misterjtc/
        |__ README.md <this should contain any notes about assembly, fan specs, print instructions/recommendations, build pictures, etc.>
        |__ images/
            |__<any photos of the case that you want to have live in the repo. You might have to push the photos first, then go back to edit the README.md with the links to them so they can be displayed in-page (or I can make the edits if needed)
        |__ STLs/
            |__ <your STLs for this case>

-- factory-assembled
    |__ none yet, so this would just be a placeholder for now.
```

I think this structure would be ideal to keep potential future contributions organized and scalable, but also open to other suggestions!

misterjtc, 2023-10-20:
Sounds good to me. I'll work on it.

In the meantime, I've attached a zip file here with STLs of the most current design for printing.
[PiPower 4 Bottom V25.zip](https://github.com/David00/rpi-power-monitor/files/13049185/PiPower.4.Bottom.V25.zip)

The design is available on OnShape here:
[Assembly](https://cad.onshape.com/documents/86f7bd557cef8cbd98689106/w/b6814c3843a5479e1823c6cc/e/d6480fd9a7b78243a9c79649?renderMode=0&uiState=6531defd208a7747faad5f97)
[Top and Cap](https://cad.onshape.com/documents/984496662072eb7b5d97f476/w/61a532e86ea82be985f73f29/e/7c35e9621c646a0654b52c63?renderMode=0&uiState=6531de22208a7747faad5e01)
[Bottom](https://cad.onshape.com/documents/33e3c4ab5cd6d158e194e49c/w/7adae68ffe3d3acd19b3bf99/e/db320f86071afb8205ab55bf?renderMode=0&uiState=6531de83f45dd6138c56706e)