David00/rpi-power-monitor #112: bad GPG key for grafana repo

FreydNot, 2023-09-02:
I just did a new install using the pre-built disk image.  `apt update` gives a GPG error  for Grafana.

This issue is solved with the procedure at their webpage
https://packages.grafana.com/oss/
```
# mkdir -p /etc/apt/keyrings/
# wget -q -O - https://apt.grafana.com/gpg.key | gpg --dearmor > /etc/apt/keyrings/grafana.gpg
# echo "deb [signed-by=/etc/apt/keyrings/grafana.gpg] https://apt.grafana.com stable main" | tee /etc/apt/sources.list.d/grafana.list
```

David00, 2023-09-10:
Thank you for letting me know and for providing the procedure along with a link. I'll leave this open for a while until I build the next OS image.

David00, 2024-02-04:
This has been added to my OS build process and will be resolved in the next OS release. Thanks again.

closed 2024-02-04 by David00