David00/rpi-power-monitor #115: Version 1.1 PCB still OK to use

gaz99gh, 2023-10-28:
I have one of the old V1.1 boards that I am going to setup for my nephew and want to know if the circuit has changed since the original boards were designed.

I saw a mention that the channel numbering has changed so just need to know if there is anything I need to do to identify the board revision I have.

Is there circuit diagram of the current version available ??

David00, 2023-11-02:
Hi @gaz99gh, the circuit has not changed much at all. The original v1.1 board had the "L1" slot on the PCB for a future inductor, but this turned out to not be necessary, and a wire link was instead supplied for this slot.  In the v1.2 PCB, this L1 component was removed.

There is a new factory assembled PCB which is slightly different, but the overall circuit is the same.  The CT connectors have been changed to use more standard screw terminals, and there's a stable reference voltage for the ADC on the factory assembled PCB.  

The software is the same on both boards, and the CT numbering was only a software change to get rid of CT 0 and have the numbering start at CT 1 instead.  Now, if you have an existing power monitor running an early version of the software (v0.1.x), your CT numbers in the power monitor database will be 0 - 5.  The CT numbers were changed in v0.2.x, and I wrote a script to convert the channel numbers in your existing data so you can run the latest software without things getting mixed up.  Some notes on the upgrade process are here: https://github.com/David00/rpi-power-monitor/issues/87

gaz99gh, 2023-11-02:
OK, thanks for the update, the new board with the screw terminals is much better that the old one but as I have a couple of the old one I may as well use them.

I'll be starting with a fresh install of V0.3 so should be fine.

closed 2023-11-02 by gaz99gh