David00/rpi-power-monitor #117: Grafana Dashboards "relative time" should be 30s

JKJameson, 2023-11-15:
I would like to propose a change with the Grafana Dashboards, but the sources aren't in the repo so I can't create a pull request.

The "Relative Time" setting for the Query should be "30s" instead of the default "5s". Otherwise, the dashboard will report null values randomly if the power monitor hasn't sent an update to InfluxDB in the past 5 seconds.

Thanks for the amazing hardware guide and software, this was a fun project!

David00, 2023-11-17:
Hey @JKJameson, thanks for opening an issue. Are you seeing this happen on any specific panels or queries? I don't think any of the provided panels or queries in the two dashboards I've released have this issue, as most of them either select the last value, or have a large enough time interval to the point that if there's no data available, there's a bigger issue at hand.

JKJameson, 2023-11-21:
Hi,
Perhaps it's a problem with my data collection, but it does seem to be a fronted graphing issue as far as I can see.

Here's an example of the graph problem. At the end of the graph, it reports zero, but my live reading is actually 356 watts at that time.

[image]

If I refresh a few seconds later, the graph shows an actual reading.

[image]

Similarly, the stats at the top of the dashboard will report "No data" due to the same problem.

I have fixed it by adjusting the Relative Time from the default of 5 seconds to 30 seconds. To get the graphs to stop showing zero at the end, I have to set to 1m resolution. As the graph still shows the latest data, I see no drawback. You can also zoom in the graph quite far and I no longer see this monstrosity.

[image]

Here is with 30s min interval set.

[image]

Finally, 1m interval set. I do see some curving out of the data with this setting, but once again - it removes the zero reading at the end of the graph.

[image]