David00/rpi-power-monitor #119: power-monitor.service is failing to start

lenrsmith, 2023-11-17:
I updated to the most recent image, 0.3.1, and I had everything working great last night. Today, when I checked on things, no data was displayed in the Grafana dashboard. Upon closer investigation, I found the following logs:

> Nov 17 17:18:17 powermonitor systemd[1]: Started Raspberry Pi Power Monitor.
Nov 17 17:18:17 powermonitor systemd[1]: power-monitor.service: Main process exited, code=exited, status=217/USER
Nov 17 17:18:17 powermonitor systemd[1]: power-monitor.service: Failed with result 'exit-code'.
Nov 17 17:18:19 powermonitor systemd[1]: power-monitor.service: Scheduled restart job, restart counter is at 413.
Nov 17 17:18:19 powermonitor systemd[1]: Stopped Raspberry Pi Power Monitor.
Nov 17 17:18:19 powermonitor systemd[1337]: power-monitor.service: Failed to determine user credentials: No such process
Nov 17 17:18:19 powermonitor systemd[1337]: power-monitor.service: Failed at step USER spawning /usr/bin/python3: No such process

Any ideas on what could be going on here?

lenrsmith, 2023-11-17:
Solved this on my own. The default power-monitor.service file references the user `pi`. Because I setup a custom user, I needed to to change the reference to user `pi` and the directory location in the `ExecStart` attribute to correspond to my custom user/directory combo.

David00, 2023-11-17:
Hi @lenrsmith, thanks for following up with your fix and closing the issue. 

There's definitely an opportunity to improve the docs about this, so I will reopen the issue, add a note to the [Running as a Service](https://david00.github.io/rpi-power-monitor/docs/v0.3.0/advanced-usage.html#running-as-a-service) page in the documentation, and then close this issue with a commit containing the improvement to the docs.

lenrsmith, 2023-11-17:
Awesome! If I can help in any way, please let me know. I am not super proficient with python, but I am a developer. So thankful for this project, so I am happy to pitch in somehow.

David00, 2023-11-17:
Thanks for the offer :) This was a quick and easy change so I went ahead and took care of it. I'm sure there are other improvements to made and additional opportunities to get involved, though!

The change is now live on https://david00.github.io/rpi-power-monitor/docs/v0.3.0/advanced-usage.html#running-as-a-service

commit 16e9c61b (2023-11-17): Add note about using the non-default username in the service file. Closes #119
closed 2023-11-17 by David00