David00/rpi-power-monitor #122: Negative Power Factor / 240v

sogorman, 2024-01-28:
Good morning, I am getting a new install setup and configured. I have CT3 and CT4 on a 240v PV generation circuit  (config below). When running the the power monitor the power factor for one of the legs is negative which is throwing off the measurements. Ideas?

```
pi@rpipowermonitor:~ $ python3 ~/rpi_power_monitor/rpi_power_monitor/power_monitor.py --mode terminal
INFO : ... Starting Raspberry Pi Power Monitor
INFO : Press Ctrl-c to quit...
INFO : 
+-------------+---------+------+---------+----------+-----+-----+
|             |   ct1   | ct2  |   ct3   |   ct4    | ct5 | ct6 |
+-------------+---------+------+---------+----------+-----+-----+
|    Watts    |    0    |  0   | 975.282 | -976.732 |  0  |  0  |
|   Current   |    0    |  0   |  9.072  |   9.06   |  0  |  0  |
|     P.F.    |    0    |  0   |  0.841  |  -0.843  |  0  |  0  |
|   Voltage   | 127.853 |      |         |          |     |     |
| Sample Rate |  37.49  | kSPS |         |          |     |     |
+-------------+---------+------+---------+----------+-----+-----+
INFO : 
+-------------+---------+------+---------+----------+-----+-----+
|             |   ct1   | ct2  |   ct3   |   ct4    | ct5 | ct6 |
+-------------+---------+------+---------+----------+-----+-----+
|    Watts    |    0    |  0   | 959.483 | -961.168 |  0  |  0  |
|   Current   |    0    |  0   |  8.965  |  8.955   |  0  |  0  |
|     P.F.    |    0    |  0   |   0.84  |  -0.842  |  0  |  0  |
|   Voltage   | 127.485 |      |         |          |     |     |
| Sample Rate |   37.4  | kSPS |         |          |     |     |
+-------------+---------+------+---------+----------+-----+-----+
INFO : 
+-------------+---------+------+---------+----------+-----+-----+
|             |   ct1   | ct2  |   ct3   |   ct4    | ct5 | ct6 |
+-------------+---------+------+---------+----------+-----+-----+
|    Watts    |    0    |  0   | 969.206 | -970.976 |  0  |  0  |
|   Current   |    0    |  0   |  8.998  |  8.985   |  0  |  0  |
|     P.F.    |    0    |  0   |  0.842  |  -0.845  |  0  |  0  |
|   Voltage   | 127.925 |      |         |          |     |     |
| Sample Rate |  37.45  | kSPS |         |          |     |     |
+-------------+---------+------+---------+----------+-----+-----+
INFO : 
+-------------+---------+------+---------+----------+-----+-----+
|             |   ct1   | ct2  |   ct3   |   ct4    | ct5 | ct6 |
+-------------+---------+------+---------+----------+-----+-----+
|    Watts    |    0    |  0   | 952.866 | -953.815 |  0  |  0  |
|   Current   |    0    |  0   |  8.908  |  8.895   |  0  |  0  |
|     P.F.    |    0    |  0   |  0.839  |  -0.841  |  0  |  0  |
|   Voltage   | 127.524 |      |         |          |     |     |
| Sample Rate |  37.61  | kSPS |         |          |     |     |
+-------------+---------+------+---------+----------+-----+-----+
^Cpi@rpipowermonitor:~ $ ^C

```

```

general]
name = "PORT AU PRINCE"            # Enter a name for this power monitor. This name will be used to tag all entries in the DB, which is useful 
                            # if you have multiple monitors saving data to a centralized InfluxDB server.

# Enable 3-phase mode by setting this to true.
three_phase_mode = false

[database]
host = "localhost"
port = 8086
username = "root"
password = "password"
database_name = "power_monitor"

[grid_voltage]
grid_voltage = 123.8
ac_transformer_output_voltage = 10.5
frequency = 60
voltage_calibration = 1

---- SOME CT REMOVED ----

[current_transformers.channel_3]
name = 'SOLAR PHASE A'
rating = 60
type = 'production'
two_pole = false
enabled = true
calibration = 1.0
watts_cutoff_threshold = 0
reversed = false
phase_angle = 0

[current_transformers.channel_4]
name = 'SOLAR PHASE B'
rating = 60
type = 'production'
two_pole = false
enabled = true
calibration = 1.0
watts_cutoff_threshold = 0
reversed = false
phase_angle = 0

```

David00, 2024-01-29:
Hi @sogorman, the leg is negative due to the orientation of the sensor.  You can fix this several ways:

* Set `reversed = true` in the config file for `channel_4`, or
* Swap the polarity of CT4's wires connected to the PCB, or
* Unclip the CT and reverse the direction that it's clamped over the wire

Yes, this will mean that the sensors technically have different orientations - the reason is because the current waveform on each leg is actually opposite from one another.

Some additional information is here in the Post Install Validation section of the docs:

https://david00.github.io/rpi-power-monitor/docs/general/hardware-installation.html#post-install-validation

David00, 2024-02-04:
Hi @sogorman, going to go ahead and close this now, but feel free to follow up if you need some more assistance :)

closed 2024-02-04 by David00