David00/rpi-power-monitor #129: Average consumption per hour of the day - grafana panel

averter, 2024-04-21:
I'm struggling to create a grafana panel that displays hours of the day in the X axis and average consumption (power) in the Y axis. In influxdb v2 this is a relatively easy task, see an example [here](https://community.influxdata.com/t/group-by-every-hour-of-the-day/24213), but what for V1? This could be useful to many who use the rpi-power-monitor I think. Thanks.

David00, 2024-04-22:
Hey @averter, start by identifying the name of the measurement that you want to view the average power for. Details for the database measurement names (and fields associated with each) are in the docs here:

https://david00.github.io/rpi-power-monitor/docs/v0.3.0/database-info.html#structure

Also note the difference between the high resolution measurements, and the downsampled 5 minute averages here:
https://david00.github.io/rpi-power-monitor/docs/v0.3.0/database-info.html#data-retention-and-continuous-queries

I'm assuming you want to average the power every hour? If so, we'll use the `GROUP BY time(1h)` statement in the query.  Most of the power graphs in the [provided default dashboards](https://david00.github.io/rpi-power-monitor/docs/v0.3.0/accessing-the-dashboard.html#viewing-dashboards) are already doing what you want, but on a much smaller GROUP BY time() interval so you can see the near real time load.

Here's a query that will do what you want using the high resolution data:

`SELECT mean("power") FROM "home_load" WHERE $timeFilter GROUP BY time(1h)`

[image]

If you want to look back in time beyond the 30 day limit for high resolution data, you'll need to adjust the query to pull from the `rp_5min` retention policy, like this:

`SELECT mean("power") FROM "rp_5min"."home_load_5m" WHERE $timeFilter GROUP BY time(1h)`

averter, 2024-04-23:
Thanks @David00. I was thinking of using the long-term data from the rp_5min retention policy, so that I can look at a longer time horizon, 6 months (say). The main problem is always how to group the data for those 6 months based on the hour of the day. I assume you achieve this via `$timeFilter` ? How is it defined? At the moment with your code I get an x-axis spanning 6 months with a point for each and every hour.

David00, 2024-04-23:
So the `$timeFilter` variable is specific to Grafana, and it just represents the start and stop times as selected in the Grafana time range selector. 

> The main problem is always how to group the data for those 6 months based on the hour of the day

I see, I wasn't understanding what you were trying to achieve, but I do now. It seems that you want to view average power over the last six months for each hour-long period within a day, like "what's my average power between 11am - 12pm each day for the last six months?".

I don't think that's possible in Influx v1's InfluxQL.  However, as you've pointed out, you can do that in Influx v2; more specifically, it's the Flux language that enables it, which is default in Influx v2. 

You can actually enable Flux in v1.8+ and give it a shot!  See the Influx docs for details on enabling it:
https://docs.influxdata.com/influxdb/v1/flux/installation/

Your old queries and Grafana panels will still work fine. To use Flux though, you'll have to setup a new datasource in Grafana, and set the Query language to Flux.

Details on how to configure the new data source in Grafana are here:

https://docs.influxdata.com/influxdb/v1/tools/grafana/?t=Flux#configure-grafana-to-use-flux

When you create a new query in Grafana, just select the new Flux enabled data source, and then you can experiment with the queries there. 

The initial Flux query framework for the power field of the home_load_5m measurement would look like this:

```flux
import "date"
from(bucket: "power_monitor/rp_5min")
  |> range(start: v.timeRangeStart, stop: v.timeRangeStop)
  |> filter(fn: (r) => r["_measurement"] == "home_load_5m")
  |> filter(fn: (r) => r["_field"] == "power")
```

averter, 2024-04-24:
I had NO IDEA that I could use flux in v1.8. Thanks very much for sharing this info and the other links are also super helpful. 
I've tried and unfortunately it seems that the flux version that ships with 1.8.10 contains a [bug in the mean function](https://github.com/influxdata/influxdb/issues/21649), as I always get a panic:runtime error ... 
However I've been able to use the max and min functions, which "sort of" give me an idea of where my power consumption is along the day. I'm sharing the flux query below for anyone interested 
````flux
import "date"
from(bucket: "PMONITOR/rp_5min")
  |> range(start: v.timeRangeStart, stop: v.timeRangeStop)
  |> filter(fn: (r) => r._measurement == "home_load_5m"  and r._field == "power")
  |> map(fn: (r) => ({ r with hour: date.hour(t: r._time) }))  
  |> group(columns: ["hour"], mode:"by")
  |> max(column: "_value") 
  |> drop(columns: ["_time", "_start", "_stop", "_field", "_measurement"])
  |> group()
  |> sort(columns: ["hour"])
````
as well as a chart. Notes: this is a XY chart grafana panel and the units are not correct (should be kW rather than MWh)
[image]

closed 2024-04-24 by averter

David00, 2024-04-24:
Very cool @averter, thanks for sharing! You could probably calculate the mean on your own through the use of `map()` and/or `aggregateWindow()`. 

I'd really like to make Influx v2 the default for this project sometime soon. It won't be too hard of a switch from the code side - the hard part will be handling users that want to upgrade from v1 to v2! I'll have to look into that.

averter, 2024-04-24:
I've tried to replace mean with àggregateWindow()` or `sum()` but anything that depends of a sum or mean always outputs that error... It's surprising given how fundamental these operations are.
I'll keep trying different ways to replace the mean function and report here if I find a workaround. Thanks again.

David00, 2024-04-24:
That's a big bummer! Out of curiosity, I was looking into the upgrade options for migrating from v1 to v2. It looks like Influx has upgrade functionality built in that will handle migrating all the data:
https://docs.influxdata.com/influxdb/v2/install/upgrade/v1-to-v2/automatic-upgrade/

... but the continuous queries will have to be manually rewritten to convert them into tasks.

This is definitely something I'd like to get to eventually, but there will be a lot to it, including new documentation, new code, and lots of testing.

averter, 2024-04-25:
I can imagine all the work involved in such change, but it would be worth it (considering the flexibility of the Flux language)! 
Meanwhile I've managed to create a proxy of the plot I was looking for, by using median() instead of the mean(). By definition, the median is less sensitive to outliers, and therefore more representative of the actual power consumption (if it contains occasional spikes) which I think is the case for my consumption.

Plot of consumption in a typical day
[image]

Plot of the median consumption per hour of the day for a duration of 9 months
[image]

and the corresponding flux query is
````
import "date"
from(bucket: "PMONITOR/rp_5min")
  |> range(start: v.timeRangeStart, stop: v.timeRangeStop)
  |> filter(fn: (r) => r._measurement == "home_load_5m"  and r._field == "power")
  |> map(fn: (r) => ({ r with hour: date.hour(t: r._time) }))  
  |> group(columns: ["hour"], mode:"by")
  |> median(column: "_value") 
  |> drop(columns: ["_time", "_start", "_stop", "_field", "_measurement"])
  |> group()
  |> sort(columns: ["hour"])
````